jgromes/RadioLib #112: APRS Support

jgromes, 2020-02-14:
RadioLib supports AX.25 as of update 3.3.0, it should be possible to add support for APRS as well.

rnbokade, 2020-03-22:
APRS uses AFSK. However, most modules supported here use GFSK or FSK. which is not supported by AFSK receivers.

jgromes, 2020-03-22:
That doesn't mean it can't be implemented, in a similar way as AX.25. The only issue with decoding is that you have to demodulate the signal as CW, and then decode the sound as AFSK. I was able to decode AX.25 sent by SX1268 using soundmodem without any problems.

There's also the option of implementing "real" AFSK for modules that have FSK direct mode (SX1278 and CC1101). Then you can feed the module PWM, and this will result in AFSK (which can be demodulated as FM).

rnbokade, 2020-03-22:
That's fantastic info. Didn't know this. This will help me a lot in my
satellite. We are using DRF1268 in it. Wanted to have some easily
decodeable telemetry modulation for it. Otherwise it was using LORA. This
solves a lot of problems. Will appreciate your help in our satellite's
program
Thanks in advance.

rnbokade, 2020-03-22:
Any way the same could be decoded in GNU-RADIO?? If this can be done, it will be boon for us.

jgromes, 2020-03-22:
Most likely so. Like I said, you only have to demodulate as CW and then decode the APRS. Not sure if there's an APRS decoder available in GNURadio though - this currently isn't being worked on, so I wasn't looking for APRS decoder to verify with yet.

rnbokade, 2020-03-22:
is there any way to similarly decode FSK sent from sx1268 modules. I mean e.g. I am sending AX.25 at 9.6kbps. Then how can I decode it? is there a way in soundmodem or some other software

jgromes, 2020-03-22:
Do you want to decode FSK or AX.25? If it's just FSK, then you can use [Universal Radio Hacker](https://github.com/jopohl/urh) - it will decode 9k6 (G)FSK even at 5 kHz frequency deviation. It can't handle SX126x whitening yet, I'm trying to figure that one out myself.

If it's AX.25, then you can use something like SDRConsole or SDRSharp to demodulate the FSK as CW and pipe the audio to [soundmodem](http://uz7.ho.ua/packetradio.htm) (using virtual audio cable). If you're on Linux, you could also use GQRX to decode the AX.25 frames.

rnbokade, 2020-03-22:
AX.25 is just a packet format right? for making fsk transmission compatible with AFSK reciever I'll have to demodulate it via cw right??

rnbokade, 2020-03-22:
I went into sx1268's datasheet. one can disable whitening.

jgromes, 2020-03-22:
I think you might be misunderstanding - if you use the `AX25Client` class, RadioLib will handle all required configuration for you, including the encoding. The note about whitening would only apply if you wanted to decode plain FSK.

MrTalon63, 2021-09-25:
So, is it possible now to send APRS compatible data with CC1101?

jgromes, 2021-09-25:
I don't see why not, though I haven't tested it with actual repeaters.

jgromes, 2021-11-28:
Official APRS support is now in. Position reporting only so far, but it should be fairly easy to add in more data types.

closed 2021-11-28 by jgromes