jgromes/RadioLib #115: CC1101 | Arduino UNO Wifi Rev2 Error Code -2

trisp3ar, 2020-02-20:
Hi,

I have a CC1101 Modul connected to an UNO Wifi Rev 2.0 in standard configuration running the receive example. When initializing I get error code -2 (Radio chip was not found).
I've tested the sender/receiver examples on NANOs and it's working there.

- IDE v1.8.10
- lib v3.3.0

jgromes, 2020-02-20:
Please include debug mode output: uncomment [debug macro definitions in TypeDef.h](https://github.com/jgromes/RadioLib/blob/master/src/TypeDef.h#L36) and post the output.

ERR_CHIP_NOT_FOUND is usually incorrect wiring, please double check that too.

trisp3ar, 2020-02-20:
I checked the wiring like ten times, it's correct.

Console Output:

`12:40:17.139 -> [CC1101] Initializing ... CC1101 not found! (1 of 10 tries) CC1101_REG_VERSION == 0x00FF, expected 0x0014`
`12:40:18.179 -> CC1101 not found! (2 of 10 tries) CC1101_REG_VERSION == 0x00FF, expected 0x0014`
`12:40:19.219 -> CC1101 not found! (3 of 10 tries) CC1101_REG_VERSION == 0x00FF, expected 0x0014`
`12:40:20.219 -> CC1101 not found! (4 of 10 tries) CC1101_REG_VERSION == 0x00FF, expected 0x0014`
`12:40:21.219 -> CC1101 not found! (5 of 10 tries) CC1101_REG_VERSION == 0x00FF, expected 0x0014`
`12:40:22.259 -> CC1101 not found! (6 of 10 tries) CC1101_REG_VERSION == 0x00FF, expected 0x0014`
`12:40:23.259 -> CC1101 not found! (7 of 10 tries) CC1101_REG_VERSION == 0x00FF, expected 0x0014`
`12:40:24.285 -> CC1101 not found! (8 of 10 tries) CC1101_REG_VERSION == 0x00FF, expected 0x0014`
`12:40:25.285 -> CC1101 not found! (9 of 10 tries) CC1101_REG_VERSION == 0x00FF, expected 0x0014`
`12:40:26.325 -> CC1101 not found! (10 of 10 tries) CC1101_REG_VERSION == 0x00FF, expected 0x0014`
`12:40:27.325 -> No CC1101 found!`
`12:40:27.365 -> failed, code -2`

jgromes, 2020-02-20:
> I checked the wiring like ten times, it's correct.

How did you verify it? For example, do you have an oscilloscope to check the SPI lines, or another SPI device to test that SPI is indeed working as expected? Have you tested the wires for continuity?

I'm not sure where SPI pins are located on Uno Wifi, but [this page](https://www.arduino.cc/en/Guide/ArduinoUnoWiFiRev2#toc10) seems to imply it's only available on the ICSP connector, and not on the usual D13-D11 pins as on Uno.

trisp3ar, 2020-02-20:
I understand, SPI was indeed only available at ICSP. Is there any chance I can redefine the ICSP pins for use with the CC1101? If so what's the best way to go?

jgromes, 2020-02-20:
Take a look at the schematic: https://content.arduino.cc/assets/UNOWiFiREV2V4.0_sch.pdf The SPI pins (PC0/PC1/PC2) are only available on the ICSP connector, so in order to use the hardware SPI, you have to connect CC1101 to the ICSP connector.

Alternatively, you could try to find software SPI implementation for Uno Wifi, and use that on any pins it will work with RadioLib as long as it's compatible with Arduino `SPIClass`.

closed 2020-02-20 by jgromes