jgromes/RadioLib #118: Feature Request for SX126x/SX127x Continuous Wave (CW) transmission

who-has-seen-the-wind, 2020-02-26:
On the SX127x and the newer SX126x, it would be useful to be able to transmit a continuous wave (CW) signal, i.e. an unmodulated carrier. One reason would be to look at oscillator drift, which is an issue for Lora with longer symbols and packets. 

I have used RadioLib to transmit bursts of a CW signal (by setting the FSK  frequency deviation to zero), but a continuous signal would also be nice for drift, power measurements, and possibly FCC tests.

(The nice RadioLib library  operation does support FSK direct data transmission,
but that means that one must provide data and clock DIO1 and DIO2.)

A solution within RadioLib seems possible, since the Semtech library has, for example, SX1276SetTxContinuousWave()
https://github.com/Lora-net/LoRaMac-node/blob/develop/src/radio/sx1276/sx1276.c#L1139

RadioLib seems to be a terrific library!

jgromes, 2020-02-26:
RadioLib already has this feature:

https://github.com/jgromes/RadioLib/blob/1a721715ea974c3410f12b0d2aaad7561ab3f42c/src/protocols/PhysicalLayer/PhysicalLayer.h#L163-L171

It's available for all modules that implement PhysicalLayer, i.e. CC1101, RF69/SX1231, nRF24, SX127x/RFM9x and SX126x.  This is how RadioLib does RTTY and Morse code transmissions.

> RadioLib library operation does support FSK direct data transmission,
but that means that one must provide data and clock DIO1 and DIO2

You don't have to provide clock and data, if you don't, the module will transmit unmodulated carrier.

closed 2020-02-26 by jgromes

Bushpulbek, 2023-11-22:
Hello, everybody!

Am I right to understand that in order to obtain an unmodulated carrier 
- of some frequency _freq_ 
- on the **PA_BOOST** pin 
- of SX127x chip (SX1276, in particular) 

the following code from the examples will do?
```cpp
// activate direct mode transmitter
  state = radio.transmitDirect(freq);
  if (state != RADIOLIB_ERR_NONE) {
    Serial.println(F("[SX1278] Unable to start direct transmission mode, code "));
    Serial.println(state);
  }
```

jgromes, 2023-11-22:
@Bushpulbek I would like to point out that this thread is more than 3 years old at this point...

`transmitDirect` is correct, though you seem to have not read the documentation - `freq` is raw 24-bit value, which is probably not what you want. Just call `transmitDirect()` and the frequency you passed to `begin()` will be used.

Regarding the output pin, you need to configure it via `setOutputPower` - again, it's described in the documentation. PA_BOOST is used by default though.