jgromes/RadioLib #120: How to receive "0x00" byte?

sjd227, 2020-02-27:
Hello,

I'm using the sx1272 in OOK mode, with Manchester encoding enabled. Some of the bytes I need to receive are `0x00`, which turns into `b0101010101010101` = `0x5555` via Manchester encoding. The received packets are always being cut off wherever there's a `0x00` byte - maybe because a new preamble detector interrupt is being triggered by `0x5555` and the FIFO stops filling up? 

How can I avoid this such that I can continue to receive the full length of the message, regardless of any `0x00` bytes in the message? 

I've tried using `fixedPacketLengthMode` to set the length to the known packet length, but it still keeps getting cutoff whenever there's a `0x00`.

Thanks,
Steven

jgromes, 2020-02-27:
Are you reading the data as a string, or as a byte array? If it's the former, the string will cut off at the null byte (since that's how C-strings are defined).

sjd227, 2020-02-27:
Oh interesting; it looks like I'm reading the data as a string. 

I'm working off the "FSKModem.ino" example file, which uses: 

`    // receive FSK packet`
`String str;`
    `state = fsk.receive(str);`

How should it be changed to read the data byte array instead?

jgromes, 2020-02-27:
It's written in the same example ;)

https://github.com/jgromes/RadioLib/blob/1a721715ea974c3410f12b0d2aaad7561ab3f42c/examples/SX127x/SX127x_FSK_Modem/SX127x_FSK_Modem.ino#L118-L121

You can use that if you know the packet length in advance, or call `getPacketLength()` to get the number of received bytes.

Also, please include your code and debug output when opening new issues, thanks.

jgromes, 2020-03-12:
Closed due to inactivity.

closed 2020-03-12 by jgromes