jgromes/RadioLib #124: SX127x reset values reversed?

sjd227, 2020-03-13:
Looking at the sx1272 datasheet, it looks like reset is done by pulling the reset pin **HIGH** for 100 microseconds, then releasing it **LOW**:

[image]

But in sx127x.cpp it seems reversed:

```
void SX127x::reset() {
     Module::pinMode(_mod->getRst(), OUTPUT);
     Module::digitalWrite(_mod->getRst(), LOW);
     delayMicroseconds(100);
     Module::digitalWrite(_mod->getRst(), HIGH);
     delay(5);
}
```
Thoughts?

- Steven

jgromes, 2020-03-13:
It looks like SX1272 has a positive-logic reset (RST=1 to reset), while SX1278 has negative logic (NRST=0 to reset). The current implementation will work for for SX1276/77/78/79, though not for SX1272/73. Will fix asap.

jgromes, 2020-03-13:
Fixed in c1c991acc831cbb457cf96037eb9cf4ced261a6f, will release the fix as 3.4.0.  Thanks for reporting!

closed 2020-03-13 by jgromes

sjd227, 2020-03-13:
Sounds good, thanks!

naggie, 2020-03-16:
This explains a lot. Thanks @sjd227 and @jgromes . Definitely encountered this