jgromes/RadioLib #125: Problem with CC1101 sending data

t351206, 2020-03-14:
Sending data with a CC1101 module and an Arduino Nano clone is not correctly working. 
I am expecting to receive (with the help of SDR and Universal Radio Hacker) the preamble (4 bytes) + sync (2 bytes) + 4 * (11111111 00000000)

but I am receiving the following:
10101010101010101010101010101010 (preamble) + 0010110111010100 (sync) +
00000000111000011110001010011010000100101000010111001100001001000

Any ideas why I am not receiving the correct payload (see coede below)

```
#include <RadioLib.h>

// CC1101 has the following connections:
// CS pin:    10
// GDO0 pin:  2
// RST pin:   unused
// GDO2 pin:  3
CC1101 cc = new Module(10, 2, NC, 3);

void setup() {
  Serial.begin(9600);

  // initialize CC1101
  Serial.print(F("[CC1101] Initializing ... "));

  int state = cc.begin(868.325, 17.240, 48.0, 325.0, 0 , 4);
  if (state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }

    // 2 bytes can be set as sync word
  //byte syncword[2] = {0xd4,0x2d};
  if (cc.setSyncWord(0x2d,0xd4) == ERR_INVALID_SYNC_WORD) {
    Serial.println(F("[CC1101] Selected sync word is invalid for this module!"));
    while (true);
  }

  int crc_state = cc.setCrcFiltering(false);
  if (crc_state == 0) {
    Serial.println(F("CRC off "));
  }
  else {
    Serial.print(F("failed, code "));
    Serial.println(crc_state);
    while (true);
  }

 
  // address filtering can also be disabled
  // NOTE: Calling this method will also erase previously
  //       set node address
 
    Serial.print(F("[CC1101] Disabling address filtering ... "));
    state == cc.disableAddressFiltering();
    if(state == ERR_NONE) {
      Serial.println(F("success!"));
    } else {
      Serial.print(F("failed, code "));
      Serial.println(state);
      while(true);
    }

  // enable fixed packet length mode
 
    Serial.print(F("[CC1101] Start fixed packet length mode ... "));
    state == cc.fixedPacketLengthMode();
    if(state == ERR_NONE) {
      Serial.println(F("success!"));
    } else {
      Serial.print(F("failed, code "));
      Serial.println(state);
      while(true);
    }

  // Disable broadcast address
 
    Serial.print(F("[CC1101] Disable broadcast address ... "));
    state == cc.setNodeAddress(0,0);
    if(state == ERR_NONE) {
      Serial.println(F("success!"));
    } else {
      Serial.print(F("failed, code "));
      Serial.println(state);
      while(true);
    }

 
}

void loop() {
  Serial.print(F("[CC1101] Transmitting packet ... "));

    byte byteArr[] = {0xFF,0x00,0xFF,0x00,0xFF,0x00,0xFF,0x00};

    int state = cc.transmit(byteArr, 8);
   
 

  if (state == ERR_NONE) {
    // the packet was successfully transmitted
    Serial.println();

    for (int i = 0; i < 9; i++) {
      printBits(byteArr[i]);
      Serial.print(" ");
    }
    Serial.println();

    Serial.println(F("success!"));

  } else if (state == ERR_PACKET_TOO_LONG) {
    // the supplied packet was longer than 64 bytes
    Serial.println(F("too long!"));

  } else {
    // some other error occurred
    Serial.print(F("failed, code "));
    Serial.println(state);

  }

  // wait for a second before transmitting again
  delay(4000);
}

  void printBits(byte n) {
  byte numBits = 8;  // 2^numBits must be big enough to include the number n
  char b;
  char c = ' ';   // delimiter character
  for (byte i = 0; i < numBits; i++) {
    // shift 1 and mask to identify each bit value
    b = (n & (1 << (numBits - 1 - i))) > 0 ? '1' : '0'; // slightly faster to print chars than ints (saves conversion)
    Serial.print(b);
    if (i < (numBits - 1) && ((numBits-i - 1) % 4 == 0 )) Serial.print(c); // print a separator at every 4 bits
  }
}

```

jgromes, 2020-03-14:
By default, CC1101 is set to use whitening, so changed payload is expected. Try setting encoding to NRZ using `setEncoding(0)`.

t351206, 2020-03-14:
Great! This is the solution! Now I am getting the right payload! Thanks a lot!

Your documentation is already very comprehensive and good but would be perfect if the standard settings are included in the documentation.

jgromes, 2020-03-14:
All of the basic parameters are already written in the examples (though to be fair, the default encoding isn't one of them). The issue is I can't really document everything, there's a lot of configuration that isn't accessible via the public API.

RadioLib isn't targeted at beginners - there's a plenty of other Arduino libraries that are "good enough" for that. The target audience here are more advanced users, which already have some basic understanding of electronics and radio. The advantage is that you get a lot of flexibility which isn't available in other libraries.

closed 2020-03-14 by jgromes