jgromes/RadioLib #126: delayMicroseconds possible inaccuracy

naggie, 2020-03-16:
Hi @jgromes,

Looking at the implementation of `delayMicroseconds` in the AVR Arduino core, there are several implementations dependent on CPU frequency. Only a handful of standard frequencies are supported -- if a non-standard CPU frequency is used, the calls to `delayMicroseconds` will result in inaccurate (short, probably) timing.

This is an issue for the reset methods which rely on this function for boards with a non-standard frequency. 

As LoRa is targeted at low-power devices, I consider it likely that users of RadioLib will encounter this as it's often desirable to use a lower CPU frequency for such low-power devices.

Would you accept a PR to use `delay(1)` where appropriate instead? As far as I can see, the `delay` implementation uses timers with a counter value dependent on F_CPU. Not only that, but using a delay of 1ms would add significant margin.

Thanks!
Callan

jgromes, 2020-03-16:
Thanks for the warning - didn't know about that, I alwas assumed it's going to wait slightly longer (function overhead etc.).

Sure - go ahead, I'll accept that PR.

naggie, 2020-03-16:
Great, thanks. The function overhead is accounted for in the implementation, so there is effectively a minimum delay not an additional delay.

closed 2020-03-16 by jgromes