jgromes/RadioLib #128: morse code tone frequency???

csurf, 2020-03-20:
I tried using the included morse examples in order to generate morse tones with an RFM96* module. However, it doesn't actually produce tones; it simply produces a non-audible tone that breaks squelch. Is there a way to created an audible morse tone at an arbitrary frequency (via a basic on/off pin toggle)? 

FYI, I've managed to do achieve this with an SI4432-based module (RFM22B), by putting the module into "direct mode" and creating a toggled/PWM signal on a 'direct'-input pin. Is something like this possible with the sx127* ???

jgromes, 2020-03-20:
The way Morse code is implemented is just by outputting unmodulated carrier wave, so if you demodulate it as CW, you can then decode the Morse code symbols (e.g. using Fldigi) - don't know if this can be used in your application, you haven't provided much information about it.

RFM9x/SX127x has a similar direct mode as Si4432. This mode is active when `MorseClient` is running, so you should be able to just apply PWM to get tones: https://www.youtube.com/watch?v=a0CyXdC6qJg

jgromes, 2020-04-03:
Closed due to inactivity.

closed 2020-04-03 by jgromes