jgromes/RadioLib #129: Additon of SSDV protocol

rnbokade, 2020-03-26:
**Describe the solution you'd like**
A way to use ssdv with rf modules with the ease with which we are able to use AX.25, Morse, RTTY client etc.

**Describe alternatives you've considered**
Dave Akerman's lora ssdv

**Additional context**
reference :- https://github.com/fsphil/hadie

jgromes, 2020-03-26:
I'm assuming you're referring to https://ukhas.org.uk/guides:ssdv

Is there a complete specification somewhere? The above link only contains basic info, such as the packet structure, but doesn't go into much detail about transfer control.

Should be possible, though I'm not sure about the API, since there will be a lot of packets that need to be sent in a sequence. Maybe something like `startTransfer(imageId, , *imagePtr, imageLen)` to initialize the transfer of a single image, and than have the user call `sendPacket()` in a loop until everything is sent?

rnbokade, 2020-03-26:
Indeed I am referring to ukhas ssdv . Although I also wasn't able to find
full specs of protocol anywhere. Although Fsphil's code is very
explanatory. Will try to come up with API in 1-2 days and try my level to
rather code thia myself and create a PR. I really want to follow your
footsteps in awesome code. I think this might just be the chance to get my
hands dirty. And I hope if I am stuck u will be there to help

jgromes, 2020-03-26:
Nice! Please make sure to follow the [contributing doc](https://github.com/jgromes/RadioLib/blob/master/CONTRIBUTING.md#code-style-guidelines).

jumbo5566, 2023-02-28:
yes ,you can view this https://github.com/TomasTT7/LoRa_SSDV

Mika-1818, 2024-05-20:
Sorry to reactivate this, but I can see the implemention but not how to use it.
Maybe someone can help me.

closed 2026-10-03 by jgromes