jgromes/RadioLib #130: Question :- How to switch between different modems of same module?

rnbokade, 2020-03-26:
My application requires me to listen (non-blocking) in lora mode for 40 seconds, after which I send a string by RTTY, followed by a byte array being sent over FSK at 9600 bps, then I switch back again to lora listening. 
Ps:- This is basically for my satellite which is based on DRF1268T and my uplink is lora, and beacon is RTTY, telemetry and downlink is fsk 9600, how to switch between all 3 of these.

jgromes, 2020-03-26:
For SX126x, you can call `beginFSK` at any time to initialize FSK modem, and `begin` to initialize LoRa modem.

If you also switch between RTTY and normal FSK operation, then you have to call `RTTY::begin()` prior to sending anything via RTTY (since frequency deviation must be set to 0 to get RTTY running). When calling `RTTY::begin()`, FSK modem must be active.

So your sequence will be:

`SX1268::begin` -> LoRa listen -> `SX1268::beginFSK` -> `RTTY::begin` -> send RTTY data -> `SX1268::beginFSK` -> send FSK data.

rnbokade, 2020-03-26:
That solves it thanks

closed 2020-03-26 by jgromes