jgromes/RadioLib #132: [SX1280] Timed out while waiting for packet!

mrninhvn, 2020-04-08:
Hi, I got "[SX1280] Timed out while waiting for packet!" error when using SX128x_GFSK_Modern example.

Any help, thank you so much.

Additional info:
 - MCU: Arduino Uno.
 - Wireless module type: SX1280.
 - Arduino IDE version:  1.8.10
 - Library version: newest in master tree.

jgromes, 2020-04-08:
As the example code says:

https://github.com/jgromes/RadioLib/blob/7c9f422e0c9820c4af2d96b889eda5089353b743/examples/SX128x/SX128x_GFSK_Modem/SX128x_GFSK_Modem.ino#L6-L10

That example is just an API guide - it shows you how to use GFSK modem, instead of the default LoRa modem. To use transmit or receive actual data, you have to adapt the other examples (e.g. SX128x_Transmit and SX128x_Receive_Interrupt) to use GFSK modem. This usually only boils down to calling `beginGFSK()` instead of `begin()`.

mrninhvn, 2020-04-08:
> As the example code says:
>
> https://github.com/jgromes/RadioLib/blob/7c9f422e0c9820c4af2d96b889eda5089353b743/examples/SX128x/SX128x_GFSK_Modem/SX128x_GFSK_Modem.ino#L6-L10
>
> That example is just an [...]

Hi jgromes,
I've used SX128x_Transmit example with `beginGFSK()` and 
`state = gfsk.setFrequency(2410.5);
  state = gfsk.setBitRate(200);
  state = gfsk.setFrequencyDeviation(100.0);
  state = gfsk.setOutputPower(5);
  state = gfsk.setDataShaping(1.0);
  uint8_t syncWord[] = {0x01, 0x23, 0x45, 0x67, 0x89};
  state = gfsk.setSyncWord(syncWord, 5);`
But return the same error.

jgromes, 2020-04-08:
You don't have to change all the settings using setFrequency() etc. - once again, that's only in the example to show it can be done like this. The only modification you have to do in the other examples is call `beginGFSK()` instead of `begin()`.

Next, assuming you're using SX128x_Receive for your receiver now - timeout is expected with the basic `receive()` function. It simply means that no packet was received within the timeout period. The period is calculated based on current bit rate, and since default bit rate for SX1280 is quite high (800 kbps), it means that the receive window is very narrow. I suggest using SX1280_Receive_Interrupt instead, as that example uses external interrupt to notify Arduino of a new packet, which is a lot more reliable, as well as non-blocking.

Finally, you haven't specified which SX1280 module you're using. Some of them, such as Ebyte E28 have external PA/LNA, and need extra control on the module's TX_EN/RX_EN pins.

mrninhvn, 2020-04-08:
Hi jgromes,
I just changed `begin()` to `beginGFSK`, and get the same error.
It return error code `-5`

mrninhvn, 2020-04-08:
> You don't have to change all the settings using setFrequency() etc. - once again, that's only in the example to show it can be done like this. The only modification you have to do in the other [...]

I uses LAMBDA80C module so don't need TX_EN and RX_EN.
It work fire with lora example.

jgromes, 2020-04-08:
> It return error code -5

Does it return ERR_TX_TIMEOUT (-5) or ERR_RX_TIMEOUT (-6)? The issue title seems to suggest timeout during reception, but -5 is timeout during transmission.

mrninhvn, 2020-04-08:
> > It return error code -5
> 
> Does it return ERR_TX_TIMEOUT (-5) or ERR_RX_TIMEOUT (-6)? The issue title seems to suggest timeout during reception, but -5 is timeout during transmission.

Hi jgromes,
Sorry about my title, error code is ERR_TX_TIMEOUT (-5)

jgromes, 2020-04-08:
Could be an issue with incorrect time-on-air calculation, or possibly something else, as the SX1280 driver is brand new. I'll look into it.

mrninhvn, 2020-04-08:
@jgromes 
Do you have an example working well with your module in GFSK mode? 
Maybe I can help something

jgromes, 2020-04-08:
You can try the SX128x_Transmit_Interrupts example, that doesn't do the ToA calculation. If it gets stuck in an infinite lopp waiting for packet to be transmitted, then it's an issue somewhere else.

mrninhvn, 2020-04-09:
@jgromes 
I uses SX128x_Receive_Interrupt example, Lora mode working fire, but when replace `begin()` to `beginGFSK()` I didn't receive any packets. SX128x_Receive return TimeOut when using `beginGFSK()`.

jgromes, 2020-04-09:
As I expected, there was an issue in ToA calculation for GFSK, it should be fixed now - let me know if the issue persists, as I'm using a different SX1280 module.

I would also suggest trying to lower bit rate and frequency deviation, or trying a different carrier frequency. There's usually a lot of traffic in 2.4 GHz band - I have to increase frequency to at least 2.450 GHz to avoid active WiFi channels. Bit rate of 250 kbps with 100 kHz frequency deviation seems reliable.

mrninhvn, 2020-04-09:
Thank you @jgromes , It works for now.
Do you plan to support the BLE package for sx1280?

jgromes, 2020-04-09:
It's on the todo list, as well as the FLRC modem and ranging. But like I said, the SX1280 driver in RadioLib is brand new, so it might take me a while to get to it. I also need to get a couple of different SX1280 modules - the ones from Ebyte seem to have huge frequency drift, which unfortunately makes them unusable for most PhysicalLayer protocols.

closed 2020-04-09 by jgromes