jgromes/RadioLib #133: SX1231 is working with RF69 Setup up, but not SX1231

volpi3000, 2020-04-14:
Hello RadioLib Team,

First of all thank you so much for your awesome project, this is a real benefit for anyone working with Radio Modules and Arduino.

I have two RFM69HCW (which is as far as I'm concerned the same thing as the SX1231) from Sparkfun and connected each to an ESP8266.  I'm running the Sender Example on one and the Receiver Example on the other with no modifications  except for the CS Pin. (3.3V for the radios comes from a seperate LN Regulator).

Out of the box the example was working fine and I could receive and send messages. The only shortcome is the range. I'm lucky if I get 1 Meter of distance before they loose connection. When you put them close together, I get about -100dBm RSSI.

So I remembered from working a little bit more low level with them a while ago, that there is some kind high power option, which needs to be enabled. 

Thanks to your library it is breeze to switch over to SX1231 . So I just replaced `RF69` in the initial declaration with `SX1231`, but when I do I get this:

> Initializing ... failed, code -2

Which is some kind of SPI Error, but when I switch back to RF69 it works. This generally means that my wiring is correct.

Am I mssing something? (I attached a picture of  my simple setup, if this is relevant)

[image]

```
PLATFORM: Espressif 8266 2.4.0 > WeMos D1 R2 and mini
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 3.20603.200130 (2.6.3)
 - tool-esptool 1.413.0 (4.13)
 - tool-esptoolpy 1.20800.0 (2.8.0)
 - tool-mkspiffs 1.200.0 (2.0)
 - toolchain-xtensa 2.40802.191122 (4.8.2)

Dependency Graph
|-- <RadioLib> 3.4.0
|   |-- <EspSoftwareSerial> 6.4.0
|   |-- <SPI> 1.0
```

jgromes, 2020-04-14:
First of all, -2 is ERR_CHIP_NOT_FOUND. Most drivers will attempt to verify whether the thing they're trying to control is actually the radio the user thinks it is. If that verification fails, ERR_CHIP_NOT_FOUND is raised. In your case, you're using RFM69, so the verification works as expected for `RF69` driver. SX1231 is the same radio chip, but it has a different value in the verification register - if you enable RADIOLIB_DEBUG in BuildOpt.h, I suspect you will see something like

``` 
SX127x not found! (1  of 10 tries) SX127X_REG_VERSION == 0x0024, expected 0x0021 / 0x0022 / 0x0023
```

in Serial monitor. More to the point, why are you trying to use the SX1231 driver to control a module you know to be RFM69?

When it comes to the bad range, default output power should be set to 13 dBm, RSSI of -100 dBm is indeed rather poor, it is possible there's some configuration issue in the driver.

volpi3000, 2020-04-14:
Because I believe the RFM69HCW is the SX1231. I believe my poor range could be related to using the wrong driver since the HCW Version of the Chip offers a high power option, which i thought is enabled when using the SX1231.

volpi3000, 2020-04-14:
> indeed rather poor, it is possible there's some configuration issue in the driver.
 
I ran into the same issue with the CC1110 connected in the same fashion using your library. Worked great, but extremly low range. Keep in mind the CC1110 has actually proper attennas connected to it. So I dont believe those are the issue.

jgromes, 2020-04-14:
> Because I believe the RFM69HCW is the SX1231

That's not entirely accurate - what is sold as RFM69 is actually just the SX1231 silicon die, packaged by HopeRF into a package. There's one difference between the two, and that is the different value in one of the SPI registers. Then again, it is possible there's some other difference I've missed.

> I ran into the same issue with the CC1110

Do you mean CC1101? CC1110 is full SoC, not just radio module, and is not supported by RadioLib.

volpi3000, 2020-04-14:
I mean TI-CC1101

volpi3000, 2020-04-14:
Range wise it didn't matter if I use the CC1101 or RFM69HCW Modul, both perform very poorly. My next step could be to check if the Issue persist when I use another Library like RadioHead.

jgromes, 2020-04-14:
Checking with two CC1101 modules, I can't reproduce the issue. Next to each other on the desk, I get RSSI about -51 dBm when transmitting at the default power level 0 dBm. Raising the output power level to 10 dBm leads to the expected rise in RSSI to roughly -41 dBm, so I think the CC1101 driver handles the power level configuration correctly.

I'll check the RF69 power issue next.

jgromes, 2020-04-14:
Unfortunately I wasn't able to replicate the issue with RF69 either - once again, the RSSI roughly matches the expectations (-35 dBm with default settings and modules next to each other, changing the output power by 10 dBm has the expected effect on RSSI).

I think the modules I'm using are RFM69CW (though I can only check Google for that - I bought these quite a while ago). I think there's some slight difference in output power configuration or OCP for RFM69CW and RFM69H(CW). The datasheet doesn't specify this (since it's really just an SX1231 datasheet). Unfortunately I can't test this without the RFM69H modules, so I'm not sure there's any other way I can help you.

Let me know if you manage to fix the issue, if it's a bug in the library it should be fixed ASAP.

volpi3000, 2020-04-14:
I checked with a different Library (RadioHead). Exact same setup. I get much better RSSI (-42dBm)

This makes a RadioLib issue more likely. The main diffence, between RF69 and RF69HCW seems to be this line:

`rf69.setTxPower(14, true);`

volpi3000, 2020-04-14:
> void RH_RF69::setTxPower(int8_t power, bool ishighpowermodule = false)
> Sets the transmitter power output level.
> Be a good neighbour and set the lowest power level you need.
> Caution: legal power limits may apply in certain countries.
> After init(), the power will be set to 13dBm for a low power module.
> If you are using a high p[ower modfule such as an RFM69HW, you MUST set the power level
> with the ishighpowermodule flag set to true. Else you wil get no measurable power output.
> Simlarly if you are not using a high power module, you must NOT set the ishighpowermodule
> (which is the default)
> \param[in] power Transmitter power level in dBm. For RF69W (ishighpowermodule = false),
> valid values are from -18 to +13.; Values outside this range are trimmed.
> For RF69HW (ishighpowermodule = true), valid values are from -2 to +20.
> Caution: at +20dBm, duty cycle is limited to 1% and a
> maximum VSWR of 3:1 at the antenna port.
> \param ishighpowermodule Set to true if the connected module is a high power module RFM69HW
> 
> 

Took this from RadioHead Doc

volpi3000, 2020-04-14:
Maybe reopen ?

Pablo2048, 2020-04-14:
Hi, IIRC you have to explicitly turn on/off the PA in TESTPA1/TESTPA2 registers (available only in H modules).

volpi3000, 2020-04-14:
> Hi, IIRC you have to explicitly turn on/off the PA in TESTPA1/TESTPA2 registers (available only in H modules).

Yes, that seems correct, if you take a look here at line 395

https://github.com/adafruit/RadioHead/blob/master/RH_RF69.cpp

jgromes, 2020-04-14:
> Took this from RadioHead Doc

There's a couple of differences in RadioHead's RF69 driver and the one in RadioLib, this is the relevant section:

https://github.com/adafruit/RadioHead/blob/46ac392cc1a0e0d81ba3c4e4489e7d6b13795065/RH_RF69.cpp#L395-L437

Once again though I can't test, since the only modules I have are the "low power" RF69CW.

@Pablo2048 that should already be taken care of:

https://github.com/jgromes/RadioLib/blob/9f8ec689dd2fd8fda38c6ac1e9f788ecad748118/src/modules/RF69/RF69.cpp#L307-L308

I think there's no difference between RFM69H(CW) and RFM69CW when it comes to the actual RF chip, the differences will all be on the breakout board. RFM69H might be using the PA_BOOST pin, the others just RFIO.

volpi3000, 2020-04-14:
> RFM69H might be using the PA_BOOST pin, the others just RFIO.

At least on their schematic it doesn't look like that
RFM69H might be using the PA_BOOST pin, the others just RFIO.

jgromes, 2020-04-14:
That's just the schematic of the breakout board, I'm referring to what is on the module itself. For the low power versions, PA0 is probably used, so only RFIO would be connected to the impedance matching network and eventually the antanna. High power versions are likely using PA_BOOST to transmit, but it's just my guess.

[image]

volpi3000, 2020-04-14:
If it helps you I'm happy to supply you with some modules. I have plenty.

jgromes, 2020-04-14:
That might actually help to get this resolved quickly. The Ebay stuff is pretty cheap, but even in normal circumstances takes 3-4 weeks to arrive. Could you send me an email to the address listed in [library.properties](https://github.com/jgromes/RadioLib/blob/master/library.properties) file?

jgromes, 2020-04-21:
@volpi3000 I got the modules today (thanks!), and I can see there's definitely some issue. First of all, I'm unable to set any frequency other than 915 MHz - both modules will just straight up refuse any SPI write to the frequency registers. Did you have this problem too?

jgromes, 2020-04-21:
I modified the `setOutputPower()` method to accept a second argument which specifies whether to use high or low power module, so to set +20 dBm on RF69H, you now call `setOutputPower(20, true)`. could oyu try it out and let me know if it helps?

volpi3000, 2020-04-21:
@jgromes both modules should be 433 MHz, I tested them both on that frequency.

volpi3000, 2020-04-21:
> I modified the `setOutputPower()` method to accept a second argument which specifies whether to use high or low power module, so to set +20 dBm on RF69H, you now call `setOutputPower(20, true)`. could oyu try it out and let me know if it helps?

I will test this an come back to you.

volpi3000, 2020-04-21:
@jgromes  Alright, I tested this and now my program crashes at `rf.begin();` (which is 433MHz) and I also tried ( rf.begin(915.0, 300.0, 60.0, 250.0, 17); ) Unfortunately it still crashes at rf.begin().

Software: Example Transmitter.
Controller: ESP8266

jgromes, 2020-04-21:
Could you post the stack trace decoded with EspExceptionDecoder?

volpi3000, 2020-04-21:
```
Decoding stack results
0x402012b1: Module::init(unsigned char) at C:\Users\lukas\OneDrive\Bilder\Dokumente\Arduino\libraries\RadioLib\src\Module.cpp line 88
0x402055f1: uart_set_debug(int) at C:\Users\lukas\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266\uart.cpp line 933
0x4020218d: RF69::begin(float, float, float, float, signed char) at C:\Users\lukas\OneDrive\Bilder\Dokumente\Arduino\libraries\RadioLib\src\modules\RF69\RF69.cpp line 17
0x40203890: HardwareSerial::begin(unsigned long, SerialConfig, SerialMode, unsigned char, bool) at C:\Users\lukas\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266\HardwareSerial.cpp line 51
0x40201087: setup() at C:\Users\lukas\OneDrive\Bilder\Dokumente\Arduino\RadioLib_Transmitter/RadioLib_Transmitter.ino line 37
0x4020422c: loop_wrapper() at C:\Users\lukas\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\cores\esp8266\core_esp8266_main.cpp line 177

```

jgromes, 2020-04-21:
Looks like it crashes at 

https://github.com/jgromes/RadioLib/blob/bd5be7729e281aee5f686ada78dabecec2de83b7/src/Module.cpp#L88

which SPI bus are you using? If I recall correctly, ESP8266 has two, HSPI and VSPI.

volpi3000, 2020-04-21:
I actually have no clue, I thought this was only an issue with ESP32 which has HSPI and VSPI.
Before I updated to the new version the example was working correctly (but with the low power issue) What changes could cause this ?

jgromes, 2020-04-21:
Oops, you're right - I was thinking of ESP32, not ESP8266. Either way, there were no SPI changes lately.

jgromes, 2020-04-22:
@volpi3000 Are you using the RF69's RESET pin your setup? and if so, could you try the examples at 434 MHz without the reset pin connected (or set to RADIOLIB_NC in the Module constructor)? I noticed that if the hardware reset isn't done, the high power modules accept 434 MHz range just fine.

jgromes, 2020-05-06:
Closed due to inactivity.

I'm not sure what caused the issue with frequency configuration and reset pin - either way, high power operation seems to be working corretly now on RF69H.

closed 2020-05-06 by jgromes