jgromes/RadioLib #134: The following problem occurs in the initialization part of sx1278

lllwer, 2020-04-16:
[SX1278] Initializing ... failed, code -2

How to solve it ? Thank you

naggie, 2020-04-16:
-2 means chip not found. Check your wiring. 
https://github.com/jgromes/RadioLib/blob/master/src/TypeDef.h#L116

lllwer, 2020-04-16:
> -2 means chip not found. Check your wiring.
> https://github.com/jgromes/RadioLib/blob/master/src/TypeDef.h#L116

oh,I understood! Maybe the chip I welded was in poor contact. I hope we can continue to exchange and study next time. Thank you very much!

lllwer, 2020-04-16:
> means chip not found. Check your wiring.

I have another question. 
SX1278 has the following connections:
NSS pin:   10
DIO0 pin:  2
RESET pin: 9
DIO1 pin:  3
Why there is no  MoSI and MISO ?
What I use is Arduino Nano.

jgromes, 2020-04-16:
SPI pins on Arduino Nano (MISO, MOSI and SCK) are fixed, so you can't change them.

lllwer, 2020-04-16:
> SPI pins on Arduino Nano (MISO, MOSI and SCK) are fixed, so you can't change them.

Thank you very much for your reply!

lllwer, 2020-04-17:
> SPI pins on Arduino Nano (MISO, MOSI and SCK) are fixed, so you can't change them.

Hello author,
If I want the corresponding pin to hang, what should I fill in the brackets? For example, I just want to configure NSS pin and RESET pin, DIO0 pin and DIO1 pin are not configured.
// SX1278 has the following connections:
// NSS pin:   10
// DIO0 pin:  2
// RESET pin: 9
// DIO1 pin:  3
SX1278 lora = new Module(10, 2, 9, 3);

Thank you again for your team!

jgromes, 2020-04-17:
You can't use SX1278 without at least DIO0. Furthermore, DIO1 is required for methods like `receive()` and `scanChannel()`. If you need to mark some pin as not connected, then you can use RADIOLIB_NC macro, e.g.: `SX1278 lora = new Module(10, 2, 9, RADIOLIB_NC);` to mark DIO1 as unconnected, but some methods won't wirk without it.

lllwer, 2020-04-17:
> You can't use SX1278 without at least DIO0. Furthermore, DIO1 is required for methods like receive() and scanChannel(). If you need to mark some pin as not connected, then you can use RADIOLIB_NC [...]

What is the function of DIO0? Thanks for your reply.

jgromes, 2020-04-17:
From the appropraite [wiki page](https://github.com/jgromes/RadioLib/wiki/Basics#standalone-module):

>- **SPI** modules: `Module(cs, irq, rst, gpio)`. For example: `RF69 rf = new Module(10, 2, 3);`
    - CS: this is the pin that's used as SPI chip select.
    - IRQ: this is the interrupt pin, it's usually connected to external interrupt on Arduino. It's a fast way for the module to tell Arduino something has happened - for example, a packet was received. All SPI modules have at least one interrupt pin.
    - RST: most modules have a hardware reset pin. If the module doesn't have a reset or chip enable pin (e.g. CC1101), set `rst` to `RADIOLIB_NC`.
    - GPIO: this is the fourth possible pin. It's optional for most modules, but may be required for some. SX127x requires additional interrupt pin to perform CAD and blocking receive. SX126x also uses this pin as the BUSY indication pin. If not in use, this pin can be left unassigned, or set to `RADIOLIB_NC`.

DIO0 is the mandatory interrupt pin - it lets the Arudino know the SX127x module has finished a certain task, like transmitting a packet.

lllwer, 2020-04-17:
> DIO0 is the mandatory interrupt pin - it lets the Arudino know the SX127x module has finished a certain task, like transmitting a packet.

Your reply is very detailed. I see. Thank you !

lllwer, 2020-04-29:
Hello, author：           
 I want to send sensor data, but
int16_t startTransmit(String& str, uint8_t addr = 0);  
Can starttransmit send data of type int or float?

jgromes, 2020-04-29:
You can either format the number as a string, or you can memcpy the any datatype you want into a byte buffer, and then use the binary version `startTransmit(uint8_t* data, uint8_t len)`:

```c++
float f = 4.5;
uint8_t buff[4];
memcpy(buff, &f, sizeof(f));
radio.startTransmit(buff, 4);
```

This is more of an Arduino than RadioLib question - please use Arduino forum to ask general Arduino programming questions.

closed 2020-04-29 by jgromes