jgromes/RadioLib #135: [SX128x_BLE_Modem] Unable set Access Address, code -20.

mrninhvn, 2020-04-17:
[image]

 - MCU: Arduino Uno.
 - Wireless module type: SX1280 (LAMBDA80C)
 - Arduino IDE version: 1.8.10.
 - Library version: 3.5.0.

jgromes, 2020-04-17:
Looks like in an attempt to save on code duplicity, I used `setSyncWord` to set the access address. Obviously that method will fail for BLE modem - outsmarted by my own library :)

I also noticed there was a missing check for BLE modem in `startTransmit`, so that returned -20 as well.

Both should be fixed now - thanks for reporting!

closed 2020-04-17 by jgromes

mrninhvn, 2020-04-17:
@jgromes Hmm I encountered CRC error when receive. Can you help me?

jgromes, 2020-04-17:
It isn't entirely clear how CRC is processed in BLE mode, and there's also the PDU header to consider. SX128x doesn't create the header for you, that must be provided as the first two bytes of payload to transmit.

mrninhvn, 2020-04-17:
Hi @jgromes ,
I would be very grateful if you have examples to accomplish that.

jgromes, 2020-04-17:
Sorry, but full BLE support (on the protocol level) is not planned at the moment. SX128x only provides support for the physical layer.

mrninhvn, 2020-04-17:
Hi @jgromes,
My project using SX1280 only, can I add SX128x library into arduino sketch folder and use them without `RadioLib.h` & `Module.h` or not.
I don't understand the structure of your library so you can show me how.
Thank you a lot.

jgromes, 2020-04-17:
No, you can't do that, you always have to include RadioLib.h. SX128x.h is not a full library, only a driver for a specific module. The other radio drivers aren't used and will be optimized away during compilation.

mrninhvn, 2020-04-17:
Hi @jgromes ,
Can I add library to arduino sketch folder without adding to arduino library folder?

jgromes, 2020-04-17:
I'm not sure why would you want to do that - you'd probably have to modify the include paths for the compiler.