jgromes/RadioLib #136: Lora.receive

manueldindakis, 2020-04-19:
**IMPORTANT: Check the wiki**
I Have checked this sites. No solution found. (or maybe id did not know where to look)

**Describe the bug**
I have 2 the Duinotech LoRa Shields for UNO. I am using them with 2 MKR1500s.
I have successfully tested the SX1278_Transmit example. I can see via the serial monitor that the program is running.

I did changed the module creation line on both examples to suite the pins used on the MKR1500. 
SX1278 lora = new Module(7, 2, 3);

Now when I try running the receive example SX1278_receive it does not seem to run. It is not producing any output to the serial monitor. However, When I REM out the receive line
 int state = lora.receive(str); the program produces an output to the serial monitor but of course does not receive any LoRa data.

I have successfully tested the transmit and receive with UNO but no success with MKR1500

I have updated to the latest Arduino IDE and RadioLib library as at Apr 19, 2020

Any help will be appreciated

Regards

Manuel

jgromes, 2020-04-19:
You seem to have ommitted DIO1 from the constructor. SX127x's blocking `receive` method requires DIO1 as the source for timeout event, so I'd suggest connecting it to the MKR board and adding it to the constructor.

Also, I'd suggest checking the logic level of the shield is compatible with MKR 1500. If the shield was designed for Arduino Uno, then it most likely has 5 V logic. I'm not sure what's the logic level of MKR boards as I don't use them, but they might not be 5V-tolerant.

manueldindakis, 2020-04-20:
Yes, They are 3.3 Volts. 
I'll check the voltage levels

jgromes, 2020-05-05:
Closing due to inactivity - feel free to reopen if there's more information about the issue.

closed 2020-05-05 by jgromes