jgromes/RadioLib #137: ESB-support for built-in radio on nRF52

amsgo, 2020-04-19:
I am trying to find a way to use Enhanced Shockburst with the built-in radio on a nRF52-board (in my case it's the Adafruit Feather nRF52840) in order to communicate with the nRF24L01 and other nRF52s.

I have seen some implementations of this i.e. in the MySensors-Library, but this library is very use-case specific requiring you to use custom board definitions and so on. 

What i would love to see is a general-use ESB-library for the nRF52 chipsets so they can be used just like the nRF24 without having to physically add a nRF24-Chip to my setup (which seems a little pointless if I am already using an ESB-capable board). I have played around a little trying to create my own library but I realised that I just don't have the required low-level programming skills needed for that.

jgromes, 2020-04-19:
I'm not entirely sure how to go about achieving this - at the moment, nRF52 is only suported as a platform to run the library on. In general, RadioLib assumes that any radio module is a separate system connected to a microcontroller.

I'm not opposed to the idea in general, unfortunately I have no experience with nRF52. I'd encourage you (or anyone experienced with nRF52) to dig into it, and see if we can add that into RadioLib. The library structure should make it possible to add nRF52 as a driver in src/modules and only use it when nRF52 platform is used in BuildOpt.h

amsgo, 2020-04-20:
I found a very basic implementation of this here: https://github.com/HiveTracker/arduino_nrf_radio
I played around with it a little bit and it seems to be working as expected. It's very basic however missing some core features of ESB like the ACK. In general the setup seems to be somewhat similar to the nRF24 since it's the same protocol.

jgromes, 2021-07-27:
Upon examining nRF52 datasheet, it doesn't seem that interesting. It's very similar to nRF24, which is probably the "weakest" module in RadioLib (when it comes to hacking, ham radio modes etc.)

Unfortunately there are other priorities, so this won't be worked on.

closed 2021-07-27 by jgromes