jgromes/RadioLib #138: Proposed changes to facilitate integration

Timvrakas, 2020-04-28:
Hello! I've been integrating this into a project for school, and I had a few suggestions:

* Most Arduino configurations have `SPIClass` and `Uart`, but some don't have specific instances globally defined. You use `SPI` and `Serial1` in several places in the library, which causes compilation (header checks) to fail. My workaround was to add [these extern lines](https://github.com/stanford-ssi/RadioLib/blob/bddfc1b9d7e7d95e5138cc8283ea07e56c2bc603/src/BuildOpt.h#L30). If the user tries to use one of the Module constructors that uses these objects, and they don't exist, the linker will throw an error. If they do exist, there should not be any side effects. I think it's wise to include these lines if static or global objects are referenced, because having 0 `SPIClass` instances is a valid configuration.
* A silent `RADIOLIB_GODMODE` that doesn't put out that warning. I had to turn that on the first day and I'd love to be able to turn it off, as funny as it is
* **SX126x.h: ** I feel many of the object fields should be protected, rather than private. It's nearly impossible to override functionality without entering Godmode. I ran up against this while modifying `SX126x` for a FreeRTOS threaded environment.

Thanks so much for the great library!
timv

jgromes, 2020-04-28:
Which compiler (and platform) are you using? `RADIOLIB_HARDWARE_SERIAL_PORT` should only be used if SoftwareSerial is unavailable, which is only a case for some platforms - all of which do have `Serial1`. It is true that `SPI` instance is always used on all platforms, so it would be safer to have some macro in BuildOpt to specify it, but I haven't seen it being an issue on any of the supported platforms. On a side note, you can always create your own "platform" in BuildOpt.h. 

The warning is in there by design - godmode should never be used in production and only sparingly in development.

Most of the private fields in `SX126x` are cached parameters (since SX126x can only be configured using SPI commands with specific parameter sequences), so it should be OK to make tehm protected.

Timvrakas, 2020-04-28:
I'm building with PlatformIO under the the [Adafruit core](https://github.com/adafruit/ArduinoCore-samd). I'm using a custom variant for a board that does not have generic `SPI`, `SPI2` ports. Instead I instantiate and pass `SPIClass` objects to RadioLib. My setup looks like [this](https://github.com/stanford-ssi/SpaceSalmon/blob/5cf5e5e940cdfc0d63e5faee2500f97b9a0543d0/src/guppy/RadioTask.hpp#L53].) I'm aware this is not a common configuration, but given [this option](https://github.com/adafruit/ArduinoCore-samd/blob/d60ec0fae387c8c982937e6e1b22c2569e24ae5b/variants/feather_m4/variant.h#L137) exists, I'm inclined to believe that 0 (static) SPI interfaces is a valid number. 
The `RADIOLIB_HARDWARE_SERIAL_PORT` option is nice, but only if you have a Serial port to point it to, there's no way to leave it undefined. It also feels a bit odd to set up a Serial port to avoid build errors when using a module that uses SPI. 
I think generally It's best for libraries to make as few assumptions as possible. You `#include` the Arduino header files, It seems natural to `extern`ally declare the symbols you're expecting to be present.

jgromes, 2020-04-28:
Strictly speaking, Adafruit SAMD core is not a supported platform ;)

You should be able to pass a null pointer to the Module constructor if you don't want to use UART interface - it won't touch it unless you explicitly try to call some UART method, like AT commands.

I'd prefer to avoid adding `extern` for object that may or may not exist - linker errors can be painful to debug, I'd prefer compilation to throw an undefined reference error.

jgromes, 2020-05-12:
To address what was pointed out in this issue:

1. All platform configurations now have RADIOLIB_DEFAULT_SPI parameter. This coupled with the option to create a custom platform with RADIOLIB_CUSTOM_PLATFORM should provide users the ability to fully customize platform definition, even for non-standard platforms.
2. Godmode warning will be left as-is. It's annoying by design, godmode shouldn't be used for workarounds.
3. @Timvrakas I'm not sure which private fields in SX126x class were giving you trouble, so I'm leaving them all private for now. Feel free to open PR to make specific fields protected.

closed 2020-05-12 by jgromes