jgromes/RadioLib #139: AFSK Modulation in Direct Mode

mrpatrick1991, 2020-04-28:
Currently the AFSK / AX25 examples in the SX127x folder transmit AX.25 frames in FSK mode, which can't be decoded using the usual ham radio equipment. The main application is probably APRS, which is mostly useful because of the large deployed infrastructure. This infrastructure expects sound modulated signals over an FM signal. 

I suggest that work from e.g. Trackuino et. al. (https://github.com/trackuino/trackuino) which makes the AFSK signal using PWM could be transmitted in the SX127x direct mode. I have run a test of this successfully on a high altitude balloon and the signal was received by the standard APRS stations. 

Alternatively, I think that it might be more clear to change the language used in the examples so as not to suggest that they will work with AFSK 1200 baud equipment. To my knowledge, the examples use FSK 1200 only. 

Excellent library by the way. I appreciate all the work that has gone into it.

jgromes, 2020-04-29:
This has been pointed out before in #112 - I'll look into it.

What constitutes "the usual ham radio equipment"? I'm not a ham radio operator, and I only use SDR, so I can demodulate the signal as CW or USB, hence obtaining the audio from just binary FSK.

mrpatrick1991, 2020-04-29:
Thanks for the reply. The "usual equipment" consists of FM analog radios for voice attached to old bell 202 style modems, and is in widespread use through the APRS system.

jgromes, 2020-04-29:
I guess the affected protocols/digimodes will be AX.25, RTTY, Morse, SSTV and Hellschreiber. Would be able to test them out, once AFSK support is in (bearing in mind I have no idea when will I get to it)? It would help me quite a lot.

mrpatrick1991, 2020-04-29:
Yes, absolutely happy to test for you. I have the equipment to do this.

I think it would make the library of even greater use to us HAMs

jgromes, 2020-04-30:
That's great, thanks!

I was trying to get RTTY with AFSK to work (to test out the new audio interface), And I did manage to do it just using Arduino's `tone()` function. The issue is, it only outputs square wave, so the resulting audio has all the extra spectral lines - I'm assuming from Fourier decomposition. See the  screenshot, space at 400 Hz, 183 Hz shift.
[image]

I was wondering whether is this going to be an issue? I tried the AFSK RTTY with Fldigi (after demodulating as plain FM with SDR), and it seems to decode fine, but what about other modes, like AX.25? It looks like the trackduino project used sine-modulated PWM, which could improve things, though I'm not too keen on having to implement it, as it wouldn't be very portable between different platforms.

jgromes, 2020-04-30:
So it looks like it won't be an issue after all - I successfully decoded AFSK RTTY, Morse code and Hellschreiber using Fldigi, SSTV using MMSSTV and AX.25 using soundmodem. Will commit once I clean it all up.

jgromes, 2020-04-30:
So it looks like it builds OK for all platforms that support Arduino tone() :)

@mrpatrick1991 all the new AFSK examples are in the example directory for the appropriate protocol. I tested it with Arduino Pro Mini and SX1278, so if you can get a similar setup going on, it would be great to try it out using actual ham equipment.

mrpatrick1991, 2020-04-30:
Wow! That was fast - congratulations! I'll test these modes as soon as possible and report results here. I might even fly a sounding balloon and see what kind of range we can get.

jgromes, 2020-05-14:
I believe this can be closed for the time being (and also as per the contributing doc). Since I was able to decode everything using fairly common programs, I think it should work for HAM equipment as well.

Feel free to reopen if there are issues.

closed 2020-05-14 by jgromes

tomasbrincil, 2020-11-30:
Hello there, any recommendation for using only AFSK modulator/demodulator without need of initialization any of supported radion via PhysicalLayer?

jgromes, 2020-11-30:
I'm not sure why would you need to do that - `AFSKClient` is just a wrapper around Arduino `tone()` function with checks for automated starting/disabling direct mode, so if you were to run it without a radio, you'd basically only have the `tone()` function.