jgromes/RadioLib #140: APRS using FSK G3RUH

DF1AMB, 2020-05-01:
Hi,

I am trying to implement APRS using radiolib and SX1278/1276 chips.

The problem I am stuck at the moment is that I cannot receive Data from my Yaesu VX-8 running at 9600 baud speed in FSK mode. It also does not work the other way arraound.

So my question is what mode do I need to set radiolib to? Is it standard mode or whitening?

Thanks for your support

Martin (DF1AMB)

jgromes, 2020-05-01:
As long as you use the AX25Client class, it should take care of all the configuration. I was able to decode AX.25 frames for both FSK and AFSK using soundmodem, so I'm confident the configuration is correct.

If you're not using the AX25Client class, you will most likely have to make sure the FSK packet is as plain as possible - no CRC or addresses, encoding set to NRZ using `setEncoding(0)`, etc.

> Yaesu VX-8 running at 9600 baud speed in FSK mode

The default AX.25 example uses 1200 baud, so that's something to check. When you say FSK, do you mean AFSK, or binary FSK (BFSK)?

DF1AMB, 2020-05-01:
The VX-8 is able to do both AFSK at 1200 baud and FSK at 9600 baud.

Since the bandwidth you see when you look at the waterfall of a SDR is smaller in 9600 baud than in 1200 baud I would guess it is BFSK. But I am not sure.

The first test I tried is running the SX1278 in FSK mode.

`modem.beginFSK(rQrg, BAUDRATE_FAST, MAX_DEVIATION, 12.5, TX_POWER, 100, 16, false);`

I did not use `setEncoding()` at all.
After initiating the FSK modem I did

`modem.setDio0Action(packetReceived);
modem.startReceive();`

From what I thought the SX1278 should now be able to receive a sent Frame. For the first test I did not care about the protocoll (AX.25) used. I was hoping to see the raw packet. So I got my Handheld (VX-8) and sent some Frames. I could see the transmission on the waterfall of my SDR, but the chip did not react as expected.

jgromes, 2020-05-01:
My guess would some incompatibility between the packets. Since you mentioned you have an SDR, you could capturing the FSK packets from VX-8, to see the exact bytes that are sent, including things like preamble. I use Universal Radio Hacker for that, it's simple enough even for me (unlike GNURadio).

On SX127x, the encoding should be set to NRZ by default. Assuming all the modulation parameters are correct, the next thing that comes to mind is packet length mode (fixed or variable), or sync word.

DF1AMB, 2020-05-01:
Yes I have a SDR, but I have just been playing around with it an received audio signals. So thanks for advising me to Universal Radio Hacker.

Currently I am trying to receive the string I am sending with the SX1278. But nothing so far.

I used the protocol sniffer and set it to FSK. Maybe I have to set other parameters that I don't know yet. Or since it is only a cheap RTL SDR maybe the Hardware is useless in this case.

jgromes, 2020-05-01:
I'm using an RTL-SDR too, it's enough to do the job.

To be honest, I haven't used the protocol sniffer yet. To do basic analysis, select record signal in the file menu, then select RTL-SDR as the device, set the frequency, disable DC correction and click start.

DF1AMB, 2020-05-02:
So I did what you suggested. I took a record of the SX1278 and then put it into the interpretation window. Well what should I say, it does not decode correctly. But that is not the point.

The point is that I also took a record of the VX-8. When I had a look to the waveform it looked totally different. So I took another record of the VX-8 in 1200 bps AFSK mode and the waveform looked similar. This made me wonder if 9600 bps packet radio may also be AFSK and not as expected FSK. I found the answer on [AMSAT](https://www.amsat.org/amsat/articles/g3ruh/109.html) homepage. But I had to read that article more than once so get to the point, that with a bandwidth of up to 8 kHz a normal handheld radio is perfectly capable of doing 9600 bps AFSK (that in fact only need 4800 Hz of bandwidth)

In the actual case I will proceed with my ARPS implementation. In Fact I just will need to use two SX127x chips for my tests.

So thanks for your help and for the great lib you are providing. If I have a commitable state of my work I will provide it as an extension for radiolib

jgromes, 2020-05-02:
I'm pretty you'll be able to find a couple fo tutorials on URH - I highly recommend to give it a shot later, it's incredibly useful when developing wireless stuff.

And thank you for the time you're taking to work on APRS :) If you're planning on submitting a pull request, please take a look at the [coding style guidelines](https://github.com/jgromes/RadioLib/blob/master/CONTRIBUTING.md#code-style-guidelines). I will close this issue now, since there's already an issue for APRS support (#112).

closed 2020-05-02 by jgromes