jgromes/RadioLib #145: Advice for duplex interrupts?

wbrickner, 2020-05-09:
I want to use the async (interrupt based) communication methods. All the examples for my board (`SX1278`) use an interrupt on DIO0 and either receive or transmit, but not both. 

I tried doing it myself, by assuming that the interrupt is an RX event, trying to check the new contents of the buffer along with the status code, then falling back to the interrupt being a TX completion event if valid data was not contained.

This feels wrong and hacky and doesn't actually work (well, at least in the exact way I've written it).

What I'd really _like_ to do is register handlers for interrupts on separate pins for RX and TX events, or at the very least be able to receive these events in a clearly correct way.

I've examined the documentation and can't find any public methods for configuring things this way.

I'm an embedded radio noob, do you have any advice? 
Is this impossible?

Note: I have access to SDR hardware and two LoRa & Arduino composites which are functioning properly, it's just a matter of software now.

Thank you very much!

jgromes, 2020-05-10:
Unfortunately, on SX1278 only DIO0 can be used for TxDone or RxDone events - this is a limitation of the hardware. I can see at least two workarounds:

1. Set up ISR for Rx event - this is used as the "default" DIO0 interrupt handler, because you can't predict when a packet is going to arrive.
2. When you want to transmit, clear the Rx ISR from DIO0 using `SX127x::clearDio0Action()`.
3. Set up new ISR for Tx event by calling `SX127x::setDio0Action(txEvent)`.
4. When that triggers (or there's a timeout), clear the Tx ISR and set it back to Rx ISR.

Alternatively, you could use some flag to indicate whether you're currently transmitting, and check that flag in the DIO0 ISR. Set the flag prior to starting your transmission, then clear it afterwards and you won't have to keep changing the ISRs.

gozu42, 2020-05-14:
i ran into a variant of this issue over at https://github.com/Code-and-Response/ClusterDuck-Protocol/issues/50 
(if you prefer this as a separate issue, just holler. it seems close enough to me and my suggestion seems to solve this too)

last summary from "over there":

after reading the semtech sx127x specs some more, the IRQs have some interesting additional potential.
for example, there is an irq that fires when a lora header was received, so we could avoid transmitting while we are actualy receiving a packet right now. == cheapish CSMA

the best way to me seems capsulating all irq handling inside radiolib.
this could be exposed as isReceiving (received header, but not complete yet) and isAvailable (complete) methods.

another interface would be to register separate rx/tx/whatever ISRs with radiolib, and call them as needed after irq-state decoding.
(as in, both an "interrupt oriented" and a "polling" interface work)

i can write the code for this, but am open to suggestion on what to call the methods or implementation hints.
my testing would be limited to sx127x boards in lora mode.

jgromes, 2020-05-14:
I'm not sure how I feel about this - it might be useful to have access to all the IRQs, however, I'm not sure how to integrate it with the current API.

On modules like SX127x, RadioLib assumes there's always one dedicated IRQ pin, and optionally one GPIO pin (which can be used as DIO1 on SX127x, or something else like BUSY pin on SX126x/8x). The issue with DIO configuration on SX127x and similar modules (like RF69/SX1231 or CC1101) is that events can't be mapped to arbitrary DIO pin - below is a table for LoRa and FSK packet modes:

[image]
[image]

I don't think it would be a good idea to extend `Module` constructor for all 6 DIO pins SX127x has. It seems like it would make more sense to have bunch of new methods that configure the correct DIO pin for the requested action, e.g. `SX127x::setValidHeaderAction(pin, onValidHeader);` would map ValidHeader event to pin number and call onValidHeader as the ISR. However, there must be a way to specify which DIO should be used, as some of the events may be mapped to multiple DIO pins.

Also, I don't think your original issue with "ghost packets" will be solved by this - the fundamental problem is that only DIO0 can be used for RxDone and TxDone events. I haven't studied your code, but from the description it looks like the external interrupt fires on TxDone event, which obviously still triggers the setFlag function intended for RxDone. It doesn't seem like should be the job of the driver to determine whether that's valid or not.

gozu42, 2020-05-14:
my example code is very close to your "Receive Interrupt" example, with a bit of "Transmit" (no (expected) interrupt) mixed in.

your guess regarding the "ghost packets" is pretty much spot on, and the obvious way to avoid the problem is to clear the enableInterrupt flag before transmitting. (which is "same idea, different direction" from your "no swapping ISRs" suggestion above)
but while figuring this out i had found RegIrqFlags, but was not aware of constraints on which DIO pins these can be mapped to...

and with these constraints (to basicly just two flags via interrupt on dio0) it really doesnt make much sense to put it into radiolib. 

i am going to play around a bit with reading SX127X_REG_IRQ_FLAGS and SX127X_REG_MODEM_STAT and see if exposing them through a getIRQFlags or getModemStat might be useful even without mapping any more actual DIO pins. 

thanks for your response!

gozu42, 2020-05-14:
i ran [this sketch](https://github.com/jgromes/RadioLib/files/4630726/GhostWatcher.txt) for a bit, which basicly uses GODMODE to dump the two registers mentioned in the last comment as bits to serial every loop(), and reads any received packet (and ignores the content).

postprocessing [some resulting logfile](https://github.com/jgromes/RadioLib/files/4630885/GhostWatcherLog.txt) (while receiving some packets) with 
```
cut -d" " -f5- | uniq -c
```
ignores the loopcount (lc:) and millis (m:) since they are both just timing details,
and SX127X_REG_IRQ_FLAGS_MASK (im:) since it is always just 0 (we are not ignoring any IRQs),
and then counts how many times each "combination of values" has been seen.

an "ideal" receive looks like this:
```
    274 if:0 ms:1100100
      2 if:0 ms:1100101
      3 if:0 ms:1100111
     62 if:10000 ms:1101111
      1 if:1010000 ms:1100100 pl:112
      4 if:0 ms:1100100
```
* for the first 274 loops we see we are in receive mode, but not seeing any signal (ms & 7 == 4).
* for 2 loops we see a signal (ms & 7 == 5).
* for 3 loops we see a synchronized signal (ms & 7 == 7).
* for 62 loops we have a valid header and are still receiving (ms & 15 == 15 and if & 16). and we could get this as an IRQ on the right pin, if we were listening.
* now the RX IRQ fires, which is signaled as (if & 64), all the ms&11 bits are back to 0, and we read pl:112 bytes of received packet.
* readData called clearIRQFlags so we are back in initial state and waiting for the next packet (if==0 and ms & 7 == 4).

sadly the if&64 part (== rxdone irq flag) is not reliable at all, there are plenty of cases like this too:
```
    695 if:0 ms:1100100
      3 if:0 ms:1100101
      3 if:0 ms:1100111
     63 if:10000 ms:1101111
      1 if:10000 ms:1101111 pl:115
      9 if:0 ms:1100100
```
ms behaves exactly the same as in the first example, but if&64 is never set even though we did get the RX IRQ. 

so this is not a reliable way to distinguish "is this an RX or TX IRQ" (which we have other ways to figure out anyways as you pointed out), but the modem status seems to be a reliable indicator of receiver status, which works for "would transmitting now be a collision" detection.

so, would you prefer ...
* a getIRQFlags() and getModemStat() that basicly just does the SPI read and returns the byte?
 * or a isReceivingSignal() that basicly returns (_mod->SPIgetRegValue(SX127X_REG_MODEM_STAT) & 1) as a boolean status?
* and a bunch of new defines for the modem stat bits?

either works for me, i mainly just want a solution that works without having to set GODMODE.
if you prefer this as a PR, just give me the direction.

jgromes, 2020-05-15:
getIRQFlags() and getModemStat() seems like the more general solution here.

Side note: SX127X_REG_IRQ_FLAGS register is the source for IRQ events - I don't think it's possible for RxDone IRQ to be unreliable. If that was the case, I would expect there would be routine packet loss. I think this might be due to the way you're logging the variables - Serial.prints and String conversions take a long time, so it is possible the RxDone interrupt flag is cleared by the hardware before you have the chance to read and print it.

wbrickner, 2020-05-15:
@jgromes, I'm a little in over my head reading these replies, but (I think) it sounds like there's a way to tell the difference between these two events in principle, but there are more abstract design concerns that would make a library change sticky. 

Is there a way I can modify my own copy of RadioLib to achieve my goal of differentiating an RxDone and TxDone interrupt on the DIO0 pin? I was going to abstract away RadioLib anyway into a tidy fire-and-forget / event-driven messaging layer, so it's kindof fine if it ends up messy, but I'd need a little bit of guidance.

I also have a more specific question: is packet transmission failure / unpredictable transmission latency a real issue? Like, what is the real utility of the TxDone event?  

My current method is to allocate a 256-byte RX buffer, set the first byte to a special value that is otherwise disallowed to lead in a packet (like `'\0'`), then when I get an interrupt of unknown meaning I check the first byte for the special value, which then tells me if it's an RxDone (which must have modified the buffer) or TxDone (no modification to the buffer, illegal value remains).  Will this cause any problems (assuming I control transmissions and can guarantee packets do not lead w/ illegal value, etc)?

Can you go into more detail about how I might use the methods you mentioned?

Also, if you don't mind, can you explain what `DIOx Mapping` means in the data sheet table you referenced? Is this some sort of flag accessible from software? If so, I could just use that to differentiate interrupts on the same pin.

Thanks for your replies everyone!

wbrickner, 2020-05-15:
I'd also like to respond to your earlier comment.  The issue I see is that if you're using a flag or some other global state (maybe stored in `RadioLib`) about being in RX state or TX state, you may actually be in both and totally lose information about a new message or a transmission completion.

E.g. you've just sent a message under normal operation, you globally store that you're in transmit state, then execution is interrupted because you've gotten a new message from gateway / peer, and you assume it's a TX completion event, but it actually wasn't, lo and behold the device has dropped the ball.

I'm not exactly sure how truly async things are this low in the stack, so I could be wrong about this and your solution would be A-OK.

jgromes, 2020-05-15:
@wbrickner 

> it sounds like there's a way to tell the difference between these two events

In theory yes, you can read SX127X_REG_IRQ_FLAGS to see which events are currently active. However, this requires an SPI transaction, and I'm not entirely sure this will work from inside interrupt service routine. Furhtermore, checking a simple flag is a lot faster.

> Is there a way I can modify my own copy of RadioLib to achieve my goal of differentiating an RxDone and TxDone interrupt on the DIO0 pin

Possibly so, but again, I don't think this should be handled by the library, because both TxDone and RxDone events trigger DIO0 interrupt. Distinguishing between them on the library level would require abandoning the current approach to DIO interface (the `setDio0Action` and `clearDio0Action` and their various counterparts on other modules), and changing it to some sort of "event" interface. e.g. `onRxDone`, `onTxDone` etc. - I'm not sure how portable would this be on the other modules. It also raises the issue of priority - if the user sets functions for both  `onRxDone` and `onTxDone`, the library would have to decide (somehow) which one should be called first.

>  what is the real utility of the TxDone event?

The utility is in that it tells you when packet transmission is finished on the transmitter side - that's it. I haven't seen may tx fails (it usually means something is wrong with the hardware). Nevertheless, it is an option, so it must be accounted for (using a timeout, for example).

> My current method is to allocate a 256-byte RX buffer

I don't think that's a good idea. As long as your system is running in a real world scenario, you can't guarantee someone else won't be transmitting in your channel (by chance or by intention). You also have to check for CRC errors, so you can never really guarantee special values won't appear. Values reserved only by convention are never a good solution in my experience.

> Can you go into more detail about how I might use the methods you mentioned?

You could have an ISR which on DIO0 action sets a flag. Check that flag in your main loop and if it is set, get the contents of SX127X_REG_IRQ_FLAGS. If bit 3 is set, the event that caused the interrupt is TxDone. If bit 6 is set, it was RxDone. Note that there's really not much difference between this and storing some global Rx/Tx state - in this case that state is just not stored on your MCU. You also have to make sure you retrieve the register value before it gets modified.

> can you explain what DIOx Mapping means

SX127x only has 6 DIO pins, but a lot more events that can cause interrupts. Which event causes change on any given physical DIO pin is controlled by DIO mapping registers. The interrupt sources are also accessible in SX127X_REG_IRQ_FLAGS register (in LoRa mode; there's two of them in FSK mode). Checking the value of this register is at the core of what @gozu42 proposed.

> The issue I see is that if you're using a flag or some other global state (maybe stored in RadioLib) about being in RX state or TX state, you may actually be in both

I don't think that's the case: SX127x is half duplex, so the hardware can't receive while it's transmitting. In your example, it is true that there is a small window between the time when you stop transmitting and set the flag that denotes you're in receive mode. However, this time window will not be large enough for you to receive entire packet. Even high speed FSK packets will take up to a few ms to transmit - far longer than the few MCU instructions required to set a flag. If you're using LoRa, the transmission time jumps up to few hunderd ms at least (or to several seconds for high spreading factor), so it becomes a non-issue there. Finally, you can make sure this never happens by using CAD on your gateways/peers and mandating minimum time interval between on-air packets.

jgromes, 2020-05-28:
I added both methods @gozu42 proposed (untested but should work unless I'm dumb). @wbrickner this should also help with your issue, as long as you call `SX127x::getIRQFlags()` prior to calling `readData()`, because that method clears all active flags.

closed 2020-05-28 by jgromes

gozu42, 2020-05-29:
both functions added in f068a15 tested and working.
thank you!