jgromes/RadioLib #146: STM32F103C8 - int state = lora.begin (); - program stops

EmilM256, 2020-05-17:
I used the example attached to the Radiolib library. The problem occurs regardless of which core I choose. There is no error during compilation, but the program stops on the line: **int state = lora.begin ();**

My next command is **digitalWrite (PC13, LOW);** and it's never done. I disconnected my radio module. Same behavior.

I made another test: Arduino Pro Mini (AVR) + RadioLib + sample sketch. I did not connect the RFM95 radio. The program returned error -2 no radio. It should be. Test passed.

This is the sketch from the examples a bit modified by me. The last thing I see via serial port is: **[SX1278] Initializing GO !!**

```
// include the library
#include <RadioLib.h>

// SX1278 has the following connections:
// NSS pin:   10
// DIO0 pin:  2
// RESET pin: 9
// DIO1 pin:  3
SX1278 lora = new Module(PA0, PA1, PA2, PA3);

void setup() {
  Serial.begin(9600);
  pinMode(PC13, OUTPUT);
  // initialize SX1278 with default settings
  Serial.print(F("[SX1278] Initializing ... "));
  digitalWrite(PC13, LOW);
  Serial.println(F("[SX1278] Initializing  3 "));
  delay(1000); digitalWrite(PC13, HIGH);
  Serial.println(F("[SX1278] Initializing  2 "));
  delay(1000); digitalWrite(PC13, LOW);
  Serial.println(F("[SX1278] Initializing  1 "));
  delay(1000); digitalWrite(PC13, HIGH);
  Serial.println(F("[SX1278] Initializing GO!! "));
  
  int state = lora.begin();
  digitalWrite(PC13, LOW);
  if (state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }
  digitalWrite(PC13, LOW);
  // set the function that will be called
  // when new packet is received
  lora.setDio0Action(setFlag);

  // start listening for LoRa packets
  Serial.print(F("[SX1278] Starting to listen ... "));
  state = lora.startReceive();
  if (state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }
}

// flag to indicate that a packet was received
volatile bool receivedFlag = false;

// disable interrupt when it's not needed
volatile bool enableInterrupt = true;

void setFlag(void) {
  // check if the interrupt is enabled
  if(!enableInterrupt) {
    return;
  }

  // we got a packet, set the flag
  receivedFlag = true;
}

void loop() {
  // check if the flag is set
  if(receivedFlag) {
    // disable the interrupt service routine while
    // processing the data
    enableInterrupt = false;

    // reset flag
    receivedFlag = false;

    // you can read received data as an Arduino String
    String str;
    int state = lora.readData(str);

    if (state == ERR_NONE) {
      // packet was successfully received
      Serial.println(F("[SX1278] Received packet!"));

      // print data of the packet
      Serial.print(F("[SX1278] Data:\t\t"));
      Serial.println(str);

      // print RSSI (Received Signal Strength Indicator)
      Serial.print(F("[SX1278] RSSI:\t\t"));
      Serial.print(lora.getRSSI());
      Serial.println(F(" dBm"));

      // print SNR (Signal-to-Noise Ratio)
      Serial.print(F("[SX1278] SNR:\t\t"));
      Serial.print(lora.getSNR());
      Serial.println(F(" dB"));

      // print frequency error
      Serial.print(F("[SX1278] Frequency error:\t"));
      Serial.print(lora.getFrequencyError());
      Serial.println(F(" Hz"));

    } else if (state == ERR_CRC_MISMATCH) {
      // packet was received, but is malformed
      Serial.println(F("[SX1278] CRC error!"));

    } else {
      // some other error occurred
      Serial.print(F("[SX1278] Failed, code "));
      Serial.println(state);

    }
    lora.startReceive();
    enableInterrupt = true;

  }
}
```

**Additional info (please complete):**
 - MCU: STM32F103C8 blue pill
 - RFM95
 - Arduino IDE version 1.8.12 
 - Roger's core 2020.5.16, Official Core 1.9.0 
 - Library version 3.6.1

jgromes, 2020-05-17:
Could you post the output with debug mode enabled? Uncomment these two lines in BuildOpt.h, then post the output from Serial port:

https://github.com/jgromes/RadioLib/blob/a969517e0445a1c9d52a6bff73fdffa42a2c71ef/src/BuildOpt.h#L213-L214

EmilM256, 2020-05-17:
This is what I got after entering debug mode:
```
[SX1278] Initializing  3 
[SX1278] Initializing  2 
[SX1278] Initializing  1 
[SX1278] Initializing GO!! 
R	42	0	
SX127x not found! (1 of 10 tries) SX127X_REG_VERSION == 0x00, expected 0x0012
R	42	0	
SX127x not found! (2 of 10 tries) SX127X_REG_VERSION == 0x00, expected 0x0012
R	42	0	
SX127x not found! (3 of 10 tries) SX127X_REG_VERSION == 0x00, expected 0x0012
R	42	0	
SX127x not found! (4 of 10 tries) SX127X_REG_VERSION == 0x00, expected 0x0012
R	42	0	
SX127x not found! (5 of 10 tries) SX127X_REG_VERSION == 0x00, expected 0x0012
R	42	0	
SX127x not found! (6 of 10 tries) SX127X_REG_VERSION == 0x00, expected 0x0012
R	42	0	
SX127x not found! (7 of 10 tries) SX127X_REG_VERSION == 0x00, expected 0x0012
R	42	0	
SX127x not found! (8 of 10 tries) SX127X_REG_VERSION == 0x00, expected 0x0012
R	42	0	
SX127x not found! (9 of 10 tries) SX127X_REG_VERSION == 0x00, expected 0x0012
R	42	0	
SX127x not found! (10 of 10 tries) SX127X_REG_VERSION == 0x00, expected 0x0012
No SX127x found!
```
I do not have a Radio connected to the board,
but why I do not have information about the error: **failed, code -2** ??

jgromes, 2020-05-17:
I was able to replicate the issue on Nucleo L452RE-P. The problem was in `Module::term()` method, which on platforms without SoftwareSerial library (such as STM32) attempted to terminate Serial interface even on SPI-based modules.

Seems to be fixed now, I'll release this as patch 3.6.2. Thanks for reporting!

closed 2020-05-17 by jgromes

EmilM256, 2020-05-17:
I have progress!
Progeam started running when SX127X_REG_VERSION takes the expected value.

The second thing. I have an RFM95 radio module with an RFM96 chip. The module operates at a frequency of 868 MHz.

If I choose RFM95 **lora = new Module .....** get :
SX127x not found! (1 of 10 tries) SX127X_REG_VERSION == 0x12, expected 0x0011

if I choose RFM96 I have error -12 incorrect frequency (I set frequency 868.00)
I changed RadioLib\src\modules\RFM9xRFM95.h  **#define RFM95_CHIP_VERSION** to **0x12** progeam works. I don't know if I did the right thing.

I will check communication tomorrow.
Now I get this (I don't have DOI1 connected yet):
```
[] Initializing GO!! 
R	42	12	
Found SX127x!
R	1	F	
W	1	9	
R	1	9	
R	1	9	
R	1	9	
W	1	8	
R	1	8	
R	1	8	
W	1	88	
R	1	88	
R	1	88	
W	1	89	
R	1	89	
R	1	89	
R	1	89	
W	1	89	
R	1	89	
R	39	12	
W	39	12	
R	39	12	
R	1	89	
W	1	89	
R	1	89	
R	B	2B	
W	B	2B	
R	B	2B	
R	1	89	
W	1	89	
R	1	89	
R	1	89	
R	20	0	
W	20	0	
R	20	0	
R	21	8	
W	21	8	
R	21	8	
R	24	0	
W	24	0	
R	24	0	
R	1	89	
W	1	89	
R	1	89	
R	6	6C	
W	6	D9	
R	6	D9	
R	7	80	
W	7	0	
R	7	0	
R	8	0	
W	8	0	
R	8	0	
R	1	89	
R	1	89	
W	1	89	
R	1	89	
R	1D	72	
W	1D	72	
R	1D	72	
Symbol length: 0.01 ms
R	26	4	
W	26	4	
R	26	4	
R	1	89	
R	1	89	
W	1	89	
R	1	89	
R	1D	72	
W	1D	72	
R	1D	72	
R	1E	70	
W	1E	94	
R	1E	94	
R	31	C3	
W	31	C3	
R	31	C3	
R	37	A	
W	37	A	
R	37	A	
Symbol length: 4.10 ms
R	26	4	
W	26	4	
R	26	4	
R	1	89	
R	1	89	
W	1	89	
R	1	89	
R	1D	72	
W	1D	76	
R	1D	76	
R	1	89	
W	1	89	
R	1	89	
R	9	4F	
W	9	CF	
R	9	CF	
R	9	CF	
W	9	FF	
R	9	FF	
R	4D	84	
W	4D	84	
R	4D	84	
R	1	89	
R	1	89	
W	1	89	
R	1	89	
R	26	4	
W	26	4	
R	26	4	
R	1	89	
W	1	89	
R	1	89	
R	1	89	
R	40	0	
W	40	0	
R	40	0	
R	1	89	
W	12	FF	
R	F	0	
W	F	0	
R	F	0	
R	D	0	
W	D	0	
R	D	0	
R	1	89	
W	1	8D	
R	1	8D	
```

EmilM256, 2020-05-17:
> Seems to be fixed now, I'll release this as patch 3.6.2. Thanks for reporting!

SUPER :)

jgromes, 2020-05-17:
The issue only occured when attempting to terminate the module interface when the module was not found, so it doesn't have any effect when the module is found.

Next: 
> I have an RFM95 radio module with an RFM96 chip. The module operates at a frequency of 868 MHz.

Sooo ... you have an RFM96 radio then. RadioLib attempts to verify this by reading the register SX127X_REG_VERSION. Yours is set to 0x12, which means it's indeed RFM96. This chip is limited to 410 - 525 MHz range according to the datasheet.