jgromes/RadioLib #149: How to send a byte array over AFSK without using other modes. I need to send a custom packet made in a byte array over AFSK

rnbokade, 2020-05-21:


jgromes, 2020-05-21:
This question doesn't make much sense. AFSK is an analog modulation - it works in terms of analog frequencies (tones), not binary information (bits).

You could designate some tone as logic 1 and some other as logic 0. For example, Bell 202 AFSK modem used 1200 Hz for logic 1 (mark) and 2200 Hz for logic 0 (space), which was later used in some digital modes and protocols (like AX.25 or SSTV).

To send the "packet", you would then iterate over all bits of the message, and call `afsk.tone(1200);` if the bit is 1 or `afsk.tone(2200);` if the bit is 0. Of course, you're not limited to one bit per tone - you could just as well have four different tones and assign them to the four possible combinations of two bits, if you can later decode that.

rnbokade, 2020-05-21:
So you mean to say this is not like a Bell202 AFSK implementation? It just give ability to generate tones which can later be used to make own implementations...

jgromes, 2020-05-21:
That's correct - I don't see the reason to asume AFSK should always transmit Bell 202 tones.

rnbokade, 2020-05-21:
Ok got it.

closed 2020-05-21 by jgromes