jgromes/RadioLib #151: Does RadioLib support RFM95

szbrooks, 2020-06-01:
Hi, sorry this is sort of a dumb question, but in the readme section it states that RFM9X is support but in the GitHub page it doesn't mention that it's supported.

Could I use RadioLib to connect two MRKZeros with LoRa on RFM9X modules?

Thanks,

jgromes, 2020-06-01:
RadioLib supports all RFM9x devices, including RFM95 - the source file is here: https://github.com/jgromes/RadioLib/blob/master/src/modules/RFM9x/RFM95.cpp

> but in the GitHub page it doesn't mention that it's supported.
Which page do you mean? 

> Could I use RadioLib to connect two MRKZeros with LoRa on RFM9X modules?
I don't see why not, MKR Zero should be a supported platform.

szbrooks, 2020-06-01:
I didn't see it in : https://jgromes.github.io/RadioLib/index.html
however I could be easily misunderstanding something.

Thanks for the quick response and for such a comprehensive library!

closed 2020-06-01 by szbrooks