jgromes/RadioLib #153: NOTE Generic Solution regarding RF Solution's  LAMBDA range not working!

the-drums, 2020-06-18:
After a lot of testing/debugging, finally got the RF Solutions LAMBDA62C modules to work!

The previous remark regarding using LDO instead of DC-DC appears to **not** be the issue., DC-DC circuit is fine and works properly.

**The solution:**
DONT USE command: "SetDIO3AsTCXOCtrl"
DONT USE command: "SetDIO2AsRfSwitchCtrl" when you have a REV2 Module!

As it appears from a diagram I've retrieved somewere:
Using command: "datasheet: --> 13.3.6 "SetDIO3AsTCXOCtrl" to control TXCO reference voltage causes the malfunction: BUSY_LINE stays high @ the upfollowing command datasheet --> 13.1.12 "Calibrate Function", indicating something is wrong.
This simply is caused by DIO3 not been connected internally to anything!.
After disabling this command,  Calibrate function works  && RX / TX works on all modules!

Sidenote: 
RF Solution has 2 Revisions (one is not yet available):
REV2 (Datasheet: LAMBDA62-2) shows no diagram Important! --> DIO2 is NOT connected to RF_switch
REV3 (Datasheet: LAMBDA62-3) shows a diagram with DIO2 connected to RF_switch.

This concludes that:
REV2 --> Command: datasheet --> 13.3.5 "SetDIO2AsRfSwitchCtrl"  **should NOT be used!**
REV3 -->Command: datasheet --> 13.3.5 "SetDIO2AsRfSwitchCtrl"  can be used!

Hopefully with this I save some people a lot of time figuring it out.

Ps1. As per requested I have tried the RADIOLIB for these modules.
First thing I noticed, that if you put a "pinMode(....) in Void setup(){   with the library enabled on a SAMD21 (arduino zero(M0) / M0 feather etc.) ,  the pinmode functions do NOT work any longer and all script test crashes @ boot.  There is a bug in the override function of "pinMode" of the library (saw that this is to protect the pins used by the module, digitalread/write are working though), why this is bugging i'm not sure yet, but it makes this library not usefull for SAMD21 processers who also want to use the other pins for other functionalities which is a shame unfortunatly. 
If you could please fix this that would be great!.
Will do some futher testing on the radiolib library.

Ps2. There seems to be no support for TX_ENABLE / RX_ENABLE pins which a lot of HF modules use? 
--> RX High, TX Low == Receiving, and vice versa (both LOW @ initialization  & must be set before commands:  "setRX" or "setTX"  acc. datasheets.
Maybe interesting to integrate, should be doable?

Hopefully with this i've provided some usefull information.
And thank you for replying to my previous message to swiftly! Highly appreciated.

Kind regards, 
Paul

jgromes, 2020-06-18:
Thanks for taking the time to check this out - I don't have the LAMBDA62C modules myself, so I can't really test them.

Couple of comments:

1. As mentioned in the examples and docs, you have to set TCXO reference voltage to 0 in begin when your module is using just a plain crystal. There's no way for the library to tell whether you're using TCXO or XTAL. LAMBDA62C diagram shows XTAL, therefore trying to use TCXO will cause the radio to hang at calibration because it's waiting for TCXO to start (as you correctly deduced).

2. Using DIO2 for RF switch control also depends on the module hardware - LAMBDA62C has RF switch controlled by the TX_ENABLE/RX_ENABLE pins. Similarly to TCXO, there's no way for the library to tell if that's the case, it only knows it's connected to some SX126x module. DIO2 is used for RF switch control by default, since that's the case for the SX126x modules made by Dorji (because it seems those are the most common). Still, if your module is using the external TX_ENABLE/RX_ENABLE pins, having DIO2 RF switch control enabled shouldn't matter as long as nothing is connected to DIO2.

Ad PS1: How do you know it's the pinMode overrides that cause this beahvior? They're just a simple wrappers around the pinMode function provided by your Arduino core, so I don't think there's much that can go wrong there. Which Arduino core were you using at the time? There's a couple of them that support SAMD21 (https://github.com/arduino/ArduinoCore-samd or https://github.com/adafruit/ArduinoCore-samd).

Ad PS2: It is true that some modules have an external RF switch controlled by two RX_EN/TX_EN pins, However, this is by no means standard, and also it's a component that's external to the radio chip itself. RadioLib is primarily a driver, so I'm not entirely sure how to fit that nicely into the current API.

naggie, 2020-06-18:
> The previous remark regarding using LDO instead of DC-DC appears to not be the issue., DC-DC circuit is fine and works properly.

No, there is literally no DC-DC converter on the board. You must use it in the LDO mode, else the module will draw excess current -- but still appear to work.

> DONT USE command: "SetDIO3AsTCXOCtrl"

There is no TCXO.

> DONT USE command: "SetDIO2AsRfSwitchCtrl" when you have a REV2 Module!

Yes, the DIO2 is not connected right. You may also need to remove R3 on that board, though that might be what the rev2 fixes.

the-drums, 2020-06-18:
Thank you all for the time to make the replies.
This was my first time working with these modules haha sorry if some things appear to be unclear. Previously used RFM modules.

Will honour your recommedations and disable DC-DC for excessive current usage.

Indeed I see as you stated that the RX/TX enable pins are indeed not the standard way of doing things. Maybe you can have a look at the other library from loratracker which does support this of simply skip it leave that up to you.

I've checked the Dorji modules but it appears they dont have CE certification. Only red & FCC. Maybe they will still do that.

Regarding the pinmode.
After i cleared the override pinmode void immediatly the tests weren't crashing any longer. Using just the plain examples no other libraries. Therefor i reccon the fault should be there somewere. For example setting pin13 as Led output. pinMode(13, OUTPUT) in setup void crashes the samd21.

 I'll need to check which library for the samd21 i've used. Will come back at that later on.

naggie, 2020-06-18:
No problem

> Indeed I see as you stated that the RX/TX enable pins are indeed not the standard way of doing things. Maybe you can have a look at the other library from loratracker which does support this of simply skip it leave that up to you.

Yes, you just need to drive the RX enable and TX enable pins respectively before receiving and transmitting. According to RF solutions, the reason why 2 separate signals are used is so that the module can go into a very low power state when both circuits are off.

jgromes, 2020-06-18:
@the-drums I think you may have misunderstood - the RF switch is not a part of the SX1262 chip, it's an external component and strictly speaking, it's out of scope for this library.

That being said, I can see it could be somewhat annoying having to do this prior to any Tx/Rx, so I added automatic 2-pin RF switch control. To enable it, call `setRfSwitchPins(rxEn, txEn);` somewhere in setup after calling `begin()`. Bear in mind this is untested.

the-drums, 2020-06-18:
Thank you for clarifying.
I've just tested your latest library, and used LDO && TXCO voltage "0"

**good** - First of all RX w. interrupt is working!.
**good** - (Strangely enough no more problems with pinmode also!).
**bad** - TX example states it's transmitting but it's not receiving with the receive example?
(from monitor: "[SX1262] Sending another packet ... transmission finished!") though.
Have received a Frequency analyzer today, so can have a look.
**good** - TX example from other lib combined with RX example from Radiolib is indeed 
ending/receiving correctly, 
So think there is still something small to do on the TX side? (or on my end? checked the settings all the same for TX & RX)

Thank you in advance!.

jgromes, 2020-06-19:
The RF switching is internally implemented like this, I assume that to set the RF switch to transmit you have to set Tx enable high and Rx enable low.

https://github.com/jgromes/RadioLib/blob/d710a85985168b163267250d1cd5d814dc5968a9/src/Module.cpp#L326-L340

There was a bug in setting RF switch to Tx mode in `startReceive`, which should have the opposite effect on your setup (Tx working, Rx not), so I suggest you check the pins you provided to `setRfSwitchPins(rxEn, txEn)` are correct and do not conflict with anything.

the-drums, 2020-06-20:
Hi JGromes,

Thank you for your time and assistance so far!

I see that the TX/RX pins are working as desired.
Also see the module gives transmit confirmation on DIO1.
SERIAL OUTPUT (+added add. txt a after blank test):
"
14:09:12.337 -> [SX1262] Sending another packet ... 
14:09:12.337 -> setRfSwitch RX to:LOW
14:09:12.337 -> setRfSwitch TX to:HIGH
14:09:12.337 -> 
14:09:12.337 -> transmission finished!
"
However nothing actually transmits on the assigned frequency (checked with RTL-SDR).
Pins checked with oscilloscope.
Freq should be correctly as the RX example works well with the exact same setting.

Bit in the dark here, think there is something else wrong.
Have no time unfortunatly to analyze your entire script.
Think it must be somewere in the config send to the module (calibration / image etc.) or a timing issue of some kind (delay switching external booster on?).
Don't know how they do it @ loratracker, that one I can confirm it transmits with RTL-SDR, so must be a deviation somewere.
Identical settings used for Radiolib.
"
settings:
  int state = lora.begin(869.525, 250.0, 7, 5, 0x12, 14, 60, 8, 0, true);
"

Ps. the TX HIGH RX LOW pin pin assesments keeps repeating, but not shutting down after transmit. (LOW, LOW), should be easy to add, ofcourse if people tend to transmit continuously this is not desired, rewriting the pins to the same value isn't an issue though, or a pin-state check could be added, but that's optimalization.

Hopefully you can assist with the transmit isue with radiolib.

Kind regards, 
Paul

jgromes, 2020-06-20:
Could you try to run the transmit example with debug mode enabled? Maybe that will give some more information.

RF switch should get switched off now when calling methods such as `standby()` or `sleep()` and also at the end of blocking tx/rx methods `transmit()` and `receive()` - however, in some cases (like interrupt-driven transmit), it can't be turned off automatically, since interrupt service routine is in the Arduino sketch in that case.

the-drums, 2020-06-20:
Hi JGromes,

Have gotten the latest version of the lib again to be sure.
I've turned on verbose first, then also checked debug.

With this sketch it's transmitting on the correct frequency opposed to the interrupt sketch.
The transmitted signal is rather very low (verbose mode active)!~ compared to the other lib and has missing spots througout the frequencies?.
With verbose turned of and DEBUG turned on, there is no signal at all @ that frequencie...

OUTPUTS:
output Serial (DEBUG):  (NO TRANSMISSION @ FREQ)
17:23:20.593 -> [SX1262] Initializing ... Symbol length: 0.51 ms
17:23:20.627 -> Symbol length: 0.51 ms
17:23:20.627 -> Symbol length: 0.51 ms
17:23:20.627 -> success!
17:23:20.627 -> [SX1262] Transmitting packet ... Timeout in 30912 us
17:23:20.627 -> failed, code -707
17:23:21.628 -> [SX1262] Transmitting packet ... Timeout in 30912 us
17:23:21.628 -> success!
17:23:21.628 -> [SX1262] Datarate:	4613.17 bps
17:23:22.634 -> [SX1262] Transmitting packet ... Timeout in 30912 us
17:23:22.669 -> success!

output Serial (VERBOSE): (TRANSMISSION @ FREQ!, BUT VERY POOR)
"
17:17:33.804 -> [SX1262] Transmitting packet ... CMD	80	
17:17:33.804 -> DATW	0	A2	
17:17:33.804 -> 
17:17:33.804 -> CMD	11	
17:17:33.804 -> DATR	0	A2	0	1	
17:17:33.804 -> 
17:17:33.804 -> CMD	11	
17:17:33.838 -> DATR	0	A2	0	1	
17:17:33.838 -> 
17:17:33.838 -> CMD	11	
17:17:33.838 -> DATR	0	A2	0	1	
17:17:33.838 -> 
17:17:33.838 -> CMD	1D	7	36	
17:17:33.838 -> DATR	0	A2	0	9	
17:17:33.838 -> 
17:17:33.838 -> CMD	D	7	36	
17:17:33.838 -> DATW	9	A2	
17:17:33.838 -> 
17:17:33.838 -> CMD	8C	
17:17:33.838 -> DATW	0	A2	8	A2	0	A2	C	A2	1	A2	0	A2	
17:17:33.838 -> 
17:17:33.838 -> CMD	8	
17:17:33.838 -> DATW	2	A2	1	A2	0	A2	1	A2	0	A2	0	A2	0	A2	0	A2	
17:17:33.838 -> 
17:17:33.838 -> CMD	8F	
17:17:33.838 -> DATW	0	A2	0	A2	
17:17:33.838 -> 
17:17:33.838 -> CMD	E	0	
17:17:33.838 -> DATW	48	A2	65	A2	6C	A2	6C	A2	6F	A2	20	A2	57	A2	6F	A2	72	A2	6C	A2	64	A2	21	A2	
17:17:33.838 -> 
17:17:33.838 -> CMD	2	
17:17:33.838 -> DATW	3	A2	FF	A2	
17:17:33.838 -> 
17:17:33.838 -> CMD	1D	8	89	
17:17:33.838 -> DATR	0	A2	0	4	
17:17:33.838 -> 
17:17:33.838 -> CMD	11	
17:17:33.838 -> DATR	0	A2	0	1	
17:17:33.873 -> 
17:17:33.873 -> CMD	D	8	89	
17:17:33.873 -> DATW	4	A2	
17:17:33.873 -> 
17:17:33.873 -> CMD	83	
17:17:33.873 -> DATW	0	A2	0	A2	0	A2	
17:17:33.873 -> 
17:17:33.873 -> CMD	2	
17:17:33.873 -> DATW	3	AC	FF	AC	
17:17:33.873 -> 
17:17:33.873 -> CMD	80	
17:17:33.873 -> DATW	0	AC	
17:17:33.873 -> 
17:17:33.873 -> success!
17:17:33.873 -> [SX1262] Datarate:	5188.35 bps
"

jgromes, 2020-06-20:
This behavior seems somewhat familiar now. Could you go to SX126x.cpp and change line 1661 to the following:
`#if defined(ARDUINO_ARCH_STM32) || defined(SAMD_SERIES)` (or just remove the #if #endif lines entirely and leave only the delay(1) there).

If it doesn't help, please post the full verbose transcription, including module initialization.

the-drums, 2020-06-20:
Still very intermittent transmits with low power output.
Stated below the output per Serial monitor for a couple of transfers.

18:00:38.409 -> [SX1262] Initializing ... CMD	80	
18:00:38.409 -> DATW	0	FF	
18:00:38.409 -> 
18:00:38.441 -> CMD	80	
18:00:38.441 -> DATW	0	A2	
18:00:38.441 -> 
18:00:38.441 -> CMD	80	
18:00:38.441 -> DATW	0	A2	
18:00:38.441 -> 
18:00:38.441 -> CMD	8F	
18:00:38.441 -> DATW	0	A2	0	A2	
18:00:38.441 -> 
18:00:38.441 -> CMD	8A	
18:00:38.441 -> DATW	1	A2	
18:00:38.441 -> 
18:00:38.441 -> CMD	93	
18:00:38.441 -> DATW	20	A2	
18:00:38.441 -> 
18:00:38.441 -> CMD	88	
18:00:38.441 -> DATW	3	A2	14	A2	A	A2	0	A2	0	A2	0	A2	0	A2	
18:00:38.441 -> 
18:00:38.441 -> CMD	2	
18:00:38.441 -> DATW	3	A2	FF	A2	
18:00:38.441 -> 
18:00:38.441 -> CMD	8	
18:00:38.441 -> DATW	0	A2	0	A2	0	A2	0	A2	0	A2	0	A2	0	A2	0	A2	
18:00:38.441 -> 
18:00:38.441 -> CMD	89	
18:00:38.441 -> DATW	7F	A2	
18:00:38.441 -> 
18:00:38.475 -> CMD	11	
18:00:38.475 -> DATR	0	A2	0	1	
18:00:38.475 -> 
18:00:38.475 -> CMD	8B	
18:00:38.475 -> DATW	7	A2	4	A2	3	A2	0	A2	
18:00:38.475 -> 
18:00:38.475 -> CMD	11	
18:00:38.475 -> DATR	0	A2	0	1	
18:00:38.475 -> 
18:00:38.475 -> CMD	8B	
18:00:38.475 -> DATW	7	A2	5	A2	3	A2	0	A2	
18:00:38.475 -> 
18:00:38.475 -> CMD	11	
18:00:38.475 -> DATR	0	A2	0	1	
18:00:38.475 -> 
18:00:38.475 -> CMD	8B	
18:00:38.475 -> DATW	7	A2	5	A2	1	A2	0	A2	
18:00:38.475 -> 
18:00:38.475 -> CMD	11	
18:00:38.475 -> DATR	0	A2	0	1	
18:00:38.475 -> 
18:00:38.475 -> CMD	D	7	40	
18:00:38.475 -> DATW	14	A2	24	A2	
18:00:38.475 -> 
18:00:38.475 -> CMD	D	8	E7	
18:00:38.475 -> DATW	18	A2	
18:00:38.475 -> 
18:00:38.475 -> CMD	11	
18:00:38.475 -> DATR	0	A2	0	1	
18:00:38.475 -> 
18:00:38.475 -> CMD	1D	7	36	
18:00:38.475 -> DATR	0	A2	0	D	
18:00:38.475 -> 
18:00:38.475 -> CMD	D	7	36	
18:00:38.475 -> DATW	9	A2	
18:00:38.475 -> 
18:00:38.475 -> CMD	8C	
[... 644 lines cut ...]
18:00:50.268 -> DATW	48	A2	65	A2	6C	A2	6C	A2	6F	A2	20	A2	57	A2	6F	A2	72	A2	6C	A2	64	A2	21	A2	
18:00:50.268 -> 
18:00:50.268 -> CMD	2	
18:00:50.268 -> DATW	3	A2	FF	A2	
18:00:50.268 -> 
18:00:50.268 -> CMD	1D	8	89	
18:00:50.303 -> DATR	0	A2	0	4	
18:00:50.303 -> 
18:00:50.303 -> CMD	11	
18:00:50.303 -> DATR	0	A2	0	1	
18:00:50.303 -> 
18:00:50.303 -> CMD	D	8	89	
18:00:50.303 -> DATW	4	A2	
18:00:50.303 -> 
18:00:50.303 -> CMD	83	
18:00:50.303 -> DATW	0	A2	0	A2	0	A2	
18:00:50.303 -> 
18:00:50.303 -> CMD	2	
18:00:50.303 -> DATW	3	AC	FF	AC	
18:00:50.303 -> 
18:00:50.303 -> CMD	80	
18:00:50.303 -> DATW	0	AC	
18:00:50.303 -> 
18:00:50.303 -> success!
18:00:50.303 -> [SX1262] Datarate:	5131.22 bps

the-drums, 2020-06-20:
SX1262 lora = new Module(NSS, DIO1, NRESET, SPI, SPISettings(12000000, MSBFIRST, SPI_MODE0));

  int state = lora.begin(869.525, 250.0, 7, 5, 0x12, 14, 60, 8, 0, true);
    lora.setRfSwitchPins(RX_EN, TX_EN);

all else is acc. example (except pin numbers ofcourse).

the-drums, 2020-06-20:
#ifdef TESTBOARD
#define NSS             5
#define NRESET        6
#define RFBUSY       10 
#define DIO1          11
#define DIO2          18
#define DIO3          -1
#define RX_EN         17
#define TX_EN         19
#endif

the-drums, 2020-06-20:
[image]

the-drums, 2020-06-20:
That shows the difference for clarification, picture says more than a thousand words :).

jgromes, 2020-06-20:
Checking the debug output all seems OK, the default RadioLib configuration should match that of LoRaTracker. Could you post the sketch you're using with that library?

the-drums, 2020-06-20:
`#define Program_Version "V1.1"

#include <SPI.h>                                               //the lora device is SPI based so load the SPI library                                         
#include <SX126XLT.h>                                          //include the appropriate library  
#include "Settings.h"                                          //include the setiings file, frequencies, LoRa settings etc   

SX126XLT LT;                                                   //create a library class instance called LT

uint8_t TXPacketL;
uint32_t TXPacketCount, startmS, endmS;

uint8_t buff[] = "Hello World 1234567890";

void loop()
{
  Serial.print(TXpower);                                       //print the transmit power defined
  Serial.print(F("dBm "));
  Serial.print(F("Packet> "));
  Serial.flush();

  TXPacketL = sizeof(buff);                                    //set TXPacketL to length of array
  buff[TXPacketL - 1] = '*';                                   //replace null character at buffer end so its visible on reciver

  LT.printASCIIPacket(buff, TXPacketL);                        //print the buffer (the sent packet) as ASCII

  digitalWrite(LED1, HIGH);
  startmS =  millis();                                         //start transmit timer
  if (LT.transmit(buff, TXPacketL, 10000, TXpower, WAIT_TX))   //will return packet length sent if OK, otherwise 0 if transmit error
  {
    endmS = millis();                                          //packet sent, note end time
    TXPacketCount++;
    packet_is_OK();
  }
  else
  {
    packet_is_Error();                                 //transmit packet returned 0, there was an error
  }

  digitalWrite(LED1, LOW);
  Serial.println();
  delay(packet_delay);                                 //have a delay between packets
}

void packet_is_OK()
{
  //if here packet has been sent OK
  uint16_t localCRC;

  Serial.print(F("  BytesSent,"));
  Serial.print(TXPacketL);                             //print transmitted packet length
  localCRC = LT.CRCCCITT(buff, TXPacketL, 0xFFFF);
  Serial.print(F("  CRC,"));
  Serial.print(localCRC, HEX);                         //print CRC of transmitted packet
  Serial.print(F("  TransmitTime,"));
  Serial.print(endmS - startmS);                       //print transmit time of packet
  Serial.print(F("mS"));
  Serial.print(F("  PacketsSent,"));
  Serial.print(TXPacketCount);                         //print total of packets sent OK
}

void packet_is_Error()
{
  //if here there was an error transmitting packet
  uint16_t IRQStatus;
  IRQStatus = LT.readIrqStatus();                      //read the the interrupt register
  Serial.print(F(" SendError,"));
  Serial.print(F("Length,"));
  Serial.print(TXPacketL);                             //print transmitted packet length
  Serial.print(F(",IRQreg,"));
  Serial.print(IRQStatus, HEX);                        //print IRQ status
[... 71 lines cut ...]
  LT.setRfFrequency(Frequency, Offset);
  LT.setModulationParams(SpreadingFactor, Bandwidth, CodeRate, Optimisation);
  LT.setBufferBaseAddress(0, 0);
  //order is        preamble, header type,              packet length,   CRC,         IQ
  LT.setPacketParams(8, LORA_PACKET_VARIABLE_LENGTH,  255,             LORA_CRC_ON, LORA_IQ_NORMAL);

  
  LT.setDioIrqParams(IRQ_RADIO_ALL, (IRQ_TX_DONE + IRQ_RX_TX_TIMEOUT), 0, 0);   //set for IRQ on TX done and timeout on DIO1
  LT.setHighSensitivity();  //set for maximum gain
  LT.setSyncWord(LORA_MAC_PRIVATE_SYNCWORD);
  //***************************************************************************************************

  Serial.println();
  LT.printModemSettings();                               //reads and prints the configured LoRa settings, useful check
  Serial.println();
  LT.printOperatingSettings();                           //reads and prints the configured operating settings, useful check
  Serial.println();
  Serial.println();
  LT.printRegisters(0x00, 0x4F);                         //print contents of device registers, normally 0x00 to 0x4F
  Serial.println();
  Serial.println();

  Serial.print(F("Transmitter ready"));
  Serial.println();
}`

the-drums, 2020-06-20:
hmm github handles an other type of code syntag.
Just made a account specifically for this.

Ps. this is the "Setting.h"
`/*******************************************************************************************************
  lora Programs for Arduino - Copyright of the author Stuart Robinson - 02/03/20

  This program is supplied as is, it is up to the user of the program to decide if the program is
  suitable for the intended purpose and free from errors.
*******************************************************************************************************/

//*******  Setup hardware pin definitions here ! ***************

#define TESTBOARD

#ifdef TESTBOARD
#define NSS 5                                  //select pin on LoRa device
#define NRESET 6                                //reset pin on LoRa device
#define RFBUSY 10                               //busy pin on LoRa device 
#define DIO1 11                                 //DIO1 pin on LoRa device, used for RX and TX done 
#define DIO2 18//A4                                 //DIO2 pin on LoRa device, normally not used so set to -1 
#define DIO3 -1                                 //DIO3 pin on LoRa device, normally not used so set to -1
#define RX_EN 17//A3                                //pin for RX enable, used on some SX126X devices, set to -1 if not used
#define TX_EN 19//A5                                //pin for TX enable, used on some SX126X devices, set to -1 if not used 
#define SW -1                                   //SW pin on some Dorji LoRa devices, used to power antenna switch, set to -1 if not used
#define BUZZER -1                                //pin for buzzer, set to -1 if not used
#endif
#define LED1 13                                  //on board LED, high for on
#define LORA_DEVICE DEVICE_SX1262               //we need to define the device we are usingR 4  

//*******  Setup LoRa Parameters Here ! ***************

const uint32_t Frequency = 869525000;           //frequency of transmissions in hertz
const uint32_t Offset = 0;                      //offset frequency for calibration purposes
const uint8_t Bandwidth = LORA_BW_250;          //LoRa bandwidth
const uint8_t SpreadingFactor = LORA_SF7;       //LoRa spreading factor
const uint8_t CodeRate = LORA_CR_4_5;           //LoRa coding rate
const uint8_t Optimisation = LDRO_AUTO;         //low data rate optimisation setting, normally set to auto
const int8_t TXpower = 14;                       //LoRa transmit power in dBm

#define RXBUFFER_SIZE 32                        //RX buffer size
const uint16_t packet_delay = 1000;             //mS delay between packets
`

jgromes, 2020-06-20:
I can only see the following differences:

1. You're using 12 MHz SPI speed with RadioLib, but 8 MHz with LoRaTracker. Could you try lower speed?
2. DIO2 is set to RF switch control with RadioLib. Shouldn't matter, since from the LAMBDA62C datasheet  it doesn't looks like it's connected to anything and it isn't used internally by the library. Still, could you try disabling it with `setDio2AsRfSwitch(false)`?

the-drums, 2020-06-20:
Hi JGromes,

Have checked, both SPI settings, no difference.
setDio2.... had already been done previously.

I noticed the com ports for my PC were exceeding limits last test.
Redone the test with the 1ms delay ureka!
Checked again without the 1ms, no signal.
1ms added, everything works!

So that was the remaining fault!.
(Also i see start/end of signals have some difficulties, i think it's to RX / TX switch turning on/off times.) but all packages are received correctly @ RSSI -40dbm. (not the best modules in the world and crystal (but also due to TX settings) not temperaturecontrolled. (maybe some script Onboard temp sens. (sx1262) vs. crystal timing would be great though!.

jgromes, 2020-06-20:
Good to hear that - so it was the issue with the following lines then? I guess I will have to investigate a bit more in-depth than just adding a 1 ms delay. I'll add SAMD core to the #if condition so that delay is included both for STM32 and SAMD.

https://github.com/jgromes/RadioLib/blob/06c254bc6edd2e25cd04c1b37bfbec37ad645ad2/src/modules/SX126x/SX126x.cpp#L1661-L1663

the-drums, 2020-06-20:
Yes, if that problem is solved all is fine, 

If I come accross something which can cause the issue I will notfiy you. 
Will go through a part of the script now knowing that I can use it 👍.

Indeed it's better to remove the delay,  i also always try to prevent all delays/blocking codes in a script (except for testing).
It should not be needed, run all kinds of SPI on 16MHz without any delay required (even on samd, which is known having some hardware-specific issues).
Crystal should also not play an issue due to SCK keeps everything in sync.

Ps. tx/rx delay i mentioned earlier is only needed if switching on/off at every transmission. (don't know the required min switch time acc. specs), but can add that some time. for maximum power savings.

jgromes, 2020-06-20:
I extended the check for SAMD cores - closing for now, will let you know if at some point I figure out what's the root cause.

Thank you for testing it!

closed 2020-06-20 by jgromes