jgromes/RadioLib #155: FSK - Low communication speed

Khyros12, 2020-06-23:
Hello,
I want to sent 1kB files as fast and as often as possible. I choose FSK with 300kbps bitrate but as far as i tried i received around 48kbps. I use transmitting and receiving on interrupts. I changed bitrate, size of sent array and other configurations but sending time ussually looks same for all of them - around 40ms per transmit.
I'm ussing 2x ESP32, SX1278 (433MHz), Arduino 1.8.3

`#include <RadioLib.h>

// SX1278 has the following connections:
// NSS pin:  5  - 10
// DIO0 pin: 2  - 2
// RESET pin:14 - 9
// DIO1 pin: 26 - 3
SX1278 fsk = new Module(5, 2, 14, 26);

int i=0;
int oneperhundred=0;

volatile bool dataReceivedFlag = false;
volatile bool enableInterrupt = true;

void IRQ_flag(void) {
  // check if the interrupt is enabled
  if(!enableInterrupt) {
   return;
  }// we got a packet, set the flag
  dataReceivedFlag = true;
}

void setup() {
  Serial.begin(9600);
  // carrier frequency:           434.0 MHz
  // bit rate:                    48.0 kbps
  // frequency deviation:         50.0 kHz
  // Rx bandwidth:                125.0 kHz
  // output power:                13 dBm
  // current limit:               100 mA
  // data shaping:                Gaussian, BT = 0.3
  // sync word:                   0x2D  0x01
  // OOK modulation:              false
  int state = fsk.beginFSK(434.0, 300.0, 50.0, 250.0, 17, 100, 16, false);
  if (state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }

 /* 
  state = fsk.setFrequency(433.5);
  state = fsk.setBitRate(100.0);
  state = fsk.setFrequencyDeviation(50.0);
  state = fsk.setRxBandwidth(250.0);
  state = fsk.setOutputPower(17.0);
  state = fsk.setCurrentLimit(100);*/
  state = fsk.setDataShaping(1.0);
  
  uint8_t syncWord[] = {0x81, 0x46, 0xCD, 0xAE, 
                        0x98, 0x47, 0x54, 0xCF};                    
  state = fsk.setSyncWord(syncWord, 8);
  
  if (state != ERR_NONE) {
    Serial.print(F("Unable to set configuration, code "));
    Serial.println(state);
    while (true);
  }

  fsk.setDio0Action(IRQ_flag);
  Serial.print(F("Listening for incoming transmission ... "));

  state = fsk.startReceive();
    if (state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }
}

void loop() {
  

  // you can receive data as an Arduino String
  // NOTE: receive() is a blocking method!
  //       See example ReceiveInterrupt for details
  //       on non-blocking reception method.
  // NOTE: for spreading factor 6, the packet length
  //       must be known in advance, and provided to
  //       receive() method!
  String str;
  //int state = fsk.receive(str);

  // you can also receive data as byte array

  if(dataReceivedFlag){

    enableInterrupt = false;
    dataReceivedFlag = false;

    size_t len = 56;
    byte byteArr[len];
    int state = fsk.readData(byteArr, len);
    i=0;
    str="Data:\t";
    while(i < len){
      str = str + byteArr[i];
      i++;
    }

  if (state == ERR_NONE) {
    // packet was successfully received
   /* Serial.print(F("Start!\t"));
    Serial.println(millis());*/
    // print data of the packet
    oneperhundred++;

    if(oneperhundred%100 == 0){
      Serial.print(str);
      Serial.print(F("\ttime:\t"));
      Serial.print(millis());
      Serial.print(F("\tRRSI:\t"));
      Serial.println(fsk.getRSSI());
    }

  } else if (state == ERR_RX_TIMEOUT) {
    // timeout occurred while waiting for a packet
    Serial.println(F("timeout!"));

  } else if (state == ERR_CRC_MISMATCH) {
    // packet was received, but is malformed
    Serial.println(F("CRC error!"));

  }

  fsk.startReceive();
  enableInterrupt= true;
   
  }
}`

jgromes, 2020-06-23:
Since you haven't posted your transmitter code, it's hard to tell what could be causing the delays between successive transmissions. I guess large chunk of that is going to be the SPI transactions between SX1278 and ESP32. The SPI clock frequency is set to 2 MHz by default, so you could try to speed it up (SX1278 should be able to handle 10 MHz SPI).

Next, the 300 kbps is the nominal bit rate - that's not the actual "throughput" of the channel, i.e. "how many useful payload bits can you send per second". Apart from the payload, which is the useful information, the FSK packet also contains other fields which are going to slow you down. By default, it's the preamble (16 bits), sync word (64 bits in your case), length byte (8 bits), then your actual payload (56x8 bits) and finally, CRC (16 bits). This means that to send your 448 bit message, you also send 104 extra bits. If data rate is critical, you can try e.g. shorter sync word, or using fixed-length mode (if the packet length is known in advance).

Finally, I would question whether SX1278 is a good choice of transceiver for this use-case. At this speed, the range is going to be rather poor.

Khyros12, 2020-06-24:
Thank You for fast reply. I need to send low quality photo as fast as possible in 1 km range. I thought previously about ESP32 wifi with directional antenna but LoRa seemed interesting for my to learn something new. Do You have any experience on what range will it works with that speed ? Transmitter code in bottom
`#include <RadioLib.h>

// SX1278 has the following connections:
// NSS pin:  5  - 10
// DIO0 pin: 2  - 2
// RESET pin:14 - 9
// DIO1 pin: 26 - 3
SX1278 fsk = new Module(5, 2, 14, 26);

volatile bool dataTransmitedFlag = false;
volatile bool enableInterrupt = true;
int transmissionState = ERR_NONE;

/*byte transmitArray[] = {0xA1, 0xA2, 0xA3, 0xA4,
                          0x21, 0x22, 0x23, 0x24,
                          0x31, 0x32, 0x33, 0x34,
                          0x41, 0x42, 0x43, 0x44,
                          0x51, 0x52, 0x53, 0x54,
                          0x61, 0x62, 0x63, 0x64,
                          0x71, 0x72, 0x73, 0x74,
                          0x81, 0x82, 0x83, 0x84};*/
byte transmitArray[] = {  0xA1, 0xA2, 0xA3, 0xA4, 0xA5, 0xA6, 0xA7, 0xA8,
                          0x21, 0x22, 0x23, 0x24, 0x25, 0x26, 0x27, 0x28,
                          0x31, 0x32, 0x33, 0x34, 0x35, 0x36, 0x37, 0x38,
                          0x41, 0x42, 0x43, 0x44, 0x45, 0x46, 0x47, 0x48,
                          0x51, 0x52, 0x53, 0x54, 0x55, 0x56, 0x57, 0x58,
                          0x61, 0x62, 0x63, 0x64, 0x65, 0x66, 0x67, 0x68,
                          0x71, 0x72, 0x73, 0x74, 0x75, 0x76, 0x77, 0x78};
size_t arraySize = sizeof(transmitArray);

void IRQ_flag(void) {
  
  if(!enableInterrupt) {
   return;
  }
  dataTransmitedFlag = true;
}

void setup() {
  Serial.begin(1000000);
  // carrier frequency:           434.0 MHz
  // bit rate:                    48.0 kbps
  // frequency deviation:         50.0 kHz
  // Rx bandwidth:                125.0 kHz
  // output power:                13 dBm
  // current limit:               100 mA
  // data shaping:                Gaussian, BT = 0.3
  // sync word:                   0x2D  0x01
  // OOK modulation:              false
  int state = fsk.beginFSK(434.0, 300.0, 100.0, 250.0, 17, 100, 16, false);
  if (state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }

/*
  state = fsk.setFrequency(433.5);
  state = fsk.setBitRate(100.0);
  state = fsk.setFrequencyDeviation(50.0);
  state = fsk.setRxBandwidth(250.0);
  state = fsk.setOutputPower(17.0);
  state = fsk.setCurrentLimit(100);*/
  state = fsk.setDataShaping(1.0);
  
  uint8_t syncWord[] = {0x81, 0x46, 0xCD, 0xAE, 
                        0x98, 0x47, 0x54, 0xCF};
  state = fsk.setSyncWord(syncWord, 8);
  if (state != ERR_NONE) {
    Serial.print(F("Unable to set configuration, code "));
    Serial.println(state);
    while (true);
  }

  fsk.setDio0Action(IRQ_flag);
  Serial.print(F("Sending packet ... "));
  transmissionState= fsk.startTransmit(transmitArray, arraySize);
  
  
}
void loop() {
  
  if(dataTransmitedFlag){

    enableInterrupt = false;
    dataTransmitedFlag = false;

    if (transmissionState == ERR_NONE) {
      // packet was successfully sent
     Serial.print(F("transmission finished!\t"));

    } else {
      Serial.print(F("failed, code "));
      Serial.println(transmissionState);
    }
   // Serial.print(F("Sending packet ... "));
    transmissionState= fsk.startTransmit(transmitArray, arraySize);
    enableInterrupt = true;
  } 
}`

jgromes, 2020-06-24:
I'm willing to bet 300 kbps FSK over 1 km won't happen without directional antenna and high-power PA on the transmitter, likely requiring a license to transmit (though you will need that anyway in US or EU, since that much data will definitely violate duty cycle regulations). I haven't tried the maximum range on SX1278 with 300 kbps speed, but I'd be amazed if you get more than a hundred meters line of sight using common antenna without external Tx PA and Rx LNA.

LoRa (or any LP solution in general) is designed to transmit small packets very often. I'd suggest using GSM if speed is critical.

closed 2020-06-24 by jgromes

Khyros12, 2020-06-24:
Thank You one more time for Your help.

sdey76, 2020-06-24:
Hi
Very interesting! How are u deciding the image file

Thanks for this discussion 

Regards

 Thank You one more time for Your help.  &mdash;

sdey76, 2020-06-24:
Sorry
How are you DECODING the image file
Regards

Cloolalang, 2020-06-24:
Hey Did you look at NPR (New packet Radio)?  https://hackaday.io/project/164092-npr-new-packet-radio
And easypal?  http://www.easypal.org/  for sending images

P
ZS1dbm

jgromes, 2020-06-24:
@sdey76 @Cloolalang guys this is not a general discussion board and is reserved to issues specific to RadioLib. Please go to a general discussion forum, such as Arduino forums.

cmthomlinson, 2024-05-16:
how do you increase the SPI frequency for Radio

> Since you haven't posted your transmitter code, it's hard to tell what could be causing the delays between successive transmissions. I guess large chunk of that is going to be the SPI transactions [...]

How do increase the SPI frequency to 10MHz?