jgromes/RadioLib #156: RFM69HCW wiring

szbrooks, 2020-06-26:
Hi, I have a beginner question

I'm using a RFM69HCW breakout board with an arduino UNO
and I'm getting a -2 error code when I run the RF69_Receive example (or any other example)

My wiring is:
CS ---> pin 10
RST ---> pin 3
MOSI ---> pin 11
MISO ---> pin 12
SCK ---> pin 13 

vin ---> 5V  (the breakout board says it can operate from 3.3 to 5V)
GND ---> GND

Do you have any suggestions on how I can troubleshoot the problem?

Thanks,

jgromes, 2020-06-27:
Could you post the debug mode output, as described [here?](https://github.com/jgromes/RadioLib/wiki/Debug-mode)

Another thing to check is whether the breakout board is compatible with 5V logic - RF69 is a 3.3V-logic device, so there must be an external voltage level shifter on that breakout board to use it with Arduino Uno, regardless of the supply voltage level.

szbrooks, 2020-06-28:
Board says it's 3-5V SPI logic safe,

the Debug output is:

`avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf"

         Using Port                    : COM4
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino
         Hardware Version: 3
         Firmware Version: 4.4
         Vtarget         : 0.3 V
         Varef           : 0.3 V
         Oscillator      : 28.800 kHz
         SCK period      : 3.3 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: reading input file "C:\Users\szbro\AppData\Local\Temp\arduino_build_295847/RF69_Receive.ino.hex"
avrdude: writing flash (10852 bytes):

Writing | ################################################## | 100% 1.74s

avrdude: 10852 bytes of flash written
avrdude: verifying flash memory against C:\Users\szbro\AppData\Local\Temp\arduino_build_295847/RF69_Receive.ino.hex:
avrdude: load data flash data from input file C:\Users\szbro\AppData\Local\Temp\arduino_build_295847/RF69_Receive.ino.hex:
avrdude: input file C:\Users\szbro\AppData\Local\Temp\arduino_build_295847/RF69_Receive.ino.hex contains 10852 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 1.39s

avrdude: verifying ...
avrdude: 10852 bytes of flash verified

avrdude done.  Thank you.`

Is this what you're looking for?

jgromes, 2020-06-28:
No, the debug output while running the code is displayed in Arduino Serial Monitor, please post that.

jgromes, 2020-06-28:
> Board says it's 3-5V SPI logic safe

Board might say a lot of things ;) Unless there's a an external IC like on [Adafruit breakout boards](https://learn.adafruit.com/adafruit-rfm69hcw-and-rfm96-rfm95-rfm98-lora-packet-padio-breakouts/overview), it is not 5V tolerant despite what the manufacturer claims.

szbrooks, 2020-06-28:
That's the breakout board I'm using,

I did the following:
`#define RADIOLIB_DEBUG
#define RADIOLIB_VERBOSE

#include <RadioLib.h>
`

But my Serial Monitor is still only saying:
[RF69] Initializing ... failed, code -2

jgromes, 2020-06-29:
Looks like the compiler won't pick up on those macros when using `#ifdef` as opposed to `#if defined()` - interesting, I'll fix it.

Regarding your problem, you'll have to uncomment the debug macros directly in BuildOpt file:
https://github.com/jgromes/RadioLib/blob/738d7116987cb9505c0ea1fefad8e18b56d73f51/src/BuildOpt.h#L213-L214

zbrookshaw, 2020-06-30:
[RF69] Initializing ... R     10           0              
RF69 not found! (1 of 10 tries) RF69_REG_VERSION == 0x0000, expected 0x0024
R             10           0              
RF69 not found! (2 of 10 tries) RF69_REG_VERSION == 0x0000, expected 0x0024
R             10           0              
RF69 not found! (3 of 10 tries) RF69_REG_VERSION == 0x0000, expected 0x0024
R             10           0              
RF69 not found! (4 of 10 tries) RF69_REG_VERSION == 0x0000, expected 0x0024
R             10           0              
RF69 not found! (5 of 10 tries) RF69_REG_VERSION == 0x0000, expected 0x0024
R             10           0              
RF69 not found! (6 of 10 tries) RF69_REG_VERSION == 0x0000, expected 0x0024
R             10           0              
RF69 not found! (7 of 10 tries) RF69_REG_VERSION == 0x0000, expected 0x0024
R             10           0              
RF69 not found! (8 of 10 tries) RF69_REG_VERSION == 0x0000, expected 0x0024
R             10           0              
RF69 not found! (9 of 10 tries) RF69_REG_VERSION == 0x0000, expected 0x0024
R             10           0              
RF69 not found! (10 of 10 tries) RF69_REG_VERSION == 0x0000, expected 0x0024
No RF69 found!
failed, code -2

I'd also like to thank you for holding my hand through this while I bastardize your hard work!

jgromes, 2020-06-30:
Judging by the output, I'd say this is a hardware issue - most likely something's not connected properly. If it was the SPI then I'd expect 0xFF as the output, so my guess is power.

The only other option I can see is that the module is dead.

zbrookshaw, 2020-06-30:
I killed it :( 

Ok, my main goal was to connect an MKRZero to an RFM95, this was more of a stepping stone project. 

Quick question about assigning the NSS and RST pins to the MRKZero board, you I get to specify them or are they preset like the MISO MOSI and SCK pins?

Thanks!

jgromes, 2020-07-01:
NSS, RST and other digital I/O pins can be set to any Arduino pin. The only limitation is when using interrupt-driven control, the DIO pins must be connected to an Arduino pin which supports external interrupts. See https://github.com/jgromes/RadioLib/wiki/Basics.

BTW: You're right that the SPI pins (MISO/MOSI/SCK) are fixed on Uno and other simple platforms, however, that is not the case universally, as some other architectures (like STM32) may have the option to use different pins for SPI, I2C etc. Not sure about MKRZero in this regard, I haven't used that board.

closed 2020-07-01 by jgromes