jgromes/RadioLib #157: Module Constructor of SPI for STM32F103 Series

muhammedoney, 2020-06-27:
Hi Jan,

I was trying to use the SX1262 lora module with my STM32 board. On my PCB, lora module is connected to SPI2 port. Thats why i need to change spi instances with the module constructor. I did try the solution mentioned in issue #121 Custom SPI pins but i can not make it work handling with lots of errors. 

I need to link MISO -> PB14
                     MOSI -> PB15
                     SCK -> PB13. 

                     NSS -> PB12,
                     DIO1 -> PA0,
                     NRST -> PA12,
                     BUSY -> PB5.

Could you please help me solving the problem.

jgromes, 2020-06-27:
That issue is unrelated, since you're using STM32, not ESP32. Which STM32 core are you using, official or unofficial?

If it's the former, then you will have to create a new instance of `SPIClass`, as described [here](https://github.com/stm32duino/wiki/wiki/API#spi). Provide the pins to the `SPIClass` constructor, and then provide the new SPI interface to the `Module` constructor:

```c++
SPIClass NewSPI(MOSI, MISO, SCK);
SX1262 lora = new Module(NSS, DIO1, NRST, BUSY, NewSPI);
```

If you're using the unofficial core, I don't know, as I don't have experience with it - though I imagine it will be very similar.

muhammedoney, 2020-06-27:
I'm using official core.
Thank you Jan. The document https://github.com/stm32duino/wiki/wiki/API#change-default-spi-instance-pins that you mentioned helped so much. 

I kept the Module constructor as original and didn't use the SPIClass constructor.

SX1262 lora = new Module(PB12, PA0, PA12, PB5);

and add 

SPI.setMISO(PB14);
SPI.setMOSI(PB15);
SPI.setSCLK(PB13);
SPI.begin(2);

in void setup()  function. Now, lora module and STM32 chip can communicating with SPI2.

closed 2020-06-27 by muhammedoney