jgromes/RadioLib #159: Support for BeliefSat-1

nvshinde, 2020-07-03:
I was trying to generate a pseudo packet file to test [LORA1268F30](https://www.nicerf.com/product_193_312.html) module based on SX1268 to use with gr-satellite. However was facing some issue with packet generation as discussed in [here](https://github.com/daniestevez/gr-satellites/issues/112)

The pseudo packet generator can be found [here](https://github.com/NewLeapKjsieit/BeliefSat/tree/master/temp), while audio recordings can be found [here](https://github.com/NewLeapKjsieit/BeliefSat/tree/master/Recordings). Not sure how to proceed further.

jgromes, 2020-07-03:
Brirefly looking over the generator, it seems like the encoding is left at default, which is whitening. I don't think that was accounted for when @daniestevez was attempting to decode.

Whitening is done using LFSR with 0x01FF initial state, the polynomial is x^9 + x^5 + 1. De-whitening is not entirely straight-forward, as bit order is reversed twice. So the complete de-whitening chain is:

1. Reverse bits within each byte (e.g 0001 0111 -> 1110 1000)
2. De-whiten using x^9 + x^5 + 1 (and initial state 0x01FF).
3. Reverse bits within each byte again.

This was verified using Universal Radio Hacker (and originally discussed in https://github.com/jopohl/urh/issues/749).

Alternatively, whitening can be disabled - just call `radio.setEncoding(0)` to set encoding to NRZ.

daniestevez, 2020-07-03:
Well, scrambling was not described in [comms system specifications](https://github.com/NewLeapKjsieit/BeliefSat/blob/master/README.md), so I didn't try that. Seeing this, I suspected that maybe there was scrambling, but also thought that it was unlikely that it was enabled by default. Maybe this also caught by surprise the BeliefSat-1 team.

jgromes, 2020-07-03:
It could be argued that the default should be NRZ - I don't know what's the consensus. To be honest I'm not entirely sure why it's enabled by default, might have been due to compatibility with another SX126x driver.

rnbokade, 2020-07-04:
We thought Whitening was disabled by default. Will make a new recording asap

jgromes, 2020-07-04:
New wiki page with complete default configuration of all modules was added (https://github.com/jgromes/RadioLib/wiki/Default-configuration), should prevent any future confusion

closed 2020-07-04 by jgromes