jgromes/RadioLib #161: Possibility to define LowDataRateOptimize setting

eg321, 2020-07-07:
**Is your feature request related to a problem? Please describe.**
There is "LowDataRateOptimize" setting in SX127x chips. This setting improves robustness of the transmission and mandatory for symbol duration exceeds 16ms, but also can be enabled in higher data rates. Our investigation shows that it this setting minimize Doppler effect (that's why this setting always enabled in our chips).
Current SX127x support implementation does not allow to configure this settings and always enables it for symbol rates >= 16 ms.
Unfortunately, this setting should be the same at transmitter and receiver side to avoid CRC errors. I cannot connect with remote side with current implementation.

**Describe the solution you'd like**
There should be a possibility to define needed value of LowDataRateOptimize bit (RegModemConfig 3 register). Some method like "setLowDataRateOptimization(bool) " will be ok.

**Describe alternatives you've considered**
There is one way to enable it now - correct RadioLib library.

**Additional context**
From Semtech docs:
> Given the potentially long duration of the packet at high spreading factors the option is given to improve the robustness of
the transmission to variations in frequency over the duration of the packet transmission and reception. The bit
LowDataRateOptimize increases the robustness of the LoRa link at these low effective data rates. Its use is mandated
when the symbol duration exceeds 16ms. Note that both the transmitter and the receiver must have the same setting for
LowDataRateOptimize.

jgromes, 2020-07-07:
> Our investigation shows that this setting minimize Doppler effect (that's why this setting always enabled in our chips).

Interesting - do you know of any downsides to having LDRO always enabled?

I agree LDRO option could be made public - perhaps `setLDRO(bool enable)` would be better (shorter) name for the method. If it made public however, it should also be done in the SX126x driver.

I probably won't get to it until the end of the week, feel free to open a PR in the meantime. Also, if it's urgent, you can use godmode to write the SPI register manually.

eg321, 2020-07-07:
> do you know of any downsides to having LDRO always enabled?
no, but can't find much details about that option actually. It works good for us, but may be not so good in other conditions.
It looks like public "setLDRO" the best way for now...

I've already enabled "godmode", so it's not urgent now.

Thank you!

jgromes, 2020-07-08:
Added `forceLDRO(bool enable)` method to force LDRO bit, and `autoLDRO()` to re-enable automatic configuration in 58194483b1fe24fbc6718f513055ac99b156e2c9

closed 2020-07-08 by jgromes

eg321, 2020-07-10:
@jgromes by the way, there is details of our experiments with LoRa and Doppler effect immunity:
https://research.nsu.ru/ru/publications/experimental-study-of-lora-modulation-immunity-to-doppler-effect-

maybe it will be useful for you too (in scope of Fossa` Cubsats).

jgromes, 2020-07-10:
Thank you, I will definitely read through that study.