jgromes/RadioLib #162: nRF52840 with RFM69 errors

DTronixs, 2020-07-07:
Setup:
Feather nRF52840 Express
FeatherWing RFM69HCW 900MHz
RadioLib 3.7.0
Arduino 1.8.12

I trying to get the example RF69_Transmit.ino to run and I get the following error message:

"[RF69] Initializing ... failed, code -16"

The RFM69HCW wiring is as per Adafruit's guide (CS = 5, IRQ = 10, RST = 6) and has been visually checked and continutity checked. Code has been modified as follows:

RF69 rf = new Module(5, 10, 6);

Any suggestions for what could be wrong?

Are there any options to enable further debugging?

Tony

jgromes, 2020-07-07:
> Are there any options to enable further debugging?

Only for those who use the issue templates and/or read the wiki :P See https://github.com/jgromes/RadioLib/wiki/Debug-mode

-16 is SPI write fail - real value of some SPI register after writing doesn't match the expected one. Enable debugging mode (verbose level), and it will show all the details in Serial monitor, post the output here.

DTronixs, 2020-07-09:
Thanks Jan, 

Here's the output:

[RF69] Initializing ... Found RF69! (match by RF69_REG_VERSION == 0x24)

address:	0x7
bits:		7 0
value:	0b11011001
current:	0b11100100
mask:	0b11111111
new:		0b11011001
read:		0b11100100

address:	0x8
bits:		7 0
value:	0b0
current:	0b11000000
mask:	0b11111111
new:		0b0
read:		0b11000000

failed, code -16

jgromes, 2020-07-09:
It looks like the module refuses SPI writes to RF69_REG_FRF_MSB and RF69_REG_FRF_MID. I've seen this on a couple RFM69 modules before, when I tried to set frequency other than 915 MHz. It looks like some RFM69 modules are "locked" into a particular frequency value (0xe4c000, which corresponds to 915 MHz).

Try setting the frequency in begin() method to 915 MHz. That will bypass the error, though you won't be able to change the frequency of that module.

DTronixs, 2020-07-10:
Hi Jan
Yes, thats seems to be the problem. Changed code to 

radio.begin(915);

and the error messages goes away.

thanks for your help

closed 2020-07-10 by jgromes

phretor, 2020-07-12:
Weird. I've just found the exact same issue on the 433MHz version of this Adafruit module. I was getting the very same error when setting frequency other than 915MHz (yes, 915, no 433).

```
#include <SPI.h>
#include <RadioLib.h>

#define RFM69_SS 13
#define RFM69_IRQ 27
#define RFM69_RST 15

RF69 rf = new Module(RFM69_SS, RFM69_IRQ, RFM69_RST);

void setup()
{
  Serial.begin(115200);

  delay(5000);

  Serial.println(F("[RF69] Module is initializing..."));

  int state = rf.begin(915);

  if (state == ERR_NONE)
  {
    Serial.println(F("begin(915) success!"));
  }
  else
  {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true)
      ;
  }

  delay(1000);

  state = rf.setFrequency(433.4);

  delay(1000);

  if (state == ERR_NONE)
  {
    Serial.println(F("setFrequency(433.4) success!"));
  }
  else
  {
    Serial.print(F("failed setFrequency(433.4), code "));
    Serial.println(state);
    while (true)
      ;
  }

  rf.setFrequency(433.4);
  rf.setBitRate(1.55);
  uint8_t sw[] = {0xAA, 0xD5};
  rf.setSyncWord(sw, sizeof(sw));
}
```

Output:

```
[RF69] Module is initializing...
Found RF69! (match by RF69_REG_VERSION == 0x24)
begin(915) success!

address:        0x7
bits:           7 0
value:          0b1101100
current:        0b11100100
mask:           0b11111111
new:            0b1101100
read:           0b11100100

address:        0x8
bits:           7 0
value:          0b1011001
current:        0b11000000
mask:           0b11111111
new:            0b1011001
read:           0b11000000

failed setFrequency(433.4), code -16
```

There has to be something else, because I'm able to use this 433 module with RadioHead.

I double checked with a 915 (green dot on the package) and 433 (red dot on the package) module and I get the same behavior from RadioLib.

jgromes, 2020-07-12:
@phretor can you independently verify the RadioHead code actually transmits at 433 MHz (e.g. with an SDR)? RadioHead doesn't verify SPI writes, so if it failed the same way, the module would still be at 915 MHz, you just wouldn't know.

phretor, 2020-07-13:
# Setup
- Adafruit FeatherWing RFM69HW 433MHz on Feather ESP32

# RadioLib 4.0.1

```cpp
#include <SPI.h>
#include <RadioLib.h>

//ESP32
#define RFM69_SS 27
#define RFM69_IRQ 23
#define RFM69_RST 12
#define FREQ 433

RF69 rf = new Module(RFM69_SS, RFM69_IRQ, RFM69_RST);

void setup()
{
  Serial.begin(115200);
  while (!Serial);

  Serial.println(F("[RadioLib RF69] Module is initializing..."));

  int state = rf.begin(FREQ);

  delay(1000);

  if (state == ERR_NONE)
  {
    Serial.println(F("[RadioLib RF69] begin() success!"));
  }
  else
  {
    Serial.print(F("[RadioLib RF69] begin() failed, code: "));
    Serial.println(state);
  }
}

void loop()
{
  Serial.println(F("[RadioLib RF69] Transmitting packet ... "));

  uint8_t byteArr[] = "Hello World! Hello World! Hello World! Hello World!";
  int state = rf.transmit(byteArr, sizeof(byteArr));

  if (state == ERR_NONE)
  {
    // the packet was successfully transmitted
    Serial.println(F("[RadioLib RF69] TX success!"));
  }
  else if (state == ERR_PACKET_TOO_LONG)
  {
    // the supplied packet was longer than 64 bytes
    Serial.println(F("[RadioLib RF69] Message too long!"));
  }
  else
  {
    // some other error occurred
    Serial.print(F("[RadioLib RF69] TX failed, code: "));
    Serial.println(state);
  }

  // wait for a second before transmitting again
  delay(1000);
}
```

## Outcome
```
[RadioLib RF69] Module is initializing...
RF69 not found! (1 of 10 tries) RF69_REG_VERSION == 0x0080, expected 0x0024
RF69 not found! (2 of 10 tries) RF69_REG_VERSION == 0x0082, expected 0x0024
RF69 not found! (3 of 10 tries) RF69_REG_VERSION == 0x0005, expected 0x0024
RF69 not found! (4 of 10 tries) RF69_REG_VERSION == 0x00EA, expected 0x0024
RF69 not found! (5 of 10 tries) RF69_REG_VERSION == 0x0049, expected 0x0024
RF69 not found! (6 of 10 tries) RF69_REG_VERSION == 0x0012, expected 0x0024
RF69 not found! (7 of 10 tries) RF69_REG_VERSION == 0x0081, expected 0x0024
RF69 not found! (8 of 10 tries) RF69_REG_VERSION == 0x00F7, expected 0x0024
RF69 not found! (9 of 10 tries) RF69_REG_VERSION == 0x0011, expected 0x0024
RF69 not found! (10 of 10 tries) RF69_REG_VERSION == 0x0014, expected 0x0024
No RF69 found!
[RadioLib RF69] begin() failed, code: -2
[RadioLib RF69] Transmitting packet ... 

address:        0x1
bits:           4 2
value:          0b100
current:        0b0
mask:           0b11100
new:            0b100
read:           0b0

[RadioLib RF69] TX failed, code: -16
[RadioLib RF69] Transmitting packet ... 

address:        0x1
bits:           4 2
value:          0b100
current:        0b0
mask:           0b11100
new:            0b100
read:           0b0

[RadioLib RF69] TX failed, code: -16
[RadioLib RF69] Transmitting packet ... 

address:        0x1
bits:           4 2
value:          0b100
current:        0b0
mask:           0b11100
new:            0b100
read:           0b0

[RadioLib RF69] TX failed, code: -16
[RadioLib RF69] Transmitting packet ... 

address:        0x1
bits:           4 2
value:          0b100
current:        0b0
mask:           0b11100
new:            0b100
read:           0b0

[RadioLib RF69] TX failed, code: -16
[RadioLib RF69] Transmitting packet ... 

address:        0x1
bits:           4 2
value:          0b100
current:        0b0
mask:           0b11100
new:            0b100
read:           0b0

[RadioLib RF69] TX failed, code: -16
[RadioLib RF69] Transmitting packet ... 

address:        0x1
bits:           4 2
value:          0b100
current:        0b0
mask:           0b11100
new:            0b100
read:           0b0
```

# RadioHead (latest version pulled from official website)
```cpp
#include <SPI.h>
#include <RH_RF69.h>

//ESP32
#define RFM69_SS 13
#define RFM69_IRQ 27
#define RFM69_RST 18
#define FREQ 433

RH_RF69 rf69(RFM69_SS, RFM69_IRQ);

void setup()
{
  Serial.begin(115200);
  while (!Serial);

  Serial.println(F("[RadioHead RF69] Module is initializing..."));

  if (!rf69.init())
    Serial.println("[RadioHead RF69] init() failed");

  if (!rf69.setFrequency(FREQ))
    Serial.println("[RadioHead RF69] setFrequency failed");

  rf69.setTxPower(20, true);

  Serial.println(F("[RadioHead RF69] Setup completed..."));
}

void loop()
{
  Serial.println(F("[RadioHead RF69] Transmitting packet ... "));

  // Send a message to rf69_server
  uint8_t data[] = "Hello World! Hello World! Hello World! Hello World!";
  rf69.send(data, sizeof(data));

  if (!rf69.waitPacketSent())
    Serial.println("[RadioHead RF69] TX failed");

  delay(1000);
}
```

Outcome:
```
[RadioHead RF69] Module is initializing...
RH SPI Driver initialized
RH init with device type: 36
[RadioHead RF69] Transmitting packet ... 
[RadioHead RF69] Transmitting packet ... 
[RadioHead RF69] Transmitting packet ... 
```

And verified with a BladeRF that it does transmit.

phretor, 2020-07-13:
Please hold on; I'm working on a more thorough test.

phretor, 2020-07-14:
I confirm the bug is not there for the 433 module I was testing. Was a glitch in the connections.

jgromes, 2020-07-14:
So the module is transmitting at 433 MHz with RadioHead and returning SPI fails with RadioLib? Or is it working correctly with both?

phretor, 2020-07-14:
It’s working correctly with both.

jgromes, 2020-07-14:
OK, thanks - so I guess some of the modules really are "locked" into 915 MHz.

phretor, 2020-07-15:
Oh yes.

rifath-shaarook, 2020-08-11:
Hi, I'm having 2 Adafruit Feather 32u4 with RFM69. Both are 433 MHz versions and transmitting just fine with RadioHead Library in 433 MHz, I have checked with SDR, but I put 433 MHz in RadioLib it's showing error -16. When I put 915, the error goes away but the module is neither transmitting in 433 nor 915, again I've confirmed with SDR, but it's showing "successfully transmitted" in serial port... The board is having  Red dot on it which means it is a 433 MHz module.

Here is the details when I set in 433 MHz. 
address:	0x7
bits:		7 0
value:		0b1101100
current:	0b11100100
mask:		0b11111111
new:		0b1101100
read:		0b11100100

address:	0x8
bits:		7 0
value:		0b1000000
current:	0b11000000
mask:		0b11111111
new:		0b1000000
read:		0b11000000

failed, code -16

jgromes, 2020-08-11:
This issue is popping up way too often to be just coincidence - maybe it's the verification mechanism itself that's causing the problem (though I'm not sure why or how).

@rifath-shaarook could you try if using just a simple register write will resolve this? Go to RadioLib/src/modules/RF69/RF69.cpp and change lines 385 - 389 from

```c++
  int16_t state = _mod->SPIsetRegValue(RF69_REG_FRF_MSB, (FRF & 0xFF0000) >> 16, 7, 0);
  state |= _mod->SPIsetRegValue(RF69_REG_FRF_MID, (FRF & 0x00FF00) >> 8, 7, 0);
  state |= _mod->SPIsetRegValue(RF69_REG_FRF_LSB, FRF & 0x0000FF, 7, 0);

  return(state);
```

to

```c++
  _mod->SPIwriteRegister(RF69_REG_FRF_MSB, (FRF & 0xFF0000) >> 16);
  _mod->SPIsetRegValue(RF69_REG_FRF_MID, (FRF & 0x00FF00) >> 8);
  _mod->SPIsetRegValue(RF69_REG_FRF_LSB, FRF & 0x0000FF);

  return(ERR_NONE);
```

rifath-shaarook, 2020-08-11:
Thank you so much...

It just worked like a charm and transmitting at the correct frequency... i.e. 433 Mhz

jgromes, 2020-08-12:
It indeed looks like some RF69 modules just don't like the RF frequency registers being read. As far as I can tell, the datasheet mentions nothing about this. Usually I would be hesitant about disabling a safety feature, but in this particular case I'm more than happy to make an exception - this issue was reported a couple of times already.