jgromes/RadioLib #165: failed, code -104

QuitButton, 2020-07-15:
I'm trying to get a RFM22 module working with RadioLib on a Bluepill STM32.  I have it wired to the SPI (PA5-7) port and using

Si4432 radio = new Module(PA4, PA2, PA0);

when I then call int state = radio.begin();
I get the error code -104, which seems to be defined as
#define ERR_INVALID_RX_BANDWIDTH                      -104

I'm not at all sure why I get that error as even with radio.begin(434.0, 48.0, 50.0, 181.1, 10, 16) it returns the same error.

jgromes, 2020-07-16:
Please fill in all the information that can be found in the ["module not working" issue template](https://github.com/jgromes/RadioLib/blob/master/.github/ISSUE_TEMPLATE/module-not-working.md).

QuitButton, 2020-07-16:
**Sketch that is causing the module fail**

```
/*
   RadioLib Si443x Receive Example

   This example receives packets using Si443x FSK radio module.
   To successfully receive data, the following settings have to be the same
   on both transmitter and receiver:
    - carrier frequency
    - bit rate
    - frequency deviation
    - sync word

   Other modules from Si443x/RFM2x family can also be used.

   For default module settings, see the wiki page
   https://github.com/jgromes/RadioLib/wiki/Default-configuration#si443xrfm2x

   For full API reference, see the GitHub Pages
   https://jgromes.github.io/RadioLib/
*/

// include the library
#include <RadioLib.h>

// Si4432 has the following connections:
// nSEL pin:  10
// nIRQ pin:  2
// SDN pin:   9
Si4432 radio = new Module(PA4, PA2, PA0);

// or using RadioShield
// https://github.com/jgromes/RadioShield
//Si4432 radio = RadioShield.ModuleA;

void setup() {
  Serial.begin(9600);

  delay(1000); // Delay needed or no serial output.

  // initialize Si4432 with default settings
  Serial.print(F("[Si4432] Initializing ... "));
  int state = radio.begin();
  if (state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }
}

void loop() {
  Serial.print(F("[Si4432] Waiting for incoming transmission ... "));

  // you can receive data as an Arduino String
  String str;
  int state = radio.receive(str);

  // you can also receive data as byte array
  /*
    byte byteArr[8];
    int state = radio.receive(byteArr, 8);
  */

  if (state == ERR_NONE) {
    // packet was successfully received
    Serial.println(F("success!"));

    // print the data of the packet
    Serial.print(F("[Si4432] Data:\t\t"));
    Serial.println(str);

  } else if (state == ERR_RX_TIMEOUT) {
    // timeout occurred while waiting for a packet
    Serial.println(F("timeout!"));

  } else if (state == ERR_CRC_MISMATCH) {
    // packet was received, but is malformed
    Serial.println(F("CRC error!"));

  } else {
    // some other error occurred
    Serial.print(F("failed, code "));
    Serial.println(state);

  }
}
```

**Hardware setup**
Wiring diagram, schematic, pictures etc.

**Debug mode output**
R	1	6	
Found Si443x!
R	3	20	
R	4	3	
R	7	1	
W	7	1	
R	7	1	
R	6	3	
W	6	0	
R	6	0	
R	33	22	
W	33	82	
R	33	82	
R	32	C	
W	32	0	
R	32	0	
R	70	C	
W	70	C	
R	70	C	
W	6E	C	
W	6F	49	
R	1C	1	
R	1C	1	
R	70	C	
0
0
0
10.417	10	A	3	3	83	53
100663	18937
65535	FFFF
R	21	1	
W	21	1	
R	21	1	
R	20	64	
W	20	53	
R	20	53	
R	21	1	
W	21	1	
R	21	1	
R	22	47	
W	22	89	
R	22	89	
R	23	AE	
W	23	37	
R	23	37	
R	24	2	
W	24	7	
R	24	7	
R	25	8F	
W	25	FF	
R	25	FF	
R	71	0	
W	71	0	
R	71	0	
W	72	50	
failed, code -104

**Additional info (please complete):**
 - MCU: Bluepill F103C8
 - Link to Arduino core: https://github.com/stm32duino/BoardManagerFiles/raw/master/STM32/package_stm_index.json
 - Wireless module type: RFM22
 - Arduino IDE version: 1.8.12
 - Library version: Patch 4.0.1

I also get this message from the arduino IDE:

> 
> In file included from C:\Users\Quit\Documents\Arduino\libraries\RadioLib-master\src/TypeDef.h:4,
>                  from C:\Users\Quit\Documents\Arduino\libraries\RadioLib-master\src/RadioLib.h:43,
>                  from C:\Users\Quit\AppData\Local\Temp\arduino_modified_sketch_993972\Si443x_Receive.ino:22:
> C:\Users\Quit\Documents\Arduino\libraries\RadioLib-master\src/BuildOpt.h:124:57: note: #pragma message: RADIOLIB_PLATFORM: Arduino STM32 (official)
>   124 |     #define RADIOLIB_PLATFORM                           "Arduino STM32 (official)"
>       |                                                         ^~~~~~~~~~~~~~~~~~~~~~~~~~
> C:\Users\Quit\Documents\Arduino\libraries\RadioLib-master\src/RadioLib.h:55:41: note: in expansion of macro 'RADIOLIB_PLATFORM'
>    55 |   #pragma message "RADIOLIB_PLATFORM: " RADIOLIB_PLATFORM
>       |                                         ^~~~~~~~~~~~~~~~~

jgromes, 2020-07-16:
Thank you, I think that's all I need for now. I added a bit of tolerance to the Rx bandwidth comparison, could you try it now (using the latest commit)?

QuitButton, 2020-07-17:
I now get failed, code -11

Debug output is:
`R	1	6	
Found Si443x!
R	3	20	
R	4	3	
R	7	1	
W	7	1	
R	7	1	
R	6	3	
W	6	0	
R	6	0	
R	33	22	
W	33	82	
R	33	82	
R	32	C	
W	32	0	
R	32	0	
R	70	C	
W	70	C	
R	70	C	
W	6E	C	
W	6F	49	
R	1C	1	
R	1C	1	
R	70	C	
0
0
0
10.417	10	A	3	3	83	53
100663	18937
65535	FFFF
R	21	1	
W	21	1	
R	21	1	
R	20	64	
W	20	53	
R	20	53	
R	21	1	
W	21	1	
R	21	1	
R	22	47	
W	22	89	
R	22	89	
R	23	AE	
W	23	37	
R	23	37	
R	24	2	
W	24	7	
R	24	7	
R	25	8F	
W	25	FF	
R	25	FF	
R	71	0	
W	71	0	
R	71	0	
W	72	50	
R	1C	1	
W	1C	96	
R	1C	96	
R	1C	96	
R	1C	96	
R	70	C	
1
1
0
15.625	15	F	5	5	125	7D
67108	10624
2518	9D6
R	21	1	
W	21	1	
R	21	1	
R	20	53	
W	20	7D	
R	20	7D	
R	21	1	
W	21	1	
R	21	1	
R	22	89	
W	22	6	
R	22	6	
R	23	37	
W	23	24	
R	23	24	
R	24	7	
W	24	1	
R	24	1	
R	25	FF	
W	25	D6	
R	25	D6	
R	34	8	
W	34	2	
R	34	2	
failed, code -11
`

jgromes, 2020-07-17:
I fixed a couple more problems in the last commit, could you check again?

QuitButton, 2020-07-17:
Looking good so far, I've now got it receiving and transmitting. I'll test a bit more over the weekend.

jgromes, 2020-07-17:
Thank you - Si443x isn't a module I use much, so there are probably other bugs lurking in the code. Feel free to open new issues ore PRs.

closed 2020-07-17 by jgromes