jgromes/RadioLib #168: Can I use this with CC1101 to read data from Bresser 5-in-1 (decoding code included)

vekexasia, 2020-07-18:
Hello All,

i would like to use this library with an esp32 to be used with a cc1101 to read data from Bresser-5-in-1. ([infos here](https://github.com/merbanan/rtl_433/blob/master/src/devices/bresser_5in1.c#L103)).

The weather station uses the following parameters on the 868.3MhZ frequency:

https://github.com/merbanan/rtl_433/blob/master/src/devices/bresser_5in1.c#L155-L164

As far as I can understand (noob here) the cc1101 could be used to read this with RadioLib. But i am not sure how to start up the module within the code to match the decoding "demodulation" parameters. Ex: 868.3, FSK, preamble, also short_width and long_width I am not sure how i can do it.

jgromes, 2020-07-18:
> As far as I can understand the cc1101 could be used to read this with RadioLib

What gave you that impression? I'm not sure what exactly is "FSK-PCM encoding" supposed to be. CC1101 is a 2-FSK/OOK module, so if you're not using one of these two modulations, it won't be able to receive the data as packets. It might be able to receive them in direct mode, but that will require a lot of processing on the ESP32.

Even assuming it can receive those packets, from https://github.com/andreafabrizi/BresserWeatherCenter#reversing-and-packet-structure it looks like that station is transmitting packets with 5-byte preamble. CC1101 can't set 5-byte preamble (see the [API docs](https://jgromes.github.io/RadioLib/class_c_c1101.html#acbfa80f431f335d5597500319f0affa8) - they're quite detailed). I suppose you could get around the issue by having a 4-byte preamble, and then use the fifth byte as a part of the sync word, but that's just a wild speculation.

vekexasia, 2020-07-18:
Hello @jgromes thanks for your reply. I hope FSK-PCM is just FSK (2-FSK). I'm not sure if I should also init RadioLib with other parameters such as the short_width and long_width parameter. 

From my understanding if everything works as expected I should:
 0. call `begin` with freq = 868.3, br =??, freqDev=??, rxBw=??, preambleLength 32 
 1. `setsyncWord with 0xd4
 2 (maybe) call `disableAddressFiltering`
 4. call `startReceive`
 5. cross my fingers.

not sure about the `??` parameters above.

jgromes, 2020-07-18:
Can you post SDR recordings (or even just waterfalls) of the signal in question? Without that it will be just a guessing game, but here we go anyway:

Modulation parameters:
* Since the packet shown at https://github.com/andreafabrizi/BresserWeatherCenter#reversing-and-packet-structure was sampled at 48 KHz and is supposed to have "an average of 6 samples per bit", I guess the bit rate is going to be about 8 kbps. 
* There's no way to guess frequency deviation, you'd have to examine the signal with an SDR.
* Rx bandwidth is the bandwidth of receiver filter, so safer to put it fairly wide at 125.0 kHz (assuming good signal and not much interference in nearby channels).

Packet parameters:
* Assuming you want to try and use the 4-byte sync word, you will have to set sync to 0xAA 0x2D - bear in mind there's now going to be one extra byte (0xD4) prior to the actual payload.
* Address filtering is disabled by default.
* The packet doesn't have a length field, so you will have to use fixed length mode by calling `fixedPacketLengthMode(27)` (1 overflown sync byte + 26 data bytes).
* The packet doesn't have a CRC at the end, so call `setCrcFiltering(false)`.

I don't know what's short_width and long_width supposed to be - it's no modulation/packet parameter I have heard of, so it might be something internal in rtl_433.

vekexasia, 2020-07-18:
Hey @jgromes thanks for your response again. Here attached a sample with 300k. I opened it in audacity by setting unsigned 8bit, mono, no endianness..

Out of curiosity, what is the process of determining the frequency deviation out of the dump attached below?
[g001_868.3M_300k.zip](https://github.com/jgromes/RadioLib/files/4942278/g001_868.3M_300k.zip)

jgromes, 2020-07-19:
When you open the file in Audacity, switch from waveform to spectrogram, and adjust the scale and window size, you get something like this:

[image]

The first part of the packet is the preamble, and it looks like the high and low frequencies differ by about 700 Hz. That would put 2-FSK frequency deviation at 350 Hz (because it's the offset from carrier). The lowest frequency deviation available on CC1101 is 1.587 kHz, so now I'm pretty confident at least one of the following options is true:
1. It's not actually 2-FSK.
2. Even if it is, CC1101 won't be able to receive it as packets.

As I mentioned previously, you could try direct mode (it basically bypasses packet handling and outputs data and clock of the demodulated signal directly on the GDO pins), but even if it works it will require additional post-processing on the ESP32. What you receive this way most likely won't be the data you're looking for, though you might be able to decode it.

vekexasia, 2020-07-19:
Thanks again. Is there an example on how to use direct mode?

Thanks

jgromes, 2020-07-19:
Not really, since using it is very situational - you just have to provide the modulation parameters via `radio.begin(868.3, 8.0, 1.587, 135.0);` and then start direct mode reception with `radio.receiveDirect();`.

closed 2020-07-19 by jgromes

vekexasia, 2020-07-19:
So i opened the receiver and took a picture of the module. but 'm not sure what I'm looking at. Maybe you recognize it?

[image]

Close up view of the chip:
[image]

jgromes, 2020-07-19:
Could be an RFM23 (https://www.sparkfun.com/products/12030), aka Si443x. RadioLib supports those as well and they're pretty cheap, so it might be worth trying out instead of the CC1101.

vekexasia, 2020-07-19:
I see that the value range for freqDev for the si443x is 0.625 in the lower bound while we(you :)) thought it was arount 0.3 khz :)

Am i missing something here? :)

jgromes, 2020-07-19:
The data you've provided suggests 350 Hz frequency deviation, though it is possible I misinterpreted something. Hard to guess from one sample which I don't know much about :)

You're also saying you opened the receiver - is the transmitter the same?

vekexasia, 2020-07-19:
Hey @jgromes the trasmitter is on my roof. Not easily accessible unfortunately for me now. If there is something else i can do or dump to keep investigating (except from going on my roof :)) I'll be happy to. Anyway today i'm going to receive a CC1101. MAYBe and just MAYBE it'll work...

jgromes, 2020-07-19:
Well you could take an oscilloscope or digital analyzer, hook it up to the receiver RF chip and capture all the traffic on its SPI bus (assuming it really is RFM22). Time-consuming, but then you'd know exactly what sort of configuration is being done to it on startup.

vekexasia, 2020-07-19:
hey @jgromes unfortunately I am not equipped with those. anyhow i opened up urh to get a spectrum analyzer.

And this is what it looks like
[image]

There are 2 peaks 1 at 868.19M and the other at 868.312. 

With My little understanding of the radio frequency I also tried to measure the frequency for the 0 bit and for the 1 bit.
1 bit -> from peak to peak ~8,5us -> 117khz
0 bit -> from peak to peak ~98us -> 1000000 / 98 => 10204 Hz

But i am not sure these makes any sense. considering there is a diff of about 90khz with the uS measurements below and a diff of ~110khz with the waterfall peak inspection above.

jgromes, 2020-07-19:
The waterfall should be a better indication. 110 kHz is a lot, but within RFM22 range. You could try a different SDR program - as good as URH is at decoding, the spectrum analyzer is rather basic.

It could also explain why I got such low frequency deviation from Audacity - it won't be able to handle frequencies that high, since the data rate seems to be 8 kbps I was expecting somewhere around 4 - 16 kHz.

vekexasia, 2020-07-19:
I should be in the range of cc1101 as well, the API shows 1.876 to 380kHz. any suggestion on the SDR Program? I'm on Ubuntu.

jgromes, 2020-07-19:
Sorry, I mostly use Windows - maybe GQRX?

vekexasia, 2020-07-19:
hey @jgromes my cc1101 came through but i guess it's faulty. Error -2 was prompted so i checked the wiring and also tried with a tester to see if (on the chip) the wires were ok.

Anyhow i was thinking. Do you think it would be possible to put a d1mini within the receiver board sniffing the data?

jgromes, 2020-07-19:
> cc1101 came through but i guess it's faulty

Seems unlikely - check the debug mode output. Which SPI are you using, is it set up correctly?

> d1mini within the receiver board sniffing the data

You could set it up as SPI slave, but I'm not entirely sure the bus will still work with two slaves in parallel, I never tried something like that.

vekexasia, 2020-07-19:
Hey @jgromes 
[image]

This is the connection picture.

[image]

BTW: Is there a way i can buy you a beer? You've been very patient :)

jgromes, 2020-07-19:
It seems you haven't connected the SPI pins. On Arduino Uno (which is what you're using, though you mentioned ESP32 previously), that means connecting SCLK to pin 13, MISO to 12 and MOSI to 11.

Finally, you're connecting a 3.3V device to a 5V Arduino - it might work for a while, but eventually it will break. You should use a logic level shifter.

vekexasia, 2020-07-19:
damn.. Anyway quickly did that and still same result from the library.

I'm using arduino uno cause it was easier to test with. 

Using the `CC1101 radio = new Module(10, 2, RADIOLIB_NC, 3);` initialization params.

vekexasia, 2020-07-19:
Posting a pic for reference. Btw i am using the 3.3V pin not sure if that is ok considering what you said:

[image]

vekexasia, 2020-07-19:
if you give me some directions on the wiring i can try the esp32 and see what happens. I also tried another library using the raspi. no luck.

jgromes, 2020-07-20:
That's just the power - you also need to ensure the signals are at 3.3V as well. Arduino Uno (or the clone you have) has 5 V signals, so logic 1 on the pins is represented by a voltage of 5V. The CC1101 is only supposed to work with 3.3V signals, so if you directly apply 5V to its pins, you're risking damage. You need some logic level shifter to convert between the level (BSS138 works well and is cheap).

ESP32 is a 3.3V device, so you can connect it directly to the ESP32. You'll need to check the pinout of your ESP32 module, and connect to one of the SPIs, as well as to two GPIO pins (one for NSS, one for interrupt). Something like this (untested - I don't use ESP32 often, so I'm not familiar with its SPI handling):

```c++
#include <RadioLib.h>

SPIClass newSPI(VSPI);

Module* mod;
CC1101* radio;

void setup() {
  newSPI.begin(5, 19, 18, SS);
  mod = new Module(10, 2, RADIOLIB_NC, 9, newSPI);
  radio = new CC1101(mod);
}
```

vekexasia, 2020-07-21:
Ok new CC1101 Arrived, was able to wire it on the esp32 and it does initialize properly (I guess that is a good sign.)

I tried the params we discussed above but it seems that anything below rxBw = 135 triggers error code -104. Anyhow it does not seem to receive anything. Keep in mind that the weather station is on the roof so it's far from here.

I'm not sure if am able to replicate the signal with my SDR to have more power and check if it's an antenna problem or not.

I tried directMode by calling receiveDirect() and then radio.receive(str) => still no luck unfortunately. I used these params: radio.begin(868.3, 8, 120, 135.0, 10, 32);

vekexasia, 2020-07-21:
Posting code for reference: 

```
#include "weatherstation.h"
#include <RadioLib.h>

// Original code: https://github.com/fredilarsen/TeslaChargeDoorOpener/blob/master/TeslaChargeDoorOpener.ino
namespace elektronvolt {

SPIClass newSPI(VSPI);

CC1101 radio = new Module(5, 2, RADIOLIB_NC, 4, newSPI, SPISettings());

void WeatherStation::setup() {
    newSPI.begin(18, 19, 23, 5);

    Serial.print(F("[CC1101] Initializing ... "));
    int state = radio.begin(868.3, 8, 120, 135.0, 10, 32);
    if (state == ERR_NONE) {
        Serial.println(F("success!"));
        state = radio.setCrcFiltering(false);
        if (state != ERR_NONE) {
            Serial.println(F("FAILED SET CRC FILTERING"));
            while (true)
                ;
        }
        state = radio.fixedPacketLengthMode(27);
        if (state != ERR_NONE) {
            Serial.println(F("FAILED SET PACKET LENGTH MODE"));
            while (true)
                ;
        }

        state = radio.setSyncWord(0xAA, 0X2d, 0, false);
        if (state != ERR_NONE) {
            Serial.println(F("FAILED SET PACKET LENGTH MODE"));
            while (true)
                ;
        }
    } else {
        Serial.print(F("failed, code "));
        Serial.println(state);
        while (true)
            ;
    }
}
void WeatherStation::loop() {
    Serial.print(F("[CC1101] Waiting for incoming transmission ... "));
    String str;
    radio.receiveDirect();
    int state = radio.receive(str);
    if (state == ERR_NONE) {
        // packet was successfully received
        Serial.println(F("success!"));

        // print the data of the packet
        Serial.print(F("[CC1101] Data:\t\t"));
        Serial.println(str);

        // print RSSI (Received Signal Strength Indicator)
        // of the last received packet
        Serial.print(F("[CC1101] RSSI:\t\t"));
        Serial.print(radio.getRSSI());
        Serial.println(F(" dBm"));

        // print LQI (Link Quality Indicator)
        // of the last received packet, lower is better
        Serial.print(F("[CC1101] LQI:\t\t"));
        Serial.println(radio.getLQI());

    } else if (state == ERR_CRC_MISMATCH) {
        // packet was received, but is malformed
        Serial.println(F("CRC error!"));

    } else {
        // some other error occurred
        Serial.print(F("failed, code "));
        Serial.println(state);
    }
}
} // namespace elektronvolt

```

I ugess there is something off with the freqDev or the other params maybe?

jgromes, 2020-07-22:
I think you may have misunderstood what direct mode does - it will directly output demodulated data to the GDO0 and GDO2 pins as a stream of bits, essentially turning the CC1101 from a packet-based module into a simple FSK demodulator with serial output. Reading data like this isn't supported by the library (at least not yet), so you have to read it yourself by sampling the data output on the rising edge of clock output. Since the packet handler is disabled at that point, you also have to decide whether the bits coming out of it are actual data you're looking for or just noise and check things like preamble and sync word (basically doing everything the packet handler does manually).

Also, you shouldn't use `String` class for reading the data (or for anything really)  - it will most likely have null bytes. You have to create a byte array large enough to fit the packet (27 bytes in your case).

I don't think it's an antenna/power issue (assuming the CC1101 module has at least a bit of wire as an antenna).

I'll look into the Rx bandwidth issue. On a side note, I believe Rx bandwidth value on CC1101 is dual sideband, whereas frequency deviation is single-sideband, so to fit in signal with 120 kHz deviation you should have at least 240 kHz Rx bandwidth (preferrably more). I'd also check whether the exact frequency you're transmitting on with an SDR, as well as the frequency you're listening on. I doubt the modules have highly accurate crystals or TCXOs, so there might be some frequency offset between the two (though it shouldn't matter for 120 kHz deviation).

Other than the above, I still remain sceptical this will work at all, there's just too many unknowns.

vekexasia, 2020-07-22:
Hey @jgromes Thanks for your time. I was trying to send out the a message with the cc1101 to inspect with the spectrum if the information sent via the chip matched the peaks (not sure that have any sense).

But the .transmit seems to hang indefinitely. Maybe I didnt wire it properly?

vekexasia, 2020-07-22:
Also I was wondering is there a way i can reach you out directly? I'm willing to donate you something if you can help me troubleshoot this. For example I'm on telegram via @el_bau handle

xenio, 2020-07-22:
Hello, I have a similar weather station from Bresser and I also using an SDR usb to catch the 868Mhz signal with rtl_433 similar to @vekexasia . I am very interest in development.

vekexasia, 2020-07-22:
So After countless hours of more debugging i decided it was time to wire it up again to my arduino clone and apparently i am able to send (still not sure why esp32 fails).

Anyhow i sent some messages and tuned the parameters to match the same waveform. 

I ended up with `  int state = radio.begin(868.35, 9, 60, 270.0, 10, 32);` Which has almost the same frequency peaks.

So I 

`setCrcFiltering(false)`
`fixedPacketLengthMode(27)`
`setSyncWord(0xAA, 0x2d, 0, false)`

and later in `loop()`

`  int state = radio.receive(myData, 27);`

So... I get some data... some at LQI ~ 60 and some at ~127. Assuming 127 needs to be ignored the data with low LQI seems to be consistent:
Here some snaps:
0x71 0x42 0xEF 0xFE 0xB8 0x7F 0x3B 0xBF 0xF5 0xFC 0xB8 0xE6 0xF7 0xFF 0xE3 0x4F 0x48 0x00 0xB1 0xC0 0x32 0x30 0x04 0x81 0xB1 0x8C 0xDC 

0x71 0x42 0xEF 0xFE 0xB8 0x7F 0x5F 0xBF 0xF3 0xFC 0xB8 0x66 0xF7 0xFF 0xE3 0x4F 0x48 0x00 0xB1 0xC0 0x28 0x10 0x07 0x01 0x91 0xCE 0xE2 

0x71 0xC2 0xEF 0xFE 0xB9 0xFF 0x4F 0xBF 0xF7 0xFC 0xBB 0x66 0xF7 0xFF 0xE3 0x0F 0x48 0x00 0xB1 0x80 0x2C 0x10 0x06 0x01 0x91 0x4E 0xE2 

0x71 0xC2 0xEF 0xFE 0xB9 0xFF 0x4F 0xBF 0xF7 0xFC 0xBB 0x66 0xF7 0xFF 0xE3 0x0F 0x48 0x00 0xB1 0x80 0x2C 0x10 0x06 0x01 0x91 0x4E 0xE2 

But now what bugs me more is: Is the sync-word really working? and why do I dont see the 0xD4 as the first byte? 

seems the bitrate from 8 - 9 changes stuff. ^^ above with 9. -> Below some snaps at bitrate 8

0xE6 0x0E 0xFF 0xC3 0x3F 0xBD 0xBF 0xEF 0xF2 0xEB 0x37 0xFF 0xF8 0xCF 0xA4 0x01 0xE6 0x02 0x11 0x00 0x40 0x68 0xA6 0x44 0x00 0x00 0xDC 

0xE5 0x06 0xFF 0xAF 0x1F 0xB7 0xBF 0xE3 0xF2 0xE7 0x36 0xFF 0xF8 0xD7 0xA4 0x02 0x87 0x02 0x41 0x00 0x60 0x68 0xC6 0x44 0x00 0x00 0xDA 

0xE2 0x0E 0xFF 0x9B 0x1F 0xB5 0xBF 0xE3 0xF2 0xE3 0x37 0xFF 0xF8 0xEF 0xA4 0x03 0x27 0x02 0x51 0x00 0x70 0x68 0xE6 0x44 0x00 0x00 0xDA

jgromes, 2020-07-22:
First, I'd check whether the timing of CC1101 receiving the packets actually matches the weather station sending them. Next, how did you arrive at 9 kbps bitrate? I checked the recording you have posted and the bit rate really does seem to be around 8 kbps (8.22 to be more precise, which in itself is a strange value).

vekexasia, 2020-07-22:
Hey Jan, yes. I'm verifying with the SDR. The packets with < 127 LQI get
through at the very same moment i see them on the SDR. So i'm pretty sure
that's the data I'm looking for.

Also as you can see above the data received seems to be pretty consistent
(while the others at LQI=127 looks more like garbage). The 9 kbps was taken
by measuring the whole packet length time. I probably made some mistakes.

I tried,  just for the sake of my sanity to put 8.22 and strangely enough
all the weather station data that Come through are with LQI = 0 (which
should be perfect I guess?) The information received is a lot more stable
than before. I always get almost identical information at each packet
received:

0xE6 0x0B 0x7F 0xF7 0xCF 0xF7 0xEF 0x9C 0xFD 0x98 0xCD 0xDF 0xFF 0x19 0xF4
0x80 0x08 0x30 0x08 0x10 0x63 0x02 0x67 0x32 0x20 0x00 0xDE
0xE6 0x0B 0x7F 0xF7 0xCF 0xF7 0xEF 0x9C 0xFD 0x98 0xCD 0xDF 0xFF 0x19 0xF4
0x80 0x08 0x30 0x08 0x10 0x63 0x02 0x67 0x32 0x20 0x00 0xDD
0xE6 0x0B 0x7F 0xF7 0xCF 0xF7 0xEF 0x9C 0xFD 0x98 0xCD 0xDF 0xFF 0x19 0xF4
0x80 0x08 0x30 0x08 0x10 0x63 0x02 0x67 0x32 0x20 0x00 0xDD
0xE7 0x0B 0x7F 0xF7 0xCF 0xF7 0xEF 0x9D 0xFD 0x98 0xCD 0xDF 0xFF 0x18 0xF4
0x80 0x08 0x30 0x08 0x10 0x62 0x02 0x67 0x32 0x20 0x00 0xDC
0xE7 0x0B 0x7F 0xF7 0xCF 0xF7 0xEF 0x9D 0xFD 0x98 0xCD 0xDF 0xFF 0x18 0xF4
0x80 0x08 0x30 0x08 0x10 0x62 0x02 0x67 0x32 0x20 0x00 0xE0

Looking at the decoding documentation it says that first 13 bytes is
somehow the inverted value of the second 13 bytes and if you look closely
0xE7 => 0x18 and 0xE6 => 0x19.

Though i dont see 0xD4 I would expect. I also tried
the RADIOLIB_ENCODING_WHITENING (not sure what it does though) and the
output (still constant in time is)
0x07 0x16 0xE5 0x0C 0x4A 0xDC 0xCB 0x77 0x87 0x45 0xF4 0xAF 0x68 0x4E 0xFE
0xD4 0x63 0x1D 0xC8 0x7D 0x6F 0xB8 0xE7 0x55 0x79 0xC7 0xDC
0x08 0x16 0xE5 0x1A 0x4A 0xC4 0xCB 0x77 0x87 0x45 0xF4 0xAF 0x68 0x41 0xFE
0xD4 0x75 0x1D 0xD0 0x7D 0x6F 0xB8 0xE7 0x55 0x79 0xC7 0xDC
0x06 0x16 0xE5 0x06 0x4A 0xDC 0xCB 0x76 0x87 0x45 0xF4 0xAF 0x68 0x4F 0xFE
0xD4 0x69 0x1D 0xC8 0x7D 0x6E 0xB8 0xE7 0x55 0x79 0xC7 0xDD
0x06 0x16 0xE5 0x06 0x4A 0xDC 0xCB 0x76 0x87 0x45 0xF4 0xAF 0x68 0x4F 0xFE
0xD4 0x69 0x1D 0xC8 0x7D 0x6E 0xB8 0xE7 0x55 0x79 0xC7 0xDB
0x07 0x16 0xE5 0x06 0x4A 0xDF 0xCB 0x77 0x87 0x45 0xF4 0xAF 0x68 0x4E 0xFE
0xD4 0x69 0x1D 0xCB 0x7D 0x6F 0xB8 0xE7 0x55 0x79 0xC7 0xDB

All values with LQI = 0 and BR=8.22 seems to come through at -138dBm

-- 

Andrea Baccega
[image: Email]me@andreabaccega.com
[image: Hangouts]vekexasia@gmail.com
[image: Personal site] <http://www.andreabaccega.com/> [image: Linkedin
Profile] <http://it.linkedin.com/in/andreabaccega> [image: Facebook]
<https://www.facebook.com/andrea.baccega> [image: Google+]
<https://plus.google.com/109217393200753135791> [image: Twitter]
<http://twitter.com/veke87> [image: StackOverflow]
<http://stackoverflow.com/users/314407/andrea-baccega> [image: GitHub]
<https://github.com/vekexasia>

jgromes, 2020-07-22:
I'm not quite sure why we're not seeing the second sync byte. Maybe you could also try to use promiscuous mode, which should disable almost all filtering.

Whitening refers to a type of encoding used by some radios to introduce more edges into the data - it doesn't seem likely the weather station packets are whitened, rtl_433 would have to de-whiten them in order to properly decode.

vekexasia, 2020-07-22:
Hmm @jgromes It looks like i got everything working... One thing i noticed is that with interrupts i get the 0xd4 :)

vekexasia, 2020-07-22:
Now i just need to understand why my esp32 wiring was not ok.

vekexasia, 2020-07-22:
Ok Re-Wired. copied over the arduino code, initialization goes ok but gdo0 pin does not gets called like it currently do in arduino. 

I tried changing gd0 pin to 32 but did not suceed. Earlier today i was even unable to send out packets using esp32 wiring. Probably there is something wrong within my initialization (or within the library?) I'm now using an esp32-dev board wroom similar to this one: 
[image]

Wiring 1
[image]

Wiring Top
[image]

Colors: 
MOSI: Grey
MISO: WHITE
CSN: PURPLE
SCK: GREEN
GO2 (unconnected): BLUE
GO0: YELLOW

jgromes, 2020-07-23:
Assuming you're still using interrupt-drive reception, you have to add an `IRAM_ATTR` attribute to the `setFlag` function, like this:

```c++
void IRAM_ATTR setFlag(void) {
  // check if the interrupt is enabled
  if(!enableInterrupt) {
    return;
  }

  // we got a packet, set the flag
  receivedFlag = true;
}
```

`setFlag()` is an interrupt service routine, on ESP32, these have to be placed into RAM.

vekexasia, 2020-07-23:
Nada. i tried that and I re-tried radio transmit (by removing also `startReceive` and the interrupt flag)

```
  byte byteArr[] = {0x01, 0x23, 0x45, 0x67, 0x89, 0xAB, 0xCD, 0xEF};
  int state = radio.transmit(byteArr, 8);
```
ESP32 hangs at it's execution.

Debug + verbose mode shows this:

```
OR     2       6
W       2       6
R       2       6
R       7       4
W       7F      1       23      45      67      89      AB      CD      EF
```

Could it be wiring or maybe it's something else? The same code just works fine on the Arduino clone I have.

vekexasia, 2020-07-23:
I think i made a mistake. In your previous comment ->

> ```c++
> #include <RadioLib.h>
> 
> SPIClass newSPI(VSPI);
> CC1101 radio(10, 2, RADIOLIB_NC, 9, newSPI);
> 
> void setup() {
>   newSPI.begin(5, 19, 18, SS);
>   radio.begin();
> }
> ```

you said I need a SS pin, I set such pin to the pin 5 which is the same pin used when initializing the Module.

So my initialization is:
```c++
SPIClass newSPI(VSPI);
CC1101 radio = new Module(5, 32, RADIOLIB_NC, 4, newSPI, SPISettings());
// ...
newSPI.begin(18, 19, 23, 5);
```

But i have now seen you didn't write 10 on the SS `newSPI.begin` parameter. So maybe I just got confused?

vekexasia, 2020-07-24:
Ok @jgromes with the same wiring i'm able to make cc1101 work on my esp32, at least the sending part using another library. It looks like there is something off code side on RadioLib?

vekexasia, 2020-07-24:
For reference This code works https://github.com/LSatan/SmartRC-CC1101-Driver-Lib/blob/master/examples/CC1101%20default%20examples/cc1101_Transmitt_Hello_World/cc1101_Transmitt_Hello_World.ino

jgromes, 2020-07-25:
> I tried the params we discussed above but it seems that anything below rxBw = 135 triggers error code -104.

I checked that, but I'm unable to reproduce this - lease note that the Rx filter bandwidth values can't be set to any value, only those specified in CC1101 datasheet, page 35 table 26.