jgromes/RadioLib #169: Question : Can packets longer than 64bytes be also recieved using Si443x based recievers

rnbokade, 2020-07-19:
Can this be achieved by recieve direct or responding to fifo half full interrupts?

jgromes, 2020-07-19:
With direct mode, there's essentially no limit on the "packet" size - because the packet handler is disabled, so the concept of "packet" doesn't apply anymore. The downside is that the host MCU has to do all the processing.

Using the standard interface, it could be possible, but it's not implemented. There's currently a PR (https://github.com/jgromes/RadioLib/pull/163) that does something similar on CC1101, which also has only 64 byte internal FIFO.

rnbokade, 2020-07-19:
In the direct mode does it reciever bits only after sync word is recieved or is syncword to be detected by the host MCU too?

jgromes, 2020-07-19:
Direct mode will completely bypass packet handler - at that point you basically have an FSK demodulator with digital output. The host MCU has to do everything, even distinguish between valid signal and noise.

rnbokade, 2020-07-19:
So anything I can do right now to receive a 120byte packet with 8byte preamble and 4byte syncword. And I "have to" use rfm-22b.

jgromes, 2020-07-19:
You could extend the reception methods to allow for packets longer than 64 bytes, by following what was done in #163 - set Rx FIFO threshold interrupt, then in `readData()` read the FIFO "on the fly".

closed 2020-08-02 by jgromes