jgromes/RadioLib #170: Low LoRa range

Khyros12, 2020-07-22:
Hello,
I struggle with achieving long range of LoRa modules. I use 2x Ra-02 433Mhz (SX1278) modules with 3dBi antennas and longest range that i achieved was 400m LOS. I know that 10km range can be achieved with powerful gateway but i thought that i'll get at least 2km LOS with this setup. Before i'll attach my code i would like to ask what range did you achieve with similar or other setup ?

jgromes, 2020-07-22:
> Before i'll attach my code

You have ommited a crucial chunk of information. LoRa range varies massively with its parameters, mainly spreading factor and bandwidth. For instance, consider the LoRa demodulator SNR range that can be found in datasheet - when using SF12, it can be as low as -20 dB (or 20 dB below the noise floor), whereas when using SF6 it's "only" -5 dB. That's a huge difference.

Next, the output power - you can set a large range of different values, from 2 to 17 dBm - again, this will cause large difference in maximum ranges.

Finally, the antenna. If it's one of the cheap "3dBi 433 MHz" antennas from China, then I can almost guarantee that it's not tuned - I've seen some that claimed to be for 433 MHz, while actually resonating at around 620 MHz. Once again, the impact on range will be massive.

>  i would like to ask what range did you achieve with similar or other setup ?

There is no way to meaningfully answer that question, because I know nothing about your setup and its actual performance. With bad antenna, low power and fast LoRa you'll be lucky to get a couple hundred meters. With good antenna, and appropriate modulation parameters, 2-3 km shouldn't be a problem (I had a setup like that with SF12 and 125 kHz bandwidth - quite fast in LoRa terms). With directional antenna, PA and LNA, several hundred km has been done.

Khyros12, 2020-07-23:
Hello,
i admit that antenna is cheap 3dbi antenna which i had to cut a bit to improve effectivness. Main reason for my post was to assure does this 15km performances written over internet are between 2 small lora modules, not lora module and gateway with huge antenna. As You can see in code below i use SF12, 125kHz with cr 4/8 and i send only 12B of data. This 2-3km that You mentioned before are available between 2 Ra-02 with 3-5dBi good quality antennas or rather huge 20dBi antennas ?

```
#include <RadioLib.h>

const uint8_t NSS_pin = 5;
const uint8_t DIO0_pin = 2;
const uint8_t RESET_pin = 14;
const uint8_t DIO1_pin = 25;

const float FREQUENCY = 433.0;
const float BANDWIDTH = 125.0;
const uint8_t SPREADING_FACTOR = 12; 
const uint8_t CODING_RATE = 8;//SX1278_CR_4_8
const uint8_t SYNC_WORD = 0x12;
const int8_t POWER = 20;
const uint8_t CURRENT_LIMIT = 500; 
const uint16_t PREAMLBE_LENGHT = 8;

int state = ERR_NONE;
volatile bool dataReceivedFlag = false;
volatile bool enableInterrupt = true;
volatile bool IRQwhileTransmit = false;

close_station_message_type send_message_type;
far_station_message_type receive_message_type;
byte fAlarmCode = 0;

bool communication_ready_flag = false;
bool communication_refresh_flag = false;
int communication_refresh_cycle_counter = 0;

scoreTable fScoreTable[30];
int lastPositioScoreTable = 0;

void IRQ_flag(void);

SX1278 LoRa = new Module(NSS_pin, DIO0_pin, RESET_pin, DIO1_pin);
WiFiServer server(80);

byte cMessageArray[CLOSE_MESSAGE_LENGHT];
byte fMessageArray[FAR_MESSAGE_LENGHT];

byte DEBUG_ostatni_licznik = 0;
int DEBUG_ostatni_RSSI = 0;

void setup() {
  Serial.begin(9600);
  Serial.print(F("Initializing ... "));
  state = LoRa.begin(FREQUENCY, BANDWIDTH, SPREADING_FACTOR, CODING_RATE, SYNC_WORD, POWER, CURRENT_LIMIT, PREAMLBE_LENGHT);
  state = LoRa.setCRC(true);
  check_state(state);

  LoRa.setDio0Action(IRQ_flag);
  Serial.print(F("Listening for incoming transmission ... "));

  state = LoRa.startReceive();
  check_state(state);
  pinMode(27, OUTPUT);
}

void loop() {
    //
    if (communication_refresh_flag && (++communication_refresh_cycle_counter > MAX_REFRESH_CYCLE_VALUE)) {
      communication_ready_flag = false;
      communication_refresh_flag = false;
      communication_refresh_cycle_counter = 0;
    }

    //if (!communication_ready_flag) send_message_type = c_message_handshake; //handshake wylaczony do testow
    fPrepareMessage(cMessageArray, send_message_type, CLOSE_DEVICE, FAR_DEVICE);
    
    if (send_message_type != c_clear) {
      IRQwhileTransmit=true;
      state = LoRa.transmit(cMessageArray, CLOSE_MESSAGE_LENGHT);
      check_state(state);
      IRQwhileTransmit=false;
      Serial.println("wyslane");
      send_message_type = c_clear;
    }
    
    /*
    if (dataReceivedFlag) {
      Serial.println("obrane");
      enableInterrupt = false;
      dataReceivedFlag = false;

      state = LoRa.readData(fMessageArray, FAR_MESSAGE_LENGHT);
      check_state(state);
      for(int kew=0;kew < FAR_MESSAGE_LENGHT; kew++){
        Serial.print(fMessageArray[kew]);
        Serial.print(" , ");
      }
      LoRa.startReceive();
      enableInterrupt = true;
    }*/

    switch ( fGetMessageType(fMessageArray)) {
      case FAR_TYPE_HANDSHAKE:
        timeFromFar = fGetScoreTime(fMessageArray);
        // synchronizeCloseFarTime();
        communication_ready_flag = true;
        break;
      case FAR_TYPE_READY:
        communication_refresh_flag = false;
        communication_refresh_cycle_counter = 0;
        break;
      case FAR_TYPE_ALARM:
        fAlarmCode = fGetAlarmCode(fMessageArray);
        //
        if(fAlarmCode == 79){
        //Serial.println(LoRa.getRSSI());
        DEBUG_ostatni_RSSI = LoRa.getRSSI();
        Serial.println(DEBUG_ostatni_RSSI);
        DEBUG_ostatni_licznik = fMessageArray[11];
        //digitalWrite(27, LOW);
        fAlarmCode=0;
        }
        //
        break;
    }
    fCleanMessage(fMessageArray);
    fPrepareMessage(cMessageArray, send_message_type, CLOSE_DEVICE, FAR_DEVICE);

    if (send_message_type != c_clear) {
      state = LoRa.transmit(cMessageArray, CLOSE_MESSAGE_LENGHT);
      check_state(state);
      send_message_type = c_clear;
    }
  }
  communication_ready_flag = false;

  if (dataReceivedFlag) {
      Serial.println("obrane");
      enableInterrupt = false;
      dataReceivedFlag = false;

      state = LoRa.readData(fMessageArray, FAR_MESSAGE_LENGHT);
      check_state(state);
      for(int kew=0;kew < FAR_MESSAGE_LENGHT; kew++){
        Serial.print(fMessageArray[kew]);
        Serial.print(" , ");
      }
      LoRa.startReceive();
      enableInterrupt = true;
    }
}

void IRQ_flag(void) {
  // check if the interrupt is enabled
  if (!enableInterrupt) {
    return;
  }// we got a packet, set the flag
  
  if(IRQwhileTransmit){ 
    //Serial.println("IRQwhileTransmit");
    LoRa.startReceive();
    }else{
    //Serial.println("not_IRQwhileTransmit"); 
    dataReceivedFlag = true;
    digitalWrite(27, LOW);
    }
}
```

jgromes, 2020-07-23:
Quick glance shows a couple issues with your code:

1. Current limit - you're assigning a value of 500 to a variable of uint8_t type. The range of that type is 0 - 255, so you'll only assign the lower byte of 500, which will probably be 244. That still doesn't matter though, because the allowed range is 45 to 240.

2. Output power - SX1278 is limited to +17 dBm. Moreover, 20 dBm might not be legal in your country without license.
3. You would have known both of these things had you checked the result of `radio.begin()`.

4. You're using outdated version of the library - current limit is not a part of the begin method as of 4.0.0.

The range I got was with off-the-shelf SX126x modules with tuned 3 dBi antenna and external LDO regulator as a power source for the transmitter.

Khyros12, 2020-07-24:
I changed my program and updated library as You suggested. I also changed antenna to tuned 3dBi antenna, now i can achieve around 2 km in rural area. Thank You for your help.

closed 2020-07-24 by Khyros12