jgromes/RadioLib #173: CC1101 send/receive not working with ESP32 + RadioLib 4.0.1

vekexasia, 2020-07-24:
**Describe the bug**
Hello, as mentioned in my previous issue #168, this library seems to work just fine with an arduino clone but it does not work properly when wiring it to an ESP32.

** What does not work + debugging**
- I tried the CC1101_Transmit example, the esp32 blocks its execution at `.transmit`
- I tried to receive without interrupt and no output is received
- I tried to receive with interrupt and callback is not getting called.
- I tried [another library](https://github.com/LSatan/SmartRC-CC1101-Driver-Lib/blob/master/examples/CC1101%20default%20examples/cc1101_Transmitt_Hello_World/cc1101_Transmitt_Hello_World.ino) example to trasmit data and it works. (By "it works" I mean that the code did not block AND I inspected the signal via an SDR).

Wiring on my ESP32:
 - CS: 5
 - GDO: 26,
 - SCK: 18
 - MISO: 19
 - MOSI: 23 

**To Reproduce**
Just try the Send example here.

**Additional info (please complete):**
 - MCU: [e.g. ESP32]
 - Link to Arduino core: [e.g. https://github.com/stm32duino/Arduino_Core_STM32 when using official STM32 core. See readme for links to all supported cores]
 - Wireless module type: CC1101
 - Arduino IDE version: Tried both arduino ide and platformio
 - Library version: 4.0.1

vekexasia, 2020-07-24:
I had some time to debug further:

It looks like this line is the line that halts the execution:
https://github.com/jgromes/RadioLib/blob/master/src/modules/CC1101/CC1101.cpp#L103-104

jgromes, 2020-07-25:
Finally had some tinme to try this out with actual ESP32, turns out `SPIsendCommand()` was using the default SPI instance instead of the one provided by user. That's also the reson why it was working on Arduino Uno - it only has one SPI.

Should be fixed now (checked with SDR), thanks for reporting!

closed 2020-07-25 by jgromes