jgromes/RadioLib #175: In ISerial.cpp, call to undefined member 'stopListening' for class SoftwareSerial

helloSeen, 2020-08-08:
Building for Teensy 4.0
For Sx126X chip
IDE: vscode using platformIO plugin

When building I get the following error:

```
RadioLib\src\ISerial.cpp:35:30: error: 'class SoftwareSerial' has no member named 'stopListening'
   
return(_mod->ModuleSerial->stopListening());
```

jgromes, 2020-08-08:
You should check out the warning it produces (assuming you have warnings enabled):

> C:\Users\user\Documents\Arduino\libraries\RadioLib\src/RadioLib.h:60:4: warning: #warning "RadioLib might not be compatible with this Arduino board - check supported platforms at https://github.com/jgromes/RadioLib!" [-Wcpp]

That's because Teensy is not a supported platform. Trying to compile for Teensy 4.0 without the reference to `SoftwareSerial::stopListening()` will succeed, but I'm a bit conflicted about adding Teensy as a supported platform. As far as I can tell, Teensy cores can't be installed via arduino-cli, which I'm using to run CI testing via Travis, so that would mean no automated builds prior to new version release. Also, I don't have any Teensy boards, so I can't verify whether it will actually work with hardware.

jgromes, 2022-06-12:
Even though this issue is way past stale, Teensy support for just added for arduino-cli, which means Teensy ca now be oficially supported in RadioLib.