jgromes/RadioLib #176: Wemos D1 Mini - "ISR not in IRAM!" when trying the Interrupt Examples

EpicLPer, 2020-08-25:
I'm new to LoRa (and in general playing around with breakout boards or Arduino programming in general :) ) and I just tried out the Receive and Transmit Interrupt Examples included with the project.

The non-interrupt ones worked fine and I was able to successfully transmit data which got me hyped up for LoRa already!  
But trying the Interrupt examples results in the ESP8266 (Wemos D1 Mini clone) to constantly crash with the following crashdump and error:
```
[SX1278] Initializing ... success!
ISR not in IRAM!

User exception (panic/abort/assert)
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Abort called

>>>stack>>>

ctx: cont
sp: 3ffffeb0 end: 3fffffc0 offset: 0000
3ffffeb0:  00000008 4023cb38 00000000 40203ea5  
3ffffec0:  000000fe 00000000 00000000 00000000  
3ffffed0:  00000000 00000000 00000000 00ff0000  
3ffffee0:  5ffffe00 5ffffe00 00000000 00000000  
3ffffef0:  00000001 00000004 3ffee36c 40204a22  
3fffff00:  40100702 00000000 000a0d00 40204a34  
3fffff10:  3ffe861f 00000000 3ffee36c 40204f49  
3fffff20:  00000000 00000000 3ffee310 0000000a  
3fffff30:  40203c3c 3ffee36c 3ffe861d 0000000a  
3fffff40:  00000000 3ffee310 3ffee36c 40205010  
3fffff50:  00000000 00000008 3ffee36c 4020159f  
3fffff60:  00000000 3ffee310 3ffee36c 4020241a  
3fffff70:  3fffdad0 3ffee310 3ffee36c 402010c3  
3fffff80:  0000000a 00000008 00000000 feefeffe  
3fffff90:  feefeffe feefeffe feefeffe 3ffee3d4  
3fffffa0:  3fffdad0 00000000 3ffee394 40204630  
3fffffb0:  feefeffe feefeffe 3ffe84e4 40100ff9  
<<<stack<<<
```

With my limited understanding of Arduino stuff so far it seems to get hung up on `radio.setDio0Action(setFlag);`.

Any way to fix this? Did I maybe do something wrong or is something not right in the Library?  
Thanks already :)

lillefyr, 2020-08-25:
I don't know if this can help.
I wrote it as a comment top myself last year. Seems the link could cover it

// Note if you use an Wemos D1 as MCU, ICACHE_RAM_ATTR is required
// https://community.blynk.cc/t/error-isr-not-in-iram/37426/20
//ICACHE_RAM_ATTR

I put the ICACHE_RAM_ATTR after variables and before procedures.

EpicLPer, 2020-08-25:
> I don't know if this can help.
> I wrote it as a comment top myself last year. Seems the link could cover it
>
> // Note if you use an Wemos D1 as MCU, ICACHE_RAM_ATTR is required
> // [...]

I'm kinda lost... not sure where to put this or how.

naggie, 2020-08-25:
https://stackoverflow.com/a/58131720 explains it pretty well, see the next answer for an example.

jgromes, 2020-08-25:
@EpicLPer ICACHE_RAM_ATTR is an ESP8266 attribute that will place a function into RAM (as opposed to the usual program storage, which is flash). On ESP8266, all interrupt service routines must be placed in RAM. Try placing the attribute before the ISR, like this:

```c++
void ICACHE_RAM_ATTR  setFlag(void) {
  // check if the interrupt is enabled
  if(!enableInterrupt) {
    return;
  }

  // we got a packet, set the flag
  receivedFlag = true;
}

```

EpicLPer, 2020-08-25:
If I do this then I get the following compile error:
```
In function 'void setup()':
SX127x_Receive_Interrupt_Example:54:23: error: 'setFlag' was not declared in this scope
   radio.setDio0Action(setFlag);
                       ^
exit status 1
'setFlag' was not declared in this scope
```

EpicLPer, 2020-08-25:
Okay, fixed it :) Had to put it before void for it to work!

```
ICACHE_RAM_ATTR void setFlag(void) {
  // check if the interrupt is enabled
  if(!enableInterrupt) {
    return;
  }

  // we sent a packet, set the flag
  transmittedFlag = true;
}
```

jgromes, 2020-08-25:
Couldn't remember which way it was, I'm glad it's working now.

closed 2020-08-25 by jgromes