jgromes/RadioLib #177: Problems decoding ax.25 AFSK packets using SX1278

packet-radio, 2020-08-29:
Hello,
I tried to use the RadioLib for sending ax.25 frames using AFSK and Bell202 tones. As a radio module I used the SX1278. My objective is to use the SX1278 as an 2m APRS transmitter for a school pico balloon project. On the software side I modified the ax.25 AFSK example a bit. At first glance the AFSK transmission worked just fine. The problem is, that I wasn’t able to decode the AFSK transmission using either multimon-ng or qtmm decoder sw. Only soundmodem (by UZ7HO) was able to decode them successfully. Do you see my mistake or is there any other solution to improve the decodability of ax.25 AFSK packets?

My sketch looks like this:

```
// include the library
#include <RadioLib.h>

// SX1278 has the following connections:
// NSS pin:   10
// DIO0 pin:  2
// RESET pin: 9
// DIO1 pin:  3
SX1278 radio = new Module(10, 2, 9, 3);

// or using RadioShield
// https://github.com/jgromes/RadioShield
//SX1278 radio = RadioShield.ModuleA;

// create AFSK client instance using the FSK module
// pin 5 is connected to SX1278 DIO2
AFSKClient audio(&radio, 5);

// create AX.25 client instance using the AFSK instance
AX25Client ax25(&audio);

void setup() {
  Serial.begin(9600);

  // initialize SX1278 with default settings
  Serial.print(F("[SX1278] Initializing ... "));
  int state = radio.beginFSK();
  state = radio.setPreambleLength(240);
  state = radio.setFrequency(432.5082);
  state = radio.setBitRate(1200);
  state = radio.setFrequencyDeviation(2.2);
  state = radio.setOutputPower(12.0);

  if(state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while(true);
  }

  // initialize AX.25 client
  Serial.print(F("[AX.25] Initializing ... "));

  state = ax25.begin("N0CAL");
  if(state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while(true);
  }
}

void loop() {
  // send AX.25 unnumbered information frame
  Serial.print(F("[AX.25] Sending UI frame ... "));
  
  AX25Frame frameUI("APZQAP", 2, "N0CAL", 5, AX25_CONTROL_U_UNNUMBERED_INFORMATION |
                    AX25_CONTROL_POLL_FINAL_DISABLED | AX25_CONTROL_UNNUMBERED_FRAME,
                    AX25_PID_NO_LAYER_3, "!4903.50N/07201.75W-Test 001234");
  int state = ax25.sendFrame(&frameUI);
  
  if (state == ERR_NONE) {
    // the packet was successfully transmitted
    Serial.println(F("success!"));
   
  } else {
    // some error occurred
    Serial.print(F("failed, code "));
    Serial.println(state);

  }

  delay(2000);
}
```

I have a question concerning the bitrate function. In the documentation was written, that the bitrate is set in kbps. So, for an AFSK1200 transmission the bitrate would be “1.2”. I have tried to set the bitrate to “1.2”, but then the received audio sounded way to slow. Only when I set the bitrate to “1200” the received audio sounded ok and the transmission was decodable using soundmodem (by UZ7HO). So which unit has the setBitRate() function (kbps or baud)?

Many thanks in advance!

jgromes, 2020-08-29:
Couple of thoughts:

1. You shouldn't be modifying things like bit rate, frequency deviation or preamble length when using the AX.25 client, in both AFSK and 2-FSK mode - `ax25.begin()` will take care of the settings, so the only things to modify are things like carrier frequency or output power, that do not change the modulation.

2. You're not checking the value of `state` in between those calls:

```c++
int state = radio.beginFSK();
state = radio.setPreambleLength(240);
state = radio.setFrequency(432.5082);
state = radio.setBitRate(1200);
state = radio.setFrequencyDeviation(2.2);
state = radio.setOutputPower(12.0);
```

If any of them fails, you won't know since the value of `state` gets overwritten by the subsequent call.

3. AFSK mode is implemented by supplying square wave to the SX1278 DIO2 pin while the radio is in "direct mode", the tone frequencies and baud rate are hard-coded for Bell202/AFSK1200. In direct mode, bit rate basically becomes sample rate of the DIO2 pin, so setting it as low as 1.2 kbps (the value for `setBitRate()` is in kbps) will prevent you from sending any tone higher than 600 Hz. Leave it at the default (48 kbps I believe).

As to the decoding issues you're seeing, I'm not sure what could be the cause, try setting different AX.25 preamble length in `AX25::begin()` - it's the number of AX.25 flags sent prior to the actual transmission, not the 2-FSK preamble length of `setPreambleLength()`. I have to admit I only checked AX.25 decoding with soundmodem, so it could be possible that for some reason it doesn't decode with other software. I'll try to check decoding with multimon-ng.

packet-radio, 2020-08-30:
Thanks a lot for your quick and detailed response. I really appreciate your help! With your help, I got a better understanding of how the RadioLib AFSK transmission works. I have also included your suggestions in the sketch. Unfortunately, multimon-ng and qtmm were still only able to decode a fraction of the transmitted packets. About 2 of 50 packets were successfully decoded. The others not. My local APRS digipeater was also not able to receive the transmission. But on the other hand, soundmodem (by UZ7HO) and APRSDroid were able to decode nearly every packet send. I tried using an longer ax.25 preamble, but this did not help either. Have you experienced that behavior in any way before too? Or do you have any other suggestions?

updated sketch:
```

// include the library
#include <RadioLib.h>

// SX1278 has the following connections:
// NSS pin:   10
// DIO0 pin:  2
// RESET pin: 9
// DIO1 pin:  3
SX1278 radio = new Module(10, 2, 9, 3);

// or using RadioShield
// https://github.com/jgromes/RadioShield
//SX1278 radio = RadioShield.ModuleA;

// create AFSK client instance using the FSK module
// pin 5 is connected to SX1278 DIO2
AFSKClient audio(&radio, 5);

// create AX.25 client instance using the AFSK instance
AX25Client ax25(&audio);

void setup() {
  Serial.begin(9600);

  // initialize SX1278 with default settings
  Serial.print(F("[SX1278] Initializing ... "));
  int state = radio.beginFSK(432.5082, 48, 2.2);

  if(state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while(true);
  }

  // initialize AX.25 client
  Serial.print(F("[AX.25] Initializing ... "));

  state = ax25.begin("N0CAL");
  if(state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while(true);
  }
}

void loop() {
  // send AX.25 unnumbered information frame
  Serial.print(F("[AX.25] Sending UI frame ... "));
  
  AX25Frame frameUI("APZQAP", 2, "N0CAL", 5, AX25_CONTROL_U_UNNUMBERED_INFORMATION |
                    AX25_CONTROL_POLL_FINAL_DISABLED | AX25_CONTROL_UNNUMBERED_FRAME,
                    AX25_PID_NO_LAYER_3, "!4903.50N/07201.75W-Test 001234");
  int state = ax25.sendFrame(&frameUI);
  
  if (state == ERR_NONE) {
    // the packet was successfully transmitted
    Serial.println(F("success!"));
   
  } else {
    // some error occurred
    Serial.print(F("failed, code "));
    Serial.println(state);

  }

  delay(2000);
}
```

jgromes, 2020-08-31:
You could try different frame types, not just UI, though I'm sceptical that will change much. This will definitely require some looking into, so I'll keep this issue open until I have to investigate further. In the meantime, you could try to record the frames as waveforms and examining those.

packet-radio, 2020-09-02:
I took some audio recordings of the frames, as you have recommended. Fortunately, I have a 70cm APRS tracker (build around the Si4432) and so I could compare the frames from the working APRS transmitter with the other frames. Oddly enough I did not see big differences between the frames from the different transmitters. The frequency of the Bell202 tones also seems to be quite accurate.

As another option in my case I tried putting the SX1278 in direct transmission mode and I used a different AFSK modulator sketch to create the PWM signal. This solution is a bit crappy, but will work for my school project in the meantime.

Anyway, many thanks for your help! This really helped me a lot!  But I do not want to stress your valuable time. So, feel free to close the issue.

jgromes, 2020-09-02:
> As another option in my case I tried putting the SX1278 in direct transmission mode and I used a different AFSK modulator sketch to create the PWM signal

That's interesting - if this solution works with multiple APRS decoders, it would point so some issue with the AX.25 implementation or with the way AFSK is handled. Could you share the modulator sketch and the recordings?

I'll keep this open for now.

packet-radio, 2020-09-05:
For the AFSK modulator I used the example sketch from https://github.com/handiko/Arduino-APRS.

The audio recordings are in the attachment. In the first recording "radiolib.wav" I used the radiolib. In the second recording "AFSK_modulator_handiko.wav" I used the AFSK modulator from handiko.

[audio_recordings.zip](https://github.com/jgromes/RadioLib/files/5178049/audio_recordings.zip)

jgromes, 2020-09-05:
Here's a spectrogram of the files in question.

[image]

Both of them can be decoded with soundmodem as:

```
1:Fm N0CAL-5 To APRS Via WIDE2-1 <UI R Pid=F0 Len=45> [16:59:49R] [+++]
!5233.17NH01326.20Ea A=000216/P244S7T27V244RH
```

Two things stand out:
1. RadioLib recording (the bottom one) has a lot shorter preamble than Arduino-APRS (the top one) - RadioLib uses 8 preamble flags by default, Arduino-APRS has 100. It also seems to be sending the low tone as the preamble.
2. The frame sent by RadioLib is a bit shorter - that could mean the baud rate is slightly off.

packet-radio, 2020-09-05:
Could the problem be that RadioLib sends high tones as preamble? Is it possible to change this?

jgromes, 2020-09-05:
Interestingly it looks like when the AX.25 preamble length is longer than the default 8 bytes, it's a high tone as well. Possibly there's something wrong in the NRZI encoding - could you try sending RadioLib AX.25 frames with 100 bytes of preamble (set in the ax25.begin() method)?

packet-radio, 2020-09-05:
audio recording with 100 byte preample:

[radiolib_100_byte_preamble.zip](https://github.com/jgromes/RadioLib/files/5178355/radiolib_100_byte_preamble.zip)

jgromes, 2020-09-05:
Still seems to have issues decoding with qtmm, I'll try a couple more things, like slightly adjusting the baudrate.

jgromes, 2020-09-06:
Changing baudrate seems to have no effect, however, examining the frames in Universal Radio Hacker, I noticed the following:

[image]

This should be the preamble (0x7E), but it seems there's something wrong with NRZ-I encoding or bit stuffing in RadioLib.

fazerlab, 2020-10-16:
I used Radiolib with SX1278 + ESP8266 and work fine decoding with SoundModem, Qtmm AFSK1200 Decoder and Direwolf. 
See [video](https://youtu.be/0kRAGgj-4nc), [Firmware and preliminary documentation](https://fazerlab.wordpress.com/2020/10/13/transmissao-ax-25-afsk-com-sx1278-e-esp8266/). 

Radiolib is a great librarie. Thank you jrgomes.

jgromes, 2020-10-16:
Good to know it's working with Direwolf, but I still think there's something weird going on with NRZ-I and bit stuffing, since the effect is clearly visible in the URH output above.

rnbokade, 2020-12-23:
I got it why your local digipeater didn't recieve it... Observe your spectrum that you have also posted on the blog. The spectrum bandwidth is more than 100khz. The receiver bandwidth of general digipeaters is few khz. The radiolib implementation is working perfect but the resultant bandwidth is too much... Normal AFSK 1200 should be 3khz AFAIK. This is too big than than. After you FM demodulate it, obviously audio is shifted as required, but before it, things are getting spread like anything... That's why this observation.

jgromes, 2020-12-23:
@rnbokade your point about AFSK bandwidth is a very valid one, but unfortunately ties into the "hack" RadioLib uses to achieve AFSK. It's done by directly modulating the output of the module, but it uses the default module FSK bit rate, which on SX1278 happens to be 48 kbps, hence the large resulting bandwidth.

I think what we could do instead is check the required AFSK bit rate, and then set bit rate low enough to still be able to oversample the tones (so 2.4 kbps at least for AX.25). Might also need some synchronization between the tones and radio transmissions since it's not going to be so massively oversampled as it is right now.

rnbokade, 2020-12-24:
I can try this post Christmas... Maybe by calling something like setBitrate(2400) before starting the tx.
Although at 2400 bps, the occupied be will be around 6khz. Should world with most radios. Maybe I'll test with my baofeng uv-5r radio. I have to anyway do testings on APRSLib based afsk1200 modem we are working on for one of othe satellites being developed at my work. Though in that we are using proper dra818v fm modules to modulate the sine waves generated by Arduino, which result in much cleaner and compatible tx. 
Though this makes me wonder, if direct mode is basically modulating the output of the transceiver, can't a similar approach of generating sinewaves with additional filter components on outside, be used to feed cleaner signal to the DIO2 and get proper hackless afsk on these modules too??.... Definitely worth test!

jgromes, 2020-12-25:
> Maybe by calling something like setBitrate(2400) before starting the tx

Probably won't work right away. Also, I made a mistake in my previous comment, you'll need at least 4.4 kbps since the highest tone in Bell 202 is 2200 Hz and not 1200.

I think you're misunderstanding how direct mode works - it can't do analog modulations, only digital keying. So you can't feed any waveforms to DIO2, just binary information. That's the hack, by feeding in square wave of varying frequency, we can get get sort-of AFSK on the other end. With loads of additional spectral components from the decomposition of course, but that's the price to pay for not using actual FM modulator.

matthias-straub, 2021-02-09:
> @rnbokade your point about AFSK bandwidth is a very valid one, but unfortunately ties into the "hack" RadioLib uses to achieve AFSK. It's done by directly modulating the output of the module, but it [...]

Hi there. I also did some tests with the AFSK examples and found that the high bandwidth is due to the 50khz default deviation in FSK mode for the sx127x chips. After putting "setFrequencyDeviation(0.6)" in the code examples, the bandwidth of the AFSK signals is much closer to what you would expect. From my understanding the bit rate only limits the maximum possible tone frequencies and should not have an effect on bandwidth for low frequency Bell 202 tones as long as you keep it above the required "2*highest tone" level.
On a side note - in OOK mode I got a nearly perfect AM modulated and narrow AFSK signal.

jgromes, 2021-02-09:
@matthias-straub the note about using OOK is very interesting - thanks! Maybe modules that support OOK should use it preferentially for AFSK, and only those that don't have it (like SX126x) should fall back on using narrow-band FSK.

matthias-straub, 2021-02-09:
> 
> 
> @matthias-straub the note about using OOK is very interesting - thanks! Maybe modules that support OOK should use it preferentially for AFSK, and only those that don't have it (like SX126x) should fall back on using narrow-band FSK.

Thanks for getting back to me so quickly. I agree - OOK seems to be the "cleaner" option due to the arbitrary 600hz deviation limit for FSK on sx127x. The only downside is that the result is AM modulated. This limits its interoperability (e.g. with digipeaters).

jgromes, 2021-02-13:
@rnbokade @packet-radio There's been some improvements to the FSK/AFSK layer configuration in the latest batch of commits. Checking with qtmm, it decodes perfectly in FSK mode (USB demodulator), but still has issues with AFSK. soundmodem works fine with both.

This leads me to believe that there's no encoding issue in the AX.25 implementation. Unfortunately, qtmm doesn't provide any feedback as to what's wrong with the signal. So I'm closing this for now, feel free to reopen if there's any new information.

closed 2021-02-13 by jgromes