jgromes/RadioLib #178: Add TRNG Functionality simply by adding some code! SX1262 tested & verified!

the-drums, 2020-08-30:
Hi There,

For my project I required a TRNG (true random number generator). (32bit)
I found out that the SX1262 has a specific register for it! (already specified in "sx1262.h"):
`
#define SX126X_REG_RANDOM_NUMBER_0                    0x0819
#define SX126X_REG_RANDOM_NUMBER_1                    0x081A
#define SX126X_REG_RANDOM_NUMBER_2                    0x081B
#define SX126X_REG_RANDOM_NUMBER_3                    0x081C
`
This gets it's value from the radio RX noise.

-------------------------------- Changes required:
Simply add: to sx1262.h:
`
uint32_t GetRandomValue(void);
`

& in sx1262.cpp add:
`
uint32_t SX1262::GetRandomValue(void){
    uint8_t val[] = { 0, 0, 0, 0 };
    startReceive(0);
    delay(1);
    readRegister(SX126X_REG_RANDOM_NUMBER_0, val, 4);
    standby();
    return ( val[0] << 24 ) | ( val[1] << 16 ) | ( val[2] << 8 ) | val[3];
}
`

Note: Only tested on SX1262 series, but works flawlessly!

Kind Regards, 
Paul

jgromes, 2020-08-31:
Nice - feel free to open a pull request for this. On a side note, there should be a similar register on some other modules supported by RadioLib (SX127x and RF69 IIRC), so maybe this `getRandom()` method could be included in the `PhysicalLayer` interface - it should be pretty easy to just read RSSI (without signal) to get TRNG.

the-drums, 2020-09-01:
Thank you for your reply. 
Haven't done any pull_requests before (quite new @ github). (Can't find a way to upload the changed script-file, but need to get into this).
But indeed I see that other units also carry a register, if all than it's better to add it higher level.
The measurements from the RX noise (maybe it's indeed simply the RSSI value) , are a proper true number generator and very random, so OK.
Even tho I have added some software encryption to enhance it even further.

Will have a look later on, once i figure out the pull_request stuff over here haha.

jgromes, 2020-09-01:
Pull requests are the way to propose modifications to the repository source. You create a fork of the repository (essentially making a copy into your own account), make the changes and then open a pull request into the original repository, which lets the owner/maintainer review the changes, and then merge your fork into the source repo.

Please make sure to follow the [contributing guidelines](https://github.com/jgromes/RadioLib/blob/master/CONTRIBUTING.md#code-style-guidelines) If you decide to go ahead with the PR, thanks.

the-drums, 2020-09-01:
Thank you for the information, that clarifies it up a bit.
Have some other modules lying around which I also need to cover.
Will try a fork by then.

jgromes, 2020-09-13:
Implemented in e7fb5551921c06a2d9bef2eac317fc49958ca749 - I also added `random` methods to almost all the other `PhysicalLayer` modules.

closed 2020-09-13 by jgromes