jgromes/RadioLib #181: Explicit Header SX127x

juanjole, 2020-09-02:
Hello!

I am developing a project of LoRa based on RadioLib and I would like to know if explicit header is already supported by SX127x module because it seems it is not and only implicit header is supported. I would need this feature to implement a variable-length packet.

Thank you for all the great job!

jgromes, 2020-09-02:
Explicit header is the default on SX127x (unless SF is set to 6, in that case implicit header is the only possible one).
Furthermore, when using SX127x in LoRa mode, variable packet length is the only supported mode of operation at this time.

So in fact, RadioLib doesn't support the inverse of what you're trying to do - implicit header and fixed-length LoRa packets. I don't think there's any particular reason why, so I will accept PRs implementing these features.

juanjole, 2020-09-02:
That's right! I thought that because the `receive` and `readData` methods require the expected number of bytes. Is there a way to access the actual length of the received packet?

jgromes, 2020-09-03:
Of course - see `SX127x::getPacketLength()`.

The length parameters in `receive` and `readData` methods are not required to be the number of bytes you expect to receive: if you specify more than what was actually received, you'll only get the received bytes. If you specify less, then you only get the amount of data you requested.

juanjole, 2020-09-04:
Okay, thank you for the information!

jgromes, 2020-09-04:
I'd rather keep this open until implicit header and fixed-length LoRa packets are implemented on SX127x - much like they are on SX126x.

jgromes, 2021-02-21:
Implemented in the latest commit.

closed 2021-02-21 by jgromes