jgromes/RadioLib #182: RF69 and STM32F1 blue pill compiling but not starting

vagos21, 2020-09-15:
Hello,

I hope this is the right place to report a strange "bug?"

I own 2 STM32F103 blue pills and for a couple of years i've been using a few RFM69HWs of 868MHz frequency.I was using the radiohead library with the roger clark core till now, but today i switched to the official STM32duino core. As you may guess, everything was working well (rf69HW easily covering 20km range!) but as many others i bumped my head as the library wouldn't compile for the new official core (yep, that setBitOrder thingy).

So moving on, i read somewhere that this amazing library works with the bluepills and official core. I use version 4.0.3 (latest for today) and use HID bootloader to upload the programs. Blink works perfectly, so does USB Serial port. So i load the RF69_Transmit example, it compiles just fine. I upload it and then.... that's it, nothing happens. Like the program never starts. I don't even get the first serial print

`Serial.print(F("[RF69] Initializing ... "));`

So.... any ideas?
Thank you for the huge time you've put in this project!
Vangelis

jgromes, 2020-09-15:
> I hope this is the right place to report a strange "bug?"

It is a right place, but a wrong issue type - bug reports have a template ;)

Please include [debug mode output](https://github.com/jgromes/RadioLib/wiki/Debug-mode) with both RADIOLIB_DEBUG and RADIOLIB_VERBOSE enabled.

vagos21, 2020-09-16:
Good day sir, thank you for your good mood and helping everyone out so fast! ⚡ 

I had some progress, i mean total chaos! First of all, i can't really describe how many times i banged my head against the wall trying to understand what's wrong with the blue pills, regarding the USB uploading process! I believe my STM32 is not a clone, also the USB resistor R10 on the board is the correct value 1.5K (152). But uploading gets stuck all the time, reset sucks big time works 1/10 tries 🦝  So i decided to go back to the good old serial FTDI method. And man i had no idea it's now @ Serial1 for the official core. At least now the board gets programmed correctly, and resets just fine. 

So, to the point: 
the Transmit sketch now works. Everything appears as a success and package sending is also successful. For each transmission i get this debug print:

```
[RF69] Transmitting packet ... R	1	4	
W	1	4	
R	1	4	
R	25	0	
W	25	0	
R	25	0	
W	27	FF	
W	28	FF	
W	0	C	
W	0	C	
R	37	10	
W	0	48	65	6C	6C	6F	20	57	6F	72	6C	64	21	
R	13	F	
W	13	F	
R	13	F	
R	5A	5D	
W	5A	5D	
R	5A	5D	
R	5C	7C	
W	5C	7C	
R	5C	7C	
R	1	4	
W	1	C	
R	1	C	
R	1	C	
W	1	4	
R	1	4	
W	27	FF	
W	28	FF	
success!
```

As for the receive (i use RF69_Receive_Interrupt), it also compiles, uploads and starts fine, and this is what i get in the debug:

```
[RF69] Initializing ... R	10	24	
Found RF69! (match by RF69_REG_VERSION == 0x24)
R	1	4	
W	1	4	
R	1	4	
R	1	4	
W	1	84	
R	1	84	
R	13	1A	
W	13	1A	
R	13	1A	
R	2	0	
W	2	0	
R	2	0	
R	2	0	
W	2	3	
R	2	3	
R	29	FF	
W	29	E4	
R	29	E4	
W	28	10	
R	26	5	
W	26	7	
R	26	7	
R	37	10	
W	37	90	
R	37	90	
R	3D	2	
W	3D	2	
R	3D	2	
R	3D	2	
W	3D	2	
R	3D	2	
R	38	40	
W	38	FF	
R	38	FF	
R	3C	F	
W	3C	8F	
R	3C	8F	
R	2A	0	
W	2A	FF	
R	2A	FF	
R	2B	0	
W	2B	FF	
R	2B	FF	
R	6F	0	
W	6F	30	
R	6F	30	
R	1	84	
W	1	84	
R	1	84	
W	7	D9	
W	8	0	
W	9	0	
R	1	84	
W	1	84	
R	1	84	
R	3	1A	
W	3	2	
R	3	2	
R	4	B	
W	4	9A	
R	4	9A	
R	1	84	
W	1	84	
R	1	84	
R	19	86	
W	19	42	
R	19	42	
R	1	84	
W	1	84	
R	1	84	
R	5	0	
W	5	3	
R	5	3	
R	6	52	
W	6	33	
R	6	33	
R	1	84	
W	1	84	
R	1	84	
R	11	9F	
W	11	9C	
R	11	9C	
W	2C	0	
R	2D	3	
W	2D	2	
R	2D	2	
R	37	90	
W	37	90	
R	37	90	
R	38	FF	
W	38	40	
R	38	40	
R	2E	98	
W	2E	88	
R	2E	88	
W	2F	12	AD	
R	1	84	
W	1	84	
R	1	84	
R	2	3	
W	2	0	
R	2	0	
R	1	84	
W	1	84	
R	1	84	
R	37	90	
W	37	90	
R	37	90	
R	37	90	
W	37	90	
R	37	90	
success!
[RF69] Starting to listen ... R	1	84	
W	1	84	
R	1	84	
R	25	0	
W	25	40	
R	25	40	
R	2A	FF	
W	2A	FF	
R	2A	FF	
R	2B	FF	
W	2B	FF	
R	2B	FF	
W	27	FF	
W	28	FF	
R	13	1A	
W	13	1A	
R	13	1A	
R	5A	55	
W	5A	55	
R	5A	55	
R	5C	70	
W	5C	70	
R	5C	70	
R	1	84	
W	1	90	
R	1	90	
success!
```
i have both identical RFM69HW units connected to the blue pills as follows:
```
DIO0 -> PB0
RESET -> PB1
SS -> PA4
SCK -> PA5
MISO -> PA6
MOSI -> PA7
3.3 -> 3.3
GND -> G
```

and the two units are sitting on the same breadboard, so the antennas are really close (i tried various transmit power values, even the high power one, still no packets received.

and the only things i've changed in the transmit/receive sketches is:

- replaced Serial with Serial1
- initialise the radio with the line `int state = radio.begin(868.0, 48.0, 50.0, 125.0, 10, 16);` because my modules are @868MHz
- change the pinout of the module with `RF69 radio = new Module(PA4, PB0, PB1);`

I would love to try out new things as i'm really interested in this nifty library :)
Thanks again!
Vangelis

jgromes, 2020-09-16:
You mentioned you're using RFM96H - did you call `radio.setOutputPower(10, true);`? RFM96H modules have a slightly different PA/LNA setup than other RFM69 modules, so to correctly set the output power you have to call `setOutputPower` with the second parameter set to `true`.

You could also check the transmitted signal with an SDR if you have that option.

vagos21, 2020-09-16:
yes I also used this command for the transmitter code. I also tried dialing down the power (just in case...) would you like me to post the whole code for transmitter and receiver?

jgromes, 2020-09-22:
I don't see anything immediately obvious from the debug outputs you posted, so yes - please post the full sketches, even if it's just slightly modified examples.

closed 2020-10-07 by jgromes