jgromes/RadioLib #184: Increasing output power and current for EBYTE 900M30S (SX1262)

samey3, 2020-09-25:
Hello,
I have an EBYTE module [(Product page)](http://www.ebyte.com/en/product-view-news.aspx?id=453) that has a max output power of ~31dBm. Despite modifying the range in SX1262::setOutputPower to meet this, it still does not work with values beyond 22dBm. After looking at the SX1262 datasheet, it mentions "- 9 (0xF7) to +22 (0x16) dBm by step of 1 dB if high power PA is selected", however doesn't provide any indication that higher output power can be achieved.

I think(?) I was able to increase the current by modifying SX126x::setCurrentLimit, however the limit seems to be 637.5mA since the SX126X_REG_OCP_CONFIGURATION register is only 1 byte?

I was wondering if you had any insight on what changes may be needed in order to allow setting a higher current limit (e.g. 1W) and higher output power (e.g. 30dBm).

Thanks!

jgromes, 2020-09-25:
SX1262's internal PA can't output more than 22 dBm, the module you linked to seems to have an external PA. Methods like setOutputPower can only change the configuration of SX1262's internal PA, you'll have to contact Ebyte to get details about what sort of external PA they added and how to configure its power output.

Word of caution - the external PA might not be able to handle large power input, so if you attempt to `setOutputPower(22)` and then transmit, you will be sending +22 dBm into the external PA. If it can't handle that, smoke might follow. Again, contact Ebyte for details.

Similarly with the over-current protection - that's only the internal limit of SX1262, it has no way of limiting the current draw of the external PA.

jgromes, 2020-09-25:
Also - depending on your country of residence, 30 dBm transmission may not be legal without license. It certainly isn't legal in Europe.

closed 2020-10-07 by jgromes