jgromes/RadioLib #187: Compilation errors when building RadioLib for Heltec CubeCell Board AB01

amirna2, 2020-10-14:
Hello,

I am hoping someone can help me. I am trying to build RadioLib (v4.0.4) for the `Heltec CubeCell Board AB01` Arduino compatible board. The board is an ASR650x SoC with an SX1262 (https://heltec.org/project/htcc-ab01/).
When I try to build one of the examples (SX126x Channel Activity Detection Example) I get the following error:

```
In file included from sketch/SX126x_Channel_Activity_Detection.ino.cpp:1:
/Users/anathoo/Library/Arduino15/packages/CubeCell/hardware/CubeCell/1.1.0/cores/asr650x/Arduino.h:85:27: error: variable or field 'PINMODE_RADIOLIB_PIN_MODE' declared void
 #define pinMode(pin,mode) PINMODE_ ## mode(pin)
                           ^~~~~~~~
/Users/anathoo/Documents/Arduino/libraries/RadioLib/src/Module.h:386:17: note: in expansion of macro 'pinMode'
     static void pinMode(RADIOLIB_PIN_TYPE pin, RADIOLIB_PIN_MODE mode);
                 ^~~~~~~
/
```

It seems the problem is the `Arduino.h` for the Heltec board re-defines `pinMode()` and that conflicts with the RadioLib override of the same method in Module.h
Here is how the macro is defined in Arduino.h

```
#define pinMode(pin,mode) PINMODE_ ## mode(pin)
#define PINMODE_ANALOG(pin) pinMode(pin,ANALOG)
#define PINMODE_INPUT(pin) pinMode(pin,INPUT)
#define PINMODE_OUTPUT_PULLUP(pin) pinMode(pin,OUTPUT_PULLUP)
#define PINMODE_OUTPUT_PULLDOWN(pin) pinMode(pin,OUTPUT_PULLDOWN)
#define PINMODE_OD_LO(pin) pinMode(pin,OD_LO)
#define PINMODE_OD_HI(pin) pinMode(pin,OD_HI)
#define PINMODE_OUTPUT(pin) pinMode(pin,OUTPUT)
#define PINMODE_OUTPUT_PULLUP_PULLDOWN(pin) pinMode(pin,OUTPUT_PULLUP_PULLDOWN)
#define PINMODE_INPUT_PULLUP(pin) ({pinMode(pin,OUTPUT_PULLUP); digitalWrite(pin,HIGH);})
#define PINMODE_INPUT_PULLDOWN(pin) ({pinMode(pin,OUTPUT_PULLDOWN); digitalWrite(pin,LOW);})
```

I tried to rename & replace `pinMode()` to `pinModeEx()` in RadioLib Module.h but this doesn't seem to solve the problem.

Thanks in advance for your help.
Amir.

jgromes, 2020-10-14:
Unfortunately I don't think there's a simple fix for this - RadioLib assumes `pinMode` is  a function declared in the global scope, which is true for every single platform RadioLib supports, and there's a lot of them. 

In the case of CubeCell, `pinMode` is not a function, but rather a preprocessor macro - and to make things worse, it's using string concatenation operator on its argument. This will completely break current `pinMode` handling in RadioLib.

If someone figures out how to fix this in RadioLib without breaking compatibility with currently supported platforms, I will accept that as a PR. However, I don't see it as a priority as CubeCell isn't a very wide-spread platform and already has its own internal SX126x driver.

closed 2020-10-14 by jgromes

amirna2, 2020-10-14:
Thanks. This makes sense, the fix shouldn't come from RadioLib. I was able to work around by commenting out the macro, but obviously this isn't a solution. Unfortunately the CubeCell Arduino core is missing a lot of other functions, making it impossible to use with RadioLib. Obviously Heltec is trying to promote their own implementation of LoRa and doing so in a way that may not make it easy for generic solutions.

RadioLib is currently used by the ClusterDuckProtocol (http://clusterduckprotocol.org/) open source wireless mesh network solution and I am helping with the integration of a broader range of devices, including Arm Cortex M0+ and SX1262 based platforms.

Thanks for the quick response.
Amir.

timmbogner, 2023-02-01:
Hi @jgromes, I've recently started using RadioLib in my project, [Farm Data Relay System](https://github.com/timmbogner/Farm-Data-Relay-System/pull/131). I'm trying to support these Heltec boards too, and I think something new is preventing compilation. The following is the result of trying to compile any of the SX126x examples under any of the CubeCell boards in Arduino 2.0.3:

```
In file included from c:\Users\webma\Documents\Arduino\libraries\RadioLib\src/RadioLib.h:39:0,
                 from C:\Users\webma\AppData\Local\Temp\.arduinoIDE-unsaved202311-5084-8dl3ll.ict8c\SX126x_Settings\SX126x_Settings.ino:28:
c:\Users\webma\Documents\Arduino\libraries\RadioLib\src/Module.h:65:12: error: 'constexpr' does not name a type
     static constexpr RfSwitchMode_t END_OF_MODE_TABLE = {MODE_END_OF_TABLE, {}};
            ^
c:\Users\webma\Documents\Arduino\libraries\RadioLib\src/Module.h:65:12: note: C++11 'constexpr' only available with -std=c++11 or -std=gnu++11
```

amirna2, 2023-02-03:
I am still able to build RadioLib for their AB01 and AB02S CubeCell boards with RadioLib version 5.6.0 and Heltec CubeCell Arduino version 1.5.0. The new hardware now has evolved to v2 (I am still using the older v1.1).

jgromes, 2023-02-03:
As per my previous comment:

> It's using an outdated GCC

and `constexpr` is only available in C++11. This was added as one of the changes for STM32WL support. @matthijskooijman is the `constexpr` needed, or will a simple `const` suffice?

matthijskooijman, 2023-02-03:
It seems it's not strictly needed, but it needs a couple more changes. I'll confirm and prepare a PR.

jgromes, 2023-02-04:
@timmbogner this should be fixed now (by @matthijskooijman)

timmbogner, 2023-02-06:
@jgromes Thanks! Great library, I'm implementing it on my [FDRS project](https://github.com/timmbogner/Farm-Data-Relay-System) and it's really helped me a lot.

@matthijskooijman Thank you as well! I've been watching your STM32WLx work because I'm hoping it will enable me to use RadioLib on [this E5](https://www.seeedstudio.com/Grove-LoRa-E5-STM32WLE5JC-p-4867.html) module. I'm still working on flashing it, though.