jgromes/RadioLib #190: AFSK on TTGO T-Beam SX1278

raufaser, 2020-10-20:
Hello everyone,

I am trying to get AFSK (and later AX.25) running on my TTGO T-Beam (Hardware Revision 1.1). The LoRa Module is SX1278. I tried a lot but to make it simple and avoid other sources of errors, i was trying to get the [Imperial March example running](https://github.com/jgromes/RadioLib/tree/master/examples/AFSK/AFSK_Imperial_March) example running.

PinOut of the T-Beam can be found [here](https://github.com/Xinyuan-LilyGO/LilyGO-T-Beam).
Schematic is [here](https://github.com/Xinyuan-LilyGO/LilyGO-T-Beam/blob/master/schematic/LilyGo_TBeam_V1.1.pdf).
There also is a code example for the board how to use all functions. It inculdes using LoRa. Can be found [here](https://github.com/Xinyuan-LilyGO/LilyGO-T-Beam/tree/master/examples/AllFunction).

Unfortuately i only get a keyed carrier. No "audio".

Here is my code.

`#include <Arduino.h>
#include <RadioLib.h>
#include <axp20x.h>
#include "melody.h"

#define I2C_SDA         21
#define I2C_SCL         22

#define LORA_SCK        5
#define LORA_MISO       19
#define LORA_MOSI       27
#define LORA_SS         18
#define LORA_DIO0       26
#define LORA_DIO1       33
#define LORA_DIO2       32
#define LORA_RST        23

#define FREQUENCY       433

AXP20X_Class axp;
SX1278 radio = new Module(LORA_SS, LORA_DIO0, LORA_RST, LORA_DIO1);
AFSKClient audio(&radio, LORA_DIO2);

void setup() {
  Serial.begin(115200);

  delay(1000);

  Wire.begin(I2C_SDA, I2C_SCL);

  if (!axp.begin(Wire, AXP192_SLAVE_ADDRESS)) {
    Serial.println("AXP192 Begin PASS");
  } else {
    Serial.println("AXP192 Begin FAIL");
  }

  Serial.printf("DCDC1: %s\n", axp.isDCDC1Enable() ? "ENABLE" : "DISABLE");
  Serial.printf("DCDC2: %s\n", axp.isDCDC2Enable() ? "ENABLE" : "DISABLE");
  Serial.printf("LDO2: %s\n", axp.isLDO2Enable() ? "ENABLE" : "DISABLE");
  Serial.printf("LDO3: %s\n", axp.isLDO3Enable() ? "ENABLE" : "DISABLE");
  Serial.printf("DCDC3: %s\n", axp.isDCDC3Enable() ? "ENABLE" : "DISABLE");
  Serial.printf("Exten: %s\n", axp.isExtenEnable() ? "ENABLE" : "DISABLE");
  Serial.println("----------------------------------------");

  axp.setPowerOutPut(AXP192_LDO2, AXP202_ON);
  axp.setPowerOutPut(AXP192_LDO3, AXP202_ON);
  axp.setPowerOutPut(AXP192_DCDC2, AXP202_ON);
  axp.setPowerOutPut(AXP192_EXTEN, AXP202_ON);
  axp.setPowerOutPut(AXP192_DCDC1, AXP202_ON);
  axp.setDCDC1Voltage(3300);  //esp32 core VDD    3v3
  axp.setLDO2Voltage(3300);   //LORA VDD set 3v3
  axp.setLDO3Voltage(3300);   //GPS VDD      3v3

  Serial.printf("DCDC1: %s\n", axp.isDCDC1Enable() ? "ENABLE" : "DISABLE");
  Serial.printf("DCDC2: %s\n", axp.isDCDC2Enable() ? "ENABLE" : "DISABLE");
  Serial.printf("LDO2: %s\n", axp.isLDO2Enable() ? "ENABLE" : "DISABLE");
  Serial.printf("LDO3: %s\n", axp.isLDO3Enable() ? "ENABLE" : "DISABLE");
  Serial.printf("DCDC3: %s\n", axp.isDCDC3Enable() ? "ENABLE" : "DISABLE");
  Serial.printf("Exten: %s\n", axp.isExtenEnable() ? "ENABLE" : "DISABLE");
  Serial.println("----------------------------------------");

  //SPI.begin(LORA_SCK, LORA_MISO, LORA_MOSI, LORA_SS);
  int initState;

  initState = radio.beginFSK(433);
  Serial.printf("Radio Init: %d\n", initState);
}

void loop() {
  //radio.transmit("012345678901234567890123456789012345678901234567890123456789");

  // calculate whole note duration
  int wholenote = (60000 * 4) / 120;

  // iterate over the melody
  for(unsigned int note = 0; note < sizeof(melody) / sizeof(melody[0]); note += 2) {
    // calculate the duration of each note
    int noteDuration = 0;
    int divider = melody[note + 1];
    if(divider > 0) {
      // regular note, just proceed
      noteDuration = wholenote / divider;
    } else if(divider < 0) {
      // dotted notes are represented with negative durations!!
      noteDuration = wholenote / abs(divider);
      noteDuration *= 1.5; // increases the duration in half for dotted notes
    }

    // we only play the note for 90% of the duration, leaving 10% as a pause
    audio.tone(melody[note]);
    delay(noteDuration*0.9);
    audio.noTone();
    delay(noteDuration*0.1);
  }

  delay(5000);
}`

And I attach a picture of what i see on my SDR.

[image]

What am i doing wrong?

jgromes, 2020-10-21:
Unmodulated carrier points to something wrong with the DIO2 connection. Looking over the datasheet, it looks like DIO2 (as well as DIO1) isn't connected to any pin on ESP32, just to the pin header. If that's the case, connect DIO2 to some digital IO pin on ESP32 and it should work.

raufaser, 2020-10-21:
Thanks alot for your quick reply. I guess i have to take pin 33 and 32 and connect it to LoRa IO1 and LoRa IO2. Since i defined these ports

``` c++
#define LORA_DIO1       33
#define LORA_DIO2       32

SX1278 radio = new Module(LORA_SS, LORA_DIO0, LORA_RST, LORA_DIO1);
AFSKClient audio(&radio, LORA_DIO2);
```

I checked output voltage on pin 32 and 33 and it varies (according to the melody). I just want to reassure before damaging something. 

I also found this PinOut graphic of the board i am using:

[image]

jgromes, 2020-10-21:
> I guess i have to take pin 33 and 32 and connect it to LoRa IO1 and LoRa IO2

Pretty much, yes. Though checking the DIO2 voltage level with multimeter (I presume) isn't going to tell you much, since it's varying frequency, not voltage. With an oscilloscope, you should see a modualted square wave. You can also check by just connecting a piezospeaker to that pin.

raufaser, 2020-10-21:
I checked the LoRa chip pins with the scope.

Here is what i got:

LoRa Chip Pin11 Labeled IO0
[image]

That pertty much looks like the length of the tones.

LoRa Chip Pin10 Labeled IO1
[image]

LoRa Chip Pin9 Labeled IO2
[image]

I also connected the ESP pins 32 and 33 to the LoRa IOs 1 and 2 with a physical soldered wire. Nothing changed...

I think i make a fundamental mistake ...

jgromes, 2020-10-21:
That's pretty weird, the only one that makes sense is DIO1. This is what you should get:

DIO0 - constantly at 0 V
DIO1 - 48 kHz clock (that one is the same as yours)
DIO2 - frequency-modulated square wave

Your DIO1 doesn't look like tones, the duty cycle should be 50%. DIO2 also has very little amplitude, so the waveform there is probably just crosstalk from the DIO1 line (since it's also at 48 kHz). My guess is bad soldering and/or incorrect pins.

raufaser, 2020-10-28:
According to the developer DIO1 is connected to 33 and DIO2 is connected to 32. I asked in the board specific github project. [here](https://github.com/Xinyuan-LilyGO/LilyGO-T-Beam/issues/32)

I also tried out a TTGO Lora32 v2.1 1.6 board. It shows exactly the same behaviour ...

Any suggestions what i could try out or test?

btw. LoRa works just fine.

jgromes, 2020-10-28:
I just realized that ESP32 doesn't support Arduino `tone()` function - so AFSK is currently unsupported on ESP32 (as well as some other platforms like Arduino Due) and I compeltely forgot about it (sorry).

I was kind of hoping they would add tone support to ESP32 core so that I wouldn't have to implement it in RadioLib, but looking over ESP32 core GitHub this has been raised several times and isn't worked on (see e.g. https://github.com/espressif/arduino-esp32/issues/980), so I'll add a hacky tone support here.

jgromes, 2020-10-28:
I just pushed a commit that should add tone support for ESP32 - could you try it out now?

raufaser, 2020-10-28:
First of all thank you very much for your support. Unfortunately it doesn't seem to work. I tested on my T-Beam and the Lora32 board. The behaviour doesn't change at all. I added a Serial.printf("Yep!\n"); to the interesting part of the Module.cpp just to be sure that i am using the updated code. On the SDR I still receive the unmodulated carrier and on the scope i see the same signals.

jgromes, 2020-10-29:
Thanks for the feedback, I think I have an idea of what's going on: the problem here is twofold:

1. I used PWM channel 15 to try and generate the tones, but it looks like only channels 0 and 1 work (tested on Heltec WiFi LoRa 32 V2). Fixed now.

2. On your board (as well as mine, actually), DIO2 is connected to GPIO32, which is an input-only pin. To get AFSK working you will have to connect DIO2 to pin which can output PWM. I tried this with GPIO25 and it works, checked with SDR.

raufaser, 2020-10-29:
Great success :) Thank you very much for your support and your patience.

I used the following FSK configuration

``` c++
initState = radio.beginFSK(433, 1.2, 6);
```
Here is what i received on my SDR:
[image]

Wave on DIO0 (ESP pin 26)
[image]
High when tone. Low when no tone.

Wave on DIO1 (ESP pin 33)
[image]
Seems to be 1 kHz when DIO0 is High. (am I right?)

Wave on DIO2 (ESP pin 32)
[image]
Square wave in frequency of the tone to play.

> On your board (as well as mine, actually), DIO2 is connected to GPIO32, which is an input-only pin. To get AFSK working you will have to connect DIO2 to pin which can output PWM. I tried this with GPIO25 and it works, checked with SDR.

I didn't have to do that. It simply worked.

jgromes, 2020-10-29:
Glad it's working, though it's a bit strange GPIO32 works as output for you.

closed 2020-10-29 by jgromes

raufaser, 2020-10-29:
When i use the AX25Client the transmitted audio sounds weird.

I uploaded a audio sample [here](https://sndup.net/6ptz/radiolib.mp3). **LOUD!**

It should sounds similar to [that](https://sndup.net/4spg/aprs.mp3). **LOUD!**

jgromes, 2020-10-29:
It sounds like the first sample has shorter preamble. The default one is 8 bytes long, you can try longer one (e.g. 32 bytes) like this: `ax25.begin("N7LEM", 0, 32);`. Does the AX.25 frame decode?

raufaser, 2020-10-29:
Yes. It decodes. With direwolf and AFSK 1200 decoder. Sorry for the false alarm. Now lets see if i can send out APRS packets which are accepted by digipeaters. Again, thanks a lot.

ve2gqf, 2021-11-03:
Hi Jan,
How is your project ?
Is it works ? I will interested to decode and transmit APRS with a TTGO T-Beam
thank's

jgromes, 2021-11-03:
@ve2gqf receiving and decoding AX.25 is currently not supported.

I'm currently working on a more streamlined APRS transmission support, though in theory it should work already, since any AX.25 frame can be transmitted.

fagci, 2024-06-12:
Thanks to your work!

[image]

Same issue here.
Using LILYGO T3_V1.6.1. Pinout is same for DIO1, DIO2.
Checked pins using multimeter, 20 Ohms for each ESP GPIO - sx1278 DIO.
Tried to use GPIO12, GPIO25 for DIO2, no luck.
Also tried to set GPIO pins HIGH before transmitting.

```cpp
#include <RadioLib.h>
#include <SPI.h>

const uint8_t LORA_MOSI = 27;
const uint8_t LORA_MISO = 19;
const uint8_t LORA_SCLK = 5;
const uint8_t LORA_CS = 18;
const uint8_t LORA_DIO0 = 26;
const uint8_t LORA_DIO1 = 33;
const uint8_t LORA_DIO2 = 32;
const uint8_t LORA_RST = 23;

SX1278 radio = new Module(LORA_CS, LORA_DIO0, LORA_RST, LORA_DIO1);
AFSKClient audio(&radio, LORA_DIO2);

RTTYClient rtty(&audio);

float TXfreq = 434.0;

void setup() {
  Serial.begin(115200);
  SPI.begin(LORA_SCLK, LORA_MISO, LORA_MOSI);

  int state = radio.beginFSK();
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true)
      ;
  }

  state = rtty.begin(434.0, 170, 45);
  if (state == RADIOLIB_ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true)
      ;
  }

rtty.idle();
  delay(500);
  rtty.println("Test test test");
  rtty.standby();

  // for (int i = 0; i < 10; ++i) {
  //   audio.tone(1000+i * 1000);
  //   delay(100);
  // }
  // audio.noTone();
}

void loop() {}
```

RadioLib v 6.6.0.

Have no logic analyser, but DIO2 HIGH = 0.88V using standard LILYGO layout.