jgromes/RadioLib #193: Arduino 8Mhz internal clock /sx1278 error 16

gradicom, 2020-11-01:
Hi, 

After using a Arduino with 8Mhz internal clock /SX1278 with radiolib 3.3 version without problem and IDE 1.89.
I decided to migrate to a newer version of radiolib, but when trying to start an example like RTTY or Morse it gives
 me error 16, indicating that there is a problem in the spi communication. I noticed this with versions after 3.3.
 Any idea what might be the soft problem? 

Regards.

Gustavo.

jgromes, 2020-11-01:
Please follow the templates when creating new issues, since I will always ask for the following:

1. The exact MCU and Arduino core -  I'm guessing you have 8 MHz Arduino Pro Mini, so that would be an ATmega328P MCU running at 8 MHz with the official Arduino AVR core, but I need this confirmed.

2. What is the output of [debug mode](https://github.com/jgromes/RadioLib/wiki/Debug-mode)? That will usually contain some useful info, especially with error code -16 since it will tell us exactly which SPI register failed and how.

3. Which exact example does this happen with - Morse_Transmit or Morse_Transmit_AFSK? Were there any modifications to the example code? And does it only happen with Morse and RTTY or also with plain SX127x examples?

gradicom, 2020-11-03:
Hi,

I using  ATmega 328AU  with internal 8mhz clock, Breadboard Arduino. I attached the file  that take in debug mode using the RadioLib 4.0.5 now.  I tried the RTTY example this time without sucess,  but also the other examples,(MORSE & AX25), can't run.

Gustavo.

[debug.txt](https://github.com/jgromes/RadioLib/files/5484520/debug.txt)

jgromes, 2020-11-04:
There is no error reported in the attached debug output - please make sure both RADIOLIB_DEBUG and RADIOLIB_VERBOSE are uncommmented.

gradicom, 2020-11-06:
I attached a second debug file, now testing AX25 example using 4.05 lib and ide 1.8.13. Same error -16.
[degub2.txt](https://github.com/jgromes/RadioLib/files/5503433/degub2.txt)

jgromes, 2020-11-07:
Here's the contents of the above posted txt file - I fail see to any error codes. Please post a debug output that actually contains the error code.

```
[SX1278] Initializing ... R	42	12	
Found SX127x!
R	1	9	
R	1	9	
R	1	9	
W	1	9	
R	1	9	
R	1	9	
R	1	9	
W	1	9	
R	1	9	
R	2	1A	
W	2	68	
R	2	68	
R	3	B	
W	3	2A	
R	3	2A	
R	1	9	
R	1	9	
W	1	9	
R	1	9	
R	4	0	
W	4	0	
R	4	0	
R	5	52	
W	5	8	
R	5	8	
R	1	9	
R	1	9	
W	1	9	
R	1	9	
R	13	B	
W	13	2	
R	13	2	
R	12	15	
W	12	2	
R	12	2	
R	1	9	
W	1	9	
R	1	9	
R	B	2B	
W	B	23	
R	B	23	
R	1	9	
W	1	9	
R	1	9	
R	1	9	
R	25	0	
W	25	0	
R	25	0	
R	26	3	
W	26	10	
R	26	10	
R	1	9	
R	27	93	
W	27	93	
R	27	93	
R	27	93	
W	27	91	
R	27	91	
W	28	12	AD	
R	1	9	
R	30	90	
W	30	90	
R	30	90	
R	33	0	
W	33	0	
R	33	0	
R	34	0	
W	34	0	
R	34	0	
R	1	9	
R	1	9	
W	1	9	
R	1	9	
R	E	2	
W	E	2	
R	E	2	
R	E	2	
W	E	2	
R	E	2	
R	1	9	
R	30	90	
W	30	90	
R	30	90	
R	1	9	
R	30	90	
W	30	90	
R	30	90	
R	32	40	
W	32	40	
R	32	40	
R	10	FF	
W	10	FF	
R	10	FF	
W	3F	10	
R	30	90	
W	30	D0	
R	30	D0	
R	31	40	
W	31	40	
R	31	40	
R	27	91	
W	27	B1	
R	27	B1	
R	35	F	
W	35	8F	
R	35	8F	
R	35	8F	
W	35	8F	
R	35	8F	
R	20	0	
W	20	0	
R	20	0	
R	21	0	
W	21	0	
R	21	0	
R	22	0	
W	22	0	
R	22	0	
R	1F	40	
W	1F	AA	
R	1F	AA	
R	25	0	
W	25	0	
R	25	0	
R	26	10	
W	26	3	
R	26	3	
R	44	2D	
W	44	AD	
R	44	AD	
R	A	9	
W	A	49	
R	A	49	
R	1	9	
R	1	9	
W	1	9	
R	1	9	
R	6	6C	
W	6	6C	
R	6	6C	
R	7	80	
W	7	80	
R	7	80	
R	8	0	
W	8	0	
R	8	0	
R	1	9	
W	1	9	
R	1	9	
R	9	4F	
W	9	CF	
R	9	CF	
R	9	CF	
W	9	F8	
R	9	F8	
R	4D	84	
W	4D	84	
R	4D	84	
R	1	9	
R	1	9	
W	1	9	
R	1	9	
R	A	49	
W	A	9	
R	A	9	
success!
[AX.25] Initializing ... R	1	9	
R	1	9	
W	1	9	
R	1	9	
R	4	0	
W	4	0	
R	4	0	
R	5	8	
W	5	0	
R	5	0	
R	1	9	
R	30	D0	
W	30	90	
R	30	90	
success!
```

closed 2020-11-20 by jgromes