jgromes/RadioLib #195: RTTY example with ESP8266 + SI4432  failed code -102

fazerlab, 2020-11-03:
I'm trying to get a SI4432 module working with RadioLib on a ESP8266 WEMOS D1 module and RTTY example.

Wiring to ESP8266 Hardware SPI and additional pins nSEL, nIRQ, SDN
Si4432 radio = new Module(D8, D2, D0); //ESP8266 Pin

When i  call: state = rtty.begin(434.0, 156, 45); // Author Note: RTTY frequency shift  Si443x/RFM2x - 156 Hz
I get:  failed, code -102 
Defined as:
#define  ERR_INVALID_FREQUENCY_DEVIATION   -102
The supplied frequency deviation value is invalid.
I tried other values to frequency deviation and get the same error.

I try the same hardware  with a test code with RF22 librarie and work. I  get the signal in SDR: [Link to code](https://pastebin.com/hDx62yf7)

Here the code with Radiolib and follow the debug

```
// include the library
#include <RadioLib.h>

// Si4432 has the following connections:
// nSEL pin:  10
// nIRQ pin:  2
// SDN pin:   9
//Si4432 radio = new Module(nSEL,nIRQ,SDN);
Si4432 radio = new Module(D8, D2, D0); //ESP8266 Pin

// create RTTY client instance using the FSK module
RTTYClient rtty(&radio);

void setup() {
  Serial.begin(9600);
   delay (1000);
  Serial.println("");

    // when using one of the non-LoRa modules for RTTY
  // (RF69, CC1101, Si4432 etc.), use the basic begin() method
  int state = radio.begin();

  if(state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while(true);
  }

  // initialize RTTY client
  // NOTE: RTTY frequency shift will be rounded
  //       to the nearest multiple of frequency step size.
  //       The exact value depends on the module:
  //         SX127x/RFM9x - 61 Hz
  //         RF69 - 61 Hz
  //         CC1101 - 397 Hz
  //         SX126x - 1 Hz
  //         nRF24 - 1000000 Hz
  //         Si443x/RFM2x - 156 Hz
  //         SX128x - 198 Hz
  Serial.print(F("[RTTY] Initializing ... "));
  // low ("space") frequency:     434.0 MHz
  // frequency shift:             183 Hz
  // baud rate:                   45 baud
  // encoding:                    ASCII (7-bit)
  // stop bits:                   1
  state = rtty.begin(434.0, 156, 45);
  if(state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    //while(true);
    delay(30000);
  }
}

void loop() {
  Serial.print(F("[RTTY] Sending RTTY data ... "));
  // send out idle condition for 500 ms
  rtty.idle();
  delay(500);
  // Arduino String class
  String aStr = "Arduino String";
  rtty.println(aStr);  
  Serial.println(F("done!"));
  // wait for a second before transmitting again
  delay(1000);
}
```

Debug:

```
iniciando. . . 
R	1	6	
Found Si443x!
R	3	20	3	
W	6	0	
R	33	22	
W	33	82	
R	33	82	
R	32	C	
W	32	0	
R	32	0	
R	70	C	
W	70	C	
R	70	C	
W	6E	C	
W	6F	49	
R	1C	1	
R	1C	1	
R	70	C	
0
0
0
10.417	10	A	3	3	83	53
100663	18937
65535	FFFF
R	21	1	
W	21	1	
R	21	1	
R	20	64	
W	20	53	
R	20	53	
R	21	1	
W	21	1	
R	21	1	
R	22	47	
W	22	89	
R	22	89	
R	23	AE	
W	23	37	
R	23	37	
R	24	2	
W	24	7	
R	24	7	
R	25	8F	
W	25	FF	
R	25	FF	
R	71	0	
W	71	0	
R	71	0	
W	72	50	
R	1C	1	
W	1C	96	
R	1C	96	
R	1C	96	
R	1C	96	
R	70	C	
1
1
0
15.625	15	F	5	5	125	7D
67108	10624
2518	9D6
R	21	1	
W	21	1	
R	21	1	
R	20	53	
W	20	7D	
R	20	7D	
R	21	1	
W	21	1	
R	21	1	
R	22	89	
W	22	6	
R	22	6	
R	23	37	
W	23	24	
R	23	24	
R	24	7	
W	24	1	
R	24	1	
R	25	FF	
W	25	D6	
R	25	D6	
R	34	8	
W	34	2	
R	34	2	
R	35	2A	
W	35	42	
R	35	42	
R	33	82	
W	33	82	
R	33	82	
W	36	12	AD	
R	71	0	
W	71	20	
R	71	20	
R	70	C	
W	70	8	
R	70	8	
R	70	8	
W	70	8	
R	70	8	
R	75	75	
W	75	53	
R	75	53	
R	76	BB	
W	76	4A	
R	76	4A	
R	77	80	
W	77	FF	
R	77	FF	
R	6D	18	
W	6D	1B	
R	6D	1B	
success!
[RTTY] Initializing ... R	71	20	
W	71	20	
R	71	20	
W	72	0	
failed, code -102

```
I´m working in a educational project of a HAB and would like use Radiolib. Thank you

jgromes, 2020-11-03:
Thanks, there was a missing return statement which caused 0 kHz deviation to be rejected by a range check. Should be fixed in 063b643.

fazerlab, 2020-11-04:
> Thanks, there was a missing return statement which caused 0 kHz deviation to be rejected by a range check. Should be fixed in [063b643](https://github.com/jgromes/RadioLib/commit/063b643a7b5bf233f35e9f046318b9acdff5c5db).

Thank´s for your support.
This fix work, no more failed, code -102, but  i can´t see the signal in SDR.
The next step i enable Debug mode and could see signal in SDR. Maybe a delay?!
I didn´t try decode RTTY yet.
firmware used, and follow debug log:
```
// include the library
#include <RadioLib.h>

// Si4432 has the following connections:
// nSEL pin:  10
// nIRQ pin:  2
// SDN pin:   9
//Si4432 radio = new Module(nSEL,nIRQ,SDN);
Si4432 radio = new Module(D8, D2, D0);

// or using RadioShield
// https://github.com/jgromes/RadioShield
//SX1278 radio = RadioShield.ModuleA;

// create RTTY client instance using the FSK module
RTTYClient rtty(&radio);

void setup() {
  Serial.begin(9600); 
  Serial.println("starting.. ");
  delay(1000);

    // when using one of the non-LoRa modules for RTTY
  // (RF69, CC1101, Si4432 etc.), use the basic begin() method
  int state = radio.begin(433.0, 48.0, 50.0, 181.1, 10, 16);

  if(state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while(true);
  }

  // initialize RTTY client
  // NOTE: RTTY frequency shift will be rounded
  //       to the nearest multiple of frequency step size.
  //       The exact value depends on the module:
  //         SX127x/RFM9x - 61 Hz
  //         RF69 - 61 Hz
  //         CC1101 - 397 Hz
  //         SX126x - 1 Hz
  //         nRF24 - 1000000 Hz
  //         Si443x/RFM2x - 156 Hz
  //         SX128x - 198 Hz
  Serial.print(F("[RTTY] Initializing ... "));
  // low ("space") frequency:     434.0 MHz
  // frequency shift:             183 Hz
  // baud rate:                   45 baud
  // encoding:                    ASCII (7-bit)
  // stop bits:                   1
  state = rtty.begin(433.0, 156, 45);
  if(state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while(true);
  }

  ESP.wdtDisable();
}
[... 766 lines cut ...]
W	7	9	
W	75	13	
W	76	4A	
W	77	FF	
R	71	0	
W	71	0	
R	71	0	
R	71	0	
W	71	0	
R	71	0	
W	7	9	
W	75	13	
W	76	4B	
W	77	0	
R	71	0	
W	71	0	
R	71	0	
R	71	0	
W	71	0	
R	71	0	
W	7	9	
done!
fim

```

jgromes, 2020-11-04:
>  Maybe a timming issue?

That's possible, I'll have to check if I have an Si4432 to test with - I'm not sure I ever used it with ESP8266 to send RTTY.

jgromes, 2020-11-05:
I was able to replicate this on my setup - turns out the solution was pretty simple, there was one extra call to enable direct mode. This wasn't an issue on other modules (because direct mode implementations differ), or slower platforms like Arduino Uno, which is what I usually test with.

So this issue only popped up on a fast enough platform (like ESP8266) with this exact module. Fixed in the previous commit.

closed 2020-11-05 by jgromes

fazerlab, 2020-11-06:
I apply [RTTY] Fixed idle (#195) in my setup and work fine. I could decode RTTY with software SDR Console V.3.
Many, many thanks!!! I´m a Radiolib fan!! Radiolib is a great librarie