jgromes/RadioLib #196: Support for SI4063

jatin-bhosale, 2020-11-04:
Will the RadioLib support SI4063 transmitter?

jgromes, 2020-11-04:
I will accept PRs for that purpose, but don't plan on working on it myself due to time constraints.

jgromes, 2021-07-27:
Upon examining Si4063 datasheet, it doesn't seem that interesting - basically instead of full access to the SPI registers, there's a limited command interface. This means it's lacking a lot of features compared to the other modules supported by the library. There's also most likely no way to make it compatible with ham radio modes like RTTY or AX.25, which is a big part of RadioLib.

Unforuntately there are other priorities, so this won't be worked on.

closed 2021-07-27 by jgromes