jgromes/RadioLib #198: SX127x_Receive_Interrupt not working on my TTGO T-Beam V0.7

jctang46, 2020-11-07:
Hardware TTGO T-Beam V0.7 with ESP32 processor, LoRa RFM95 (SX1276) 915Mhz, and GPS.
RadioLib v4.0.5
https://github.com/jgromes/RadioLib/tree/master/examples/SX127x/SX127x_Receive_Interrupt
When the transmission side started to transfer message, At the receiver side, the flag receivedFlag in the example SX127x_Receive_Interrupt case never became TRUE.  If this is hardware problem or I still need to do 
something in the code? It seems the code radio.setDio0Action(setFlag); not working. 
But the example SX127x_Receive works fine, it could receive the message exactly the same as the transferred.

jgromes, 2020-11-07:
Assuming you're using the default unedited Receive_Interrupt example, you'll have to add an `IRAM_ATTR` attribute to the `setFlag` function, like this:

```c++
void IRAM_ATTR setFlag(void) {
  // check if the interrupt is enabled
  if(!enableInterrupt) {
    return;
  }

  // we got a packet, set the flag
  receivedFlag = true;
}
```

`setFlag()` is an interrupt service routine, on ESP32, these have to be placed into RAM.

jctang46, 2020-11-07:
Hi Jan,
I changed my code as that you give to me. But the flag is still not changed to true. Here I attached my real code which is a GPS tracker. Please take a look, give me some idea, if something is still wrong? Thanks.
James

jgromes, 2020-11-08:
Then I'd suggest checking the pins you set in `new Module()` in your sketch, if blocking receive is working but interrupt receive isn't, it suggests some issue with interrupt handling. I'll check on my setup.

There's nothing attached to your previous messsage (you can only attach files on GitHub, not by replying to the notification email), however, it's usually a lot easier trying to fix just one issue at a time, so I'd suggest getting Receive_Interrupt working first before adding more code.

jctang46, 2020-11-08:
Jan,
You are right! I changed new module() setting. Now it works well on interrupt.
Thank you Jan!

Best

James

closed 2020-11-09 by jgromes