jgromes/RadioLib #199: SI4432 + ESP8266 Packet Mode : No Receive packet

fazerlab, 2020-11-08:
I build a setup with SI4432 module and ESP8266 WEMOs D1 module. The default SPI was used in ESP8266 and addicional pins
```c++
Si4432 radio = new Module(D8, D2, D0);
```
The firmware was based in examples Si443x_Receive.ino and Si443x_Transmit.ino

Fail description:
- firmware Si443x_Receive.ino don´t receive message

Debug actions
- I used a SDR software and could see tx signal
- I try the same hardware with RadioHead firmware and work. Transmit and receive in two directions, Here link firmware  [client](https://gist.github.com/fazerlab/69c89891f27fb4b0376ea3a796a81983) e [server.](https://gist.github.com/fazerlab/8aef202a845e0a01a6ce04be636da07c) 

firmware:
Si443x_Transmit.ino
```c++
// include the library
#include <RadioLib.h>

// Si4432 has the following connections:
// nSEL pin:  10
// nIRQ pin:  2
// SDN pin:   9
//Si4432 radio = new Module(nSEL,nIRQ,SDN);
Si4432 radio = new Module(D8, D2, D0);

void setup() {
  Serial.begin(9600);
   delay (1000);
  Serial.print("iniciando. ");
   delay (1000);
  Serial.print(". ");
   delay (1000);
  Serial.print(". ");
  delay (1000);
  Serial.println("");

  // initialize Si4432 with default settings
  Serial.print(F("[Si4432] Initializing ... "));
  int state = radio.begin();  
  if (state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }
}

void loop() {
  Serial.print(F("[Si4432] Transmitting packet ... "));

  // you can transmit C-string or Arduino string up to
  // 64 characters long
  // NOTE: transmit() is a blocking method!
  //       See example Si443x_Transmit_Interrupt for details
  //       on non-blocking transmission method.
  int state = radio.transmit("Hello World!");

  // you can also transmit byte array up to 64 bytes long
  /*
    byte byteArr[] = {0x01, 0x23, 0x45, 0x56, 0x78, 0xAB, 0xCD, 0xEF};
    int state = radio.transmit(byteArr, 8);
  */

  if (state == ERR_NONE) {
    // the packet was successfully transmitted
    Serial.println(F(" success!"));

  } else if (state == ERR_PACKET_TOO_LONG) {
    // the supplied packet was longer than 256 bytes
    Serial.println(F(" too long!"));

  } else if (state == ERR_TX_TIMEOUT) {
    // timeout occured while transmitting packet
    Serial.println(F(" timeout!"));

  } else {
    // some other error occurred
    Serial.print(F("failed, code "));
    Serial.println(state);

  }

  // wait for a second before transmitting again
  delay(1000);
}

```
Si443x_Transmit.ino
```c++
// include the library
#include <RadioLib.h>

// Si4432 has the following connections:
// nSEL pin:  10
// nIRQ pin:  2
// SDN pin:   9
//Si4432 radio = new Module(nSEL,nIRQ,SDN);
Si4432 radio = new Module(D8, D2, D0);

void setup() {
  Serial.begin(9600);  
   delay (1000);
  Serial.print("iniciando. ");
   delay (1000);
  Serial.print(". ");
   delay (1000);
  Serial.print(". ");
  delay (1000);
  Serial.println("");

  // initialize Si4432 with default settings
  //https://github.com/jgromes/RadioLib/wiki/Default-configuration#si443xrfm2x
  Serial.print(F("[Si4432] Initializing ... "));

  int state = radio.begin();  
  if (state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }
}

void loop() {
  Serial.print(F("[Si4432] Waiting for incoming transmission ... "));
  //delay(1000);
  // you can receive data as an Arduino String
  String str;
  int state = radio.receive(str);  
  // you can also receive data as byte array
  /*
    byte byteArr[8];
    int state = radio.receive(byteArr, 8);
  */

  if (state == ERR_NONE) {
    // packet was successfully received
    Serial.println(F("success!"));

    // print the data of the packet
    Serial.print(F("[Si4432] Data:\t\t"));
    Serial.println(str);

  } else if (state == ERR_RX_TIMEOUT) {
    // timeout occurred while waiting for a packet
    Serial.println(F("timeout!"));

  } else if (state == ERR_CRC_MISMATCH) {
    // packet was received, but is malformed
    Serial.println(F("CRC error!"));

  } else {
    // some other error occurred
    Serial.print(F("failed, code "));
    Serial.println(state);

  }
}

```

Debug log  (link in pastebin)
- Debug: [Si443x_Receive.ino](https://pastebin.com/Xwnug7yR) 
- Debug: [Si443x_Transmit.ino](https://pastebin.com/rjKijCZu)

Thank you

jgromes, 2020-11-08:
I can confirm this is indeed happening, both in blocking and interrupt receive. Dumping interrupt status registers during reception shows packet received interrupt doesn't get triggered, so there's probably some incorrect configuration going on.

ChassaB, 2021-07-06:
It would seem this same issue exists with ESP32 using the default SPI and Si4432 radio = new Module(5, 2, 4)

tresler, 2021-07-12:
It is look like, that I have same issue with ESP32-S2 and RFM22-S2 868MHz. Transmitter write, that transmission is ok, but reciever can't recieve any message.

tresler, 2021-07-19:
> I can confirm this is indeed happening, both in blocking and interrupt receive. Dumping interrupt status registers during reception shows packet received interrupt doesn't get triggered, so there's probably some incorrect configuration going on.

Is it some new progress about this bug?

jgromes, 2021-07-19:
@tresler I'm aware of this bug and working on it - though anyone else who wants to take a shot at it is welcome to do so, it's been a quite annoying issue.

ChassaB, 2021-07-20:
> @tresler I'm aware of this bug and working on it - though anyone else who wants to take a shot at it is welcome to do so, it's been a quite annoying issue.

I'd like to help but I doubt my amateur programming skills are up to it. I'm happy to make a donation but can't see how to.

tresler, 2021-07-20:
Fo

> @tresler I'm aware of this bug and working on it - though anyone else who wants to take a shot at it is welcome to do so, it's been a quite annoying issue.

For me is it same story. I'd like to help but I don't understand it well. I can see on SDR some traffic, but it is maximum what I can say. I can only try to use ESP32-S2 as transmitter and Arduino as reciever for test if something happens.

jgromes, 2021-07-25:
Finally fixed this, the cause was a combination of multiple bugs, mainly incorrect preamble length configuration, issues in equations from datasheet and a cheeky little electrical problem in my breadboard, which almost drive me insane. This was without a doubt the most difficult issue so far.

Will release a fixed library version in the near future.

closed 2021-07-25 by jgromes

tresler, 2021-08-01:
Great work, thank you very much 

> Finally fixed this, the cause was a combination of multiple bugs, mainly incorrect preamble length configuration, issues in equations from datasheet and a cheeky little electrical problem in my [...]

Great work. Now its is works. Thank you very much...