jgromes/RadioLib #200: SX127x corrupted packets pass CRC check

jgromes, 2020-11-12:
As per the discussion with @amirna2 in https://github.com/jgromes/RadioLib/pull/111#issuecomment-725713815

CRC check in `SX127x::readData()` should take into account whether CRC is currently enabled or not, since in some cases, corrupted LoRa headers with CRC bit set to 0 still pass (somehow), which causes decoding to fail, probably because the module incorrectly configures the coding rate based on the corrupted information from packet header.

kiryam, 2020-11-19:
+1 for that I have this problem from time to time. Also I tested suggesting check in readData and it actually fixes that.

jgromes, 2020-11-19:
Should be resolved by c127712c5e4c90ad17ffe721ccbe025fb2d0cb21

closed 2020-11-19 by jgromes

amirna2, 2020-11-19:
This is great! Thank you for adding the fix.

TimoWielink, 2020-11-20:
Awesome Thnx!

knouse1344, 2020-11-20:
very helpful